zigpy/zha-device-handlers #3043: [Device Support Request] Moes ZHT-3000-GB-SS-MS thermostat floor temperature sensor

scp93ch, 2024-03-15:
### Problem description

I have connected this thermostat to Home Assistant using Zigbee Home Automation and the SkyConnect dongle. The basic features work: the current temperature, set temperature, on/off, and it appears as a functional climate device in HA. The thermostat is for electric underfloor heating and so has a room temperature sensor (which I can read) and a floor temperature sensor which I cannot read.

I previously opened issue #2479 which was closed by a bot as I didn't respond quickly enough. The info below is up to date.

### Solution description

I would like to be able to read the floor temperature over Zigbee.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_aoclfnxz",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_electric_heating.MoesBHT"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.3.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "6.6.16-haos",
    "supervisor": "2024.03.0",
[... 573 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

muellerjo, 2024-03-20:
I got the similar issue and want to upvote!!!

- Homeassistant
- SkyConnect
- Electric floor heating
- Tuya Thermostat
- All Works, but cant read floor temperature which results in max floor temperature of 45 °C

scp93ch, 2024-03-20:
With the default set-up I can read the room temperature from the device and control the floor heating that way. As the thermostat unit controls the underfloor heating in the bathroom but is not actually inside the bathroom, this is not ideal. For what it's worth, I am currently pairing the Moes device with a separate zigbee temperature sensor with the [Better Thermostat](https://better-thermostat.org/) add-on so that the underfloor heating reacts to the bathroom temperature. I'd still like to read the floor temperature as well for better control.

muellerjo, 2024-06-14:
Do you have any updates on reading the floor temperature of the Zigbee Thermostat?
I'm also looking parrallely for another thermostat that may provide all values (including the floor temperaure) via Zigbee (ZHA) but haven't found any.

scp93ch, 2024-06-14:
> Do you have any updates on reading the floor temperature of the Zigbee Thermostat? I'm also looking parrallely for another thermostat that may provide all values (including the floor temperaure) via Zigbee (ZHA) but haven't found any.

None from me, but I have not done any further investigation myself.

scp93ch, 2024-12-16:
I can confirm that with the latest version of everything the issue persists.

closed 2025-08-14 by github-actions[bot]