zigpy/zha-device-handlers #3044: [Device Support Request] Tongou TO-Q-SY2-163JZT Smart Circuit Breaker

MnM001, 2024-03-16:
### Problem description

When adding TO-Q-SY2-163JZT to ZHA is coming up as TS011F by _TZ3000_cayepv1a - however this seems to be another product -  TO-Q-SY1.
 TO-Q-SY1 can only do monitoring. 
TO-Q-SY2-163JZT can do monitoring and more: MCB Smart Circuit Breaker Over Current Under Voltage Protection Power Metering 1-63A Remote Control Switch. 
However none of the circuit breakers ar4 exposed to ZHA.
It will be good if this can be fixed.

### Solution description

TO-Q-SY2-163JZT needs to have the following in ZHA:

temperature protection
over current protection
over voltage protection
under voltage protection
high power protection
timing
and adjustable Functions for the above

as well functional real-time energy consumption monitoring  (I say functional as current monitoring is showing wrong values - see 
https://github.com/zigpy/zha-device-handlers/issues/2652 )

This unit is fully functional in zigbee2mqtt (https://www.zigbee2mqtt.io/devices/TO-Q-SY2-163JZT.html)

### Screenshots/Video

<details><summary>Screenshots from zigbee2mqtt</summary>

[image]

<details><summary>Screenshots from ZHA</summary>

[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_cayepv1a",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Milhauz666, 2024-04-23:
This is just a tweaked quirk reusing similar definition for [_TZ3000_qeuvnohg](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py#L1246), and my comment [here](https://github.com/zigpy/zha-device-handlers/issues/2652).

Current and Summation seem to work correctly. The temperature does not work and other circuit breaker functions have not been added. It's definitely not perfect and I'd appreciate any advice.

```
"""TS011F Circuit Breaker - Tongou TO-Q-SY2-JZT."""

from typing import Any, Dict

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.quirk_ids import TUYA_PLUG_ONOFF
from zhaquirks.tuya import (
    EnchantedDevice,
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaNewManufCluster,
    TuyaZB1888Cluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBMeteringClusterWithUnit,
    TuyaZBOnOffAttributeCluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)
from zhaquirks.tuya import TuyaDPType

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature data point."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
    }

class Plug_CB_Metering_v2(EnchantedDevice):
    """Circuit breaker with monitoring, e.g. Tongou TO-Q-SY2-JZT. First one using this definition was _TZ3000_cayepv1a"""

    quirk_id = TUYA_PLUG_ONOFF
   
    signature = {
        MODEL: "TS011F",
        MODELS_INFO: [("_TZ3000_cayepv1a", "TS011F")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 1794, 2820, 1026, 57344, 57345]
            # output_clusters=[25, 10]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaTemperatureMeasurement,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

neildsb, 2024-05-16:
Sorry to hijack a thread, 
I have a TO-Q-SY2-163JZT in the post and I plan to use with ZHA, can I assume the default is off for the 'breaker' protection variables and the max amp is 64? 
My primary use case is on/off (passing up-to 32A), anything else is a bonus!
Hopefully the parameters will the exposed in ZHA soon to configure and monitor :-)

franortiz, 2024-08-16:
I am new to zha and zigpy, but after looking the Z2M code and some debug with the device i was able to create this quirk and control the thresholds and breakers
Basically i created some fake attributes that converts to custom command when you write it.
I dont know if this is the zha way  but it seems to work

Please,  test it and will try to make a proper PR if everything is OK

```
"""TS011F Circuit Breaker * Tongou TO-Q-SY2-JZT."""

from typing import Any, Optional, Union
import logging
import enum
from struct import (iter_unpack, pack)

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.quirk_ids import TUYA_PLUG_ONOFF
from zhaquirks.tuya import (
    EnchantedDevice,
    TuyaNewManufCluster,
    TuyaZB1888Cluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBMeteringClusterWithUnit,
    TuyaZBOnOffAttributeCluster,
    TuyaLocalCluster,
)

_LOGGER = logging.getLogger("ts011f")

TUYA_OPTIONS_2_DATA = 0xE6
TUYA_OPTIONS_3_DATA = 0xE7

class Breaker(t.enum8):
    Off = 0x00
    On = 0x01

breakeable_threshold_attributes = {
    0xe605: ("temperature_breaker", Breaker),
    0xe685: ("temperature_threshold", t.uint16_t),
    0xe607: ("power_breaker", Breaker),
[... 188 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeThresholdCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

drudgebg, 2024-08-19:
I confirm that it works.

> I am new to zha and zigpy, but after looking the Z2M code and some debug with the device i was able to create this quirk and control the thresholds and breakers Basically i created some fake attributes that converts to custom command when you write it. I dont know if this is the zha way but it seems to work
> 
> Please, test it and will try to make a proper PR if everything is OK
> 
> ```
> """TS011F Circuit Breaker * Tongou TO-Q-SY2-JZT."""
> 
> from typing import Any, Optional, Union
> import logging
> import enum
> from struct import (iter_unpack, pack)
> 
> from zigpy.profiles import zgp, zha
> from zigpy.quirks import CustomDevice
> import zigpy.types as t
> from zigpy.zcl import foundation
> from zigpy.zcl.clusters.general import (
>     Basic,
>     GreenPowerProxy,
>     Groups,
>     Identify,
>     OnOff,
>     Ota,
>     Scenes,
>     Time,
> )
> from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
> from zigpy.zcl.clusters.lightlink import LightLink
> from zigpy.zcl.clusters.measurement import TemperatureMeasurement
> from zigpy.zcl.clusters.smartenergy import Metering
> 
> from zhaquirks import LocalDataCluster
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODEL,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.quirk_ids import TUYA_PLUG_ONOFF
> from zhaquirks.tuya import (
>     EnchantedDevice,
>     TuyaNewManufCluster,
>     TuyaZB1888Cluster,
>     TuyaZBE000Cluster,
>     TuyaZBElectricalMeasurement,
>     TuyaZBExternalSwitchTypeCluster,
>     TuyaZBMeteringCluster,
>     TuyaZBMeteringClusterWithUnit,
>     TuyaZBOnOffAttributeCluster,
>     TuyaLocalCluster,
> )
> 
> _LOGGER = logging.getLogger("ts011f")
> 
> TUYA_OPTIONS_2_DATA = 0xE6
> TUYA_OPTIONS_3_DATA = 0xE7
> 
> class Breaker(t.enum8):
>     Off = 0x00
>     On = 0x01
> 
> breakeable_threshold_attributes = {
>     0xe605: ("temperature_breaker", Breaker),
>     0xe685: ("temperature_threshold", t.uint16_t),
>     0xe607: ("power_breaker", Breaker),
[... 193 lines cut ...]
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TemperatureMeasurement.cluster_id,
>                     TuyaZBOnOffAttributeCluster,
>                     TuyaZBMeteringCluster,
>                     TuyaZBElectricalMeasurement,
>                     TuyaZBE000Cluster,
>                     TuyaZBExternalSwitchTypeThresholdCluster,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 PROFILE_ID: zgp.PROFILE_ID,
>                 DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
> ```

franortiz, 2024-08-19:
Future improvements will be available in [this gist](https://gist.github.com/franortiz/3fd91c6fa0884f2e33fa7e9d57ad0930)

franortiz, 2025-08-06:
Updated [gist](https://gist.github.com/franortiz/3fd91c6fa0884f2e33fa7e9d57ad0930) to support 2025.08

puddly, 2025-08-07:
@franortiz could you open a PR?

franortiz, 2025-08-15:
> [@franortiz](https://github.com/franortiz) could you open a PR?

PR #4267 submitted