zigpy/zha-device-handlers #3045: SPM01-D2TZ-ZM  Zemismart Energy Sensor TS0601 _TZE200_qhlxve78

kneschi2, 2024-03-16:
### Problem description

Paring went smoothly but I don't see any sensors or entities

### Solution description

Sensors and entities to display power in W and energy in kWh

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_qhlxve78",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.3.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.6.20-haos",
    "supervisor": "2024.03.0",
    "host_os": "Home Assistant OS 12.1",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "alarmo": {
      "version": "v1.9.15",
[... 433 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

kneschi2, 2024-03-16:
[image]

stast1, 2024-05-04:
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/15209362/ts0601_din_power.zip)
Here is the corrected version

stast1, 2024-05-05:
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/15212277/ts0601_din_power.zip)
Corrected version

agomezfabian, 2024-06-25:
Correcting @stast1 code, dividing ZEMISMART_VOLTAGE_ATTR by 10 since it was reporting the decimal point as unit.

```python
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes
from zhaquirks.tuya.ts0601_din_power import TuyaElectricalMeasurement, TuyaPowerMeasurement

"""Zemismart Power Meter Attributes"""
ZEMISMART_TOTAL_ENERGY_ATTR = 0x0201
ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR = 0x0202
ZEMISMART_POWER_FACTOR_ATTR = 0x020F
ZEMISMART_FREQUENCY_ATTR = 0x0265
ZEMISMART_VOLTAGE_ATTR = 0x0266
ZEMISMART_CURRENT_ATTR = 0x0267
ZEMISMART_POWER_ATTR = 0x0268

class ZemismartManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Zemismart SPM01 Power Meter device."""

    attributes = {
        ZEMISMART_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR: ("reverse_energy", t.uint32_t, True),
        ZEMISMART_CURRENT_ATTR: ("current", t.int16s, True),
        ZEMISMART_POWER_ATTR: ("power", t.uint16_t, True),
        ZEMISMART_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        ZEMISMART_FREQUENCY_ATTR: ("frequency", t.uint16_t, True),
        ZEMISMART_POWER_FACTOR_ATTR: ("power_factor", t.uint16_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == ZEMISMART_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_deliver_reported(value * 10)
        elif attrid == ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_receive_reported(value * 10)
        elif attrid == ZEMISMART_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == ZEMISMART_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value)
        elif attrid == ZEMISMART_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value / 10)
        elif attrid == ZEMISMART_FREQUENCY_ATTR:
            self.endpoint.electrical_measurement.frequency_reported(value)
        elif attrid == ZEMISMART_POWER_FACTOR_ATTR:
            self.endpoint.electrical_measurement.power_factor_reported(value)

class TuyaZemismartPowerMeter(CustomDevice):
    """Tuya power meter device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_qhlxve78", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                    ZemismartManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

dpetrini, 2024-07-02:
HI, the file above worked for me.
However as I use in Energy Panel in HA, it was not working on it.
So I used configuration from other similar device and now is working both as device and as energy meter.
(similar to ("_TZE200_bcusnqt8", "TS0601") )
I used as below:

```
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement

from zhaquirks import LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes
from zhaquirks.tuya.ts0601_din_power import TuyaElectricalMeasurement

"""Zemismart Power Meter Attributes"""
ZEMISMART_TOTAL_ENERGY_ATTR = 0x0201
ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR = 0x0202
ZEMISMART_VCP_ATTR = 0x0006

class ZemismartPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508

    AC_VOLTAGE_MULTIPLIER = 0x0600
    AC_VOLTAGE_DIVISOR = 0x0601
    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603

    _CONSTANT_ATTRIBUTES = {
        AC_VOLTAGE_MULTIPLIER: 1,
        AC_VOLTAGE_DIVISOR: 10,
        AC_CURRENT_MULTIPLIER: 1,
        AC_CURRENT_DIVISOR: 1000,
    }

    def voltage_reported(self, value):
        """Voltage reported."""
        self._update_attribute(self.VOLTAGE_ID, value)

    def power_reported(self, value):
        """Power reported."""
        self._update_attribute(self.POWER_ID, value)

    def current_reported(self, value):
        """Ampers reported."""
        self._update_attribute(self.CURRENT_ID, value)

class ZemismartElectricalMeasurement(TuyaElectricalMeasurement):
    """Custom class for total energy measurement."""

    POWER_UNIT = 0x300
    POWER_DIVISOR = 0x302
    POWER_WATT = 0x0000  # Actually this does not work. Data is interpreted as kWh.

    """Setting unit of measurement."""
    _CONSTANT_ATTRIBUTES = {0x0300: POWER_WATT, POWER_DIVISOR: 1000}

class ZemismartManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Zemismart SPM01 Power Meter device."""

    attributes = {
        ZEMISMART_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR: ("reverse_energy", t.uint32_t, True),
        ZEMISMART_VCP_ATTR: ("vcp_raw", t.data64, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == ZEMISMART_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_deliver_reported(value)
        elif attrid == ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_receive_reported(value)
        elif attrid == ZEMISMART_VCP_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(
                (value[7] * 256) + value[6]
            )
            self.endpoint.electrical_measurement.current_reported(
                (value[5] * 256 * 256) + (value[4] * 256) + value[3]
            )
            self.endpoint.electrical_measurement.power_reported(
                (value[2] * 256 * 256) + (value[1] * 256) + value[0]
            )

class TuyaZemismartPowerMeter(CustomDevice):
    """Tuya power meter device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0,
        # user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,
        # mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>,
        # manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        # maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>,
        # *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False,
        # *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True,
        # *is_security_capable=False)",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        MODELS_INFO: [
            ("_TZE200_qhlxve78" , "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51
            # device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    ZemismartManufCluster,
                    ZemismartElectricalMeasurement,
                    ZemismartPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

VasylSamoilov, 2024-08-19:
> Correcting @stast1 code, dividing ZEMISMART_VOLTAGE_ATTR by 10 since it was reporting the decimal point as unit.

Not needed, as it reports incorrect voltage (22.1 V instead of 221 V)

VasylSamoilov, 2024-08-19:
> HI, the file above worked for me. However as I use in Energy Panel in HA, it was not working on it. So I used configuration from other similar device and now is working both as device and as energy meter. (similar to ("_TZE200_bcusnqt8", "TS0601") ) I used as below:

It reports summation_delivered incorrectly (too small)

Venom96669, 2024-08-21:
> It reports summation_delivered incorrectly (too small)

Yes. Exactly 10 times .. :-)
I had the same.
It is decided by multiplication by 10, in the 81st line.
The modified option in the attachment ..

[ts0601_din_power.zip](https://github.com/user-attachments/files/16689937/ts0601_din_power.zip)

arkpoah, 2024-10-11:
I also bought this device and try to add it to ZHA with the quirk with last @dpetrini code.
As you can see below, device is well added in ZHA and seems using the quirk, but no sensors appear
[image]

Here the logs when adding device : 
```
homeassistant    | 2024-10-11 15:09:29.773 INFO (MainThread) [homeassistant.components.zha.websocket_api] Permitting joins for 60s
homeassistant    | 2024-10-11 15:09:45.177 INFO (MainThread) [zigpy.application] New device 0x19da (bc:02:6e:ff:fe:e1:a6:dd) joined the network
homeassistant    | 2024-10-11 15:09:45.178 INFO (MainThread) [zigpy.device] [0x19da] Requesting 'Node Descriptor'
homeassistant    | 2024-10-11 15:09:45.407 INFO (MainThread) [zigpy.device] [0x19da] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
homeassistant    | 2024-10-11 15:09:45.407 INFO (MainThread) [zigpy.device] [0x19da] Discovering endpoints
homeassistant    | 2024-10-11 15:09:45.482 INFO (MainThread) [zigpy.device] [0x19da] Discovered endpoints: [1]
homeassistant    | 2024-10-11 15:09:45.483 INFO (MainThread) [zigpy.device] [0x19da] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
homeassistant    | 2024-10-11 15:09:45.483 INFO (MainThread) [zigpy.endpoint] [0x19da:1] Discovering endpoint information
homeassistant    | 2024-10-11 15:09:45.577 INFO (MainThread) [zigpy.endpoint] [0x19da:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
homeassistant    | 2024-10-11 15:09:45.648 INFO (MainThread) [zigpy.device] [0x19da] Read model 'TS0601' and manufacturer '_TZE200_qhlxve78' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
homeassistant    | 2024-10-11 15:09:45.648 INFO (MainThread) [zigpy.device] [0x19da] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_qhlxve78' nwk=0x19DA ieee=bc:02:6e:ff:fe:e1:a6:dd is_initialized=True>
homeassistant    | 2024-10-11 15:09:45.832 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.tze200_qhlxve78_ts0601_rssi
homeassistant    | 2024-10-11 15:09:45.836 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.tze200_qhlxve78_ts0601_lqi
homeassistant    | 2024-10-11 15:09:45.838 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new update.zha entity: update.tze200_qhlxve78_ts0601_micrologiciel
```
Any idea plz ?
Thanks

Jo-Kaan, 2025-02-21:
Hello, this quirk for _TZE200_qhlxve78 works with ZHA.

```python
"""SPM01_Zigbee_V2"""
"""TS0601_TZE200_qhlxve78"""

from zigpy.quirks.v2.homeassistant import (
    UnitOfElectricCurrent,
    UnitOfElectricPotential,
    UnitOfEnergy,
    UnitOfFrequency,
    UnitOfPower,
)
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t

from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE200_qhlxve78", "TS0601")
    .tuya_sensor(
        dp_id=1,
        attribute_name="energy_consumed",
        fallback_name="Total energy consumed",
        type=t.uint16_t,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        divisor=100,
        translation_key="energy_consumed",
    )
    .tuya_sensor(
        dp_id=2,
        attribute_name="energy_produced",
        fallback_name="Total energy produced",
        type=t.uint16_t,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        divisor=100,
        translation_key="energy_produced",
    )
    .tuya_sensor(
        dp_id=111,
        attribute_name="total_active_power",
        fallback_name="total_active_power",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
    )
    .tuya_sensor(
        dp_id=101,
        attribute_name="ac_frequency",
        fallback_name="ac_frequency",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.FREQUENCY,
        unit=UnitOfFrequency.HERTZ,
        divisor=100,
    )
    .tuya_sensor(
        dp_id=15,
        attribute_name="power_factor",
        fallback_name="power_factor",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER_FACTOR,
        divisor=100,
    )
    .tuya_sensor(
        dp_id=102,
        attribute_name="voltage_phase",
        fallback_name="voltage_phase",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.VOLTAGE,
        divisor=10,
        unit=UnitOfElectricPotential.VOLT,
    )
    .tuya_sensor(
        dp_id=103,
        attribute_name="current_phase",
        translation_key="current_phase",
        fallback_name="Current (phase A)",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.CURRENT,
        divisor=1000,
        unit=UnitOfElectricCurrent.AMPERE,
    )
    .tuya_sensor(
        dp_id=104,
        attribute_name="power_phase",
        fallback_name="power_phase",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
    )

    .add_to_registry()
)
``` 

[image]

putimir, 2025-06-05:
I found a solution, for anyone that faces the same problem:

As seen from the logs, the device presents itself as 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.**ON_OFF_LIGHT**: 256>, 'input_clusters': [0, 4, 5, 61184], 'output_clusters': []}}}), empahasis on ON_OFF_LIGHT.

The quirk is looking (all devices in the quirk) for a SMART_PLUG device:

` DEVICE_TYPE: zha.DeviceType.SMART_PLUG,`

So in order to make this  switch to be recognised by the quirk, you have to:

1. edit the MODELS_INFO: to include the correct id (_TZE200_wvovwe9h in this case)
2. edit the DEVICE_TYPE to ON_OFF_LIGHT:

`DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,`

Pols88, 2025-07-02:
> I found a solution, for anyone that faces the same problem:
>
> As seen from the logs, the device presents itself as 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.**ON_OFF_LIGHT**: [...]

Hi, I have the same device, SPM01-D2TZ with a _TZE284_iwn0gpzz signature. I modified the quirk code but doesn't work:

[image]

Here the code:

`"""Tuya Din Power Meter."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement

from zhaquirks import LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes
from zhaquirks.tuya.ts0601_din_power import TuyaElectricalMeasurement

"""Zemismart Power Meter Attributes"""
ZEMISMART_TOTAL_ENERGY_ATTR = 0x0201
ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR = 0x0202
ZEMISMART_VCP_ATTR = 0x0006

class ZemismartPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508

    AC_VOLTAGE_MULTIPLIER = 0x0600
    AC_VOLTAGE_DIVISOR = 0x0601
    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603

    _CONSTANT_ATTRIBUTES = {
        AC_VOLTAGE_MULTIPLIER: 1,
        AC_VOLTAGE_DIVISOR: 10,
        AC_CURRENT_MULTIPLIER: 1,
        AC_CURRENT_DIVISOR: 1000,
    }

    def voltage_reported(self, value):
        """Voltage reported."""
        self._update_attribute(self.VOLTAGE_ID, value)

    def power_reported(self, value):
        """Power reported."""
        self._update_attribute(self.POWER_ID, value)

    def current_reported(self, value):
        """Ampers reported."""
        self._update_attribute(self.CURRENT_ID, value)

class ZemismartElectricalMeasurement(TuyaElectricalMeasurement):
    """Custom class for total energy measurement."""

    POWER_UNIT = 0x300
    POWER_DIVISOR = 0x302
    POWER_WATT = 0x0000  # Actually this does not work. Data is interpreted as kWh.

    """Setting unit of measurement."""
    _CONSTANT_ATTRIBUTES = {0x0300: POWER_WATT, POWER_DIVISOR: 1000}

class ZemismartManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Zemismart SPM01 Power Meter device."""

    attributes = {
        ZEMISMART_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR: ("reverse_energy", t.uint32_t, True),
        ZEMISMART_VCP_ATTR: ("vcp_raw", t.data64, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == ZEMISMART_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_deliver_reported(value)
        elif attrid == ZEMISMART_TOTAL_REVERSE_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_receive_reported(value)
        elif attrid == ZEMISMART_VCP_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(
                (value[7] * 256) + value[6]
            )
            self.endpoint.electrical_measurement.current_reported(
                (value[5] * 256 * 256) + (value[4] * 256) + value[3]
            )
            self.endpoint.electrical_measurement.power_reported(
                (value[2] * 256 * 256) + (value[1] * 256) + value[0]
            )

class TuyaZemismartPowerMeter(CustomDevice):
    """Tuya power meter device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0,
        # user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,
        # mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>,
        # manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        # maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>,
        # *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False,
        # *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True,
        # *is_security_capable=False)",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        MODELS_INFO: [
            ("_TZE200_wvovwe9h", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51
            # device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    ZemismartManufCluster,
                    ZemismartElectricalMeasurement,
                    ZemismartPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }`

Con you help me? TY

532910, 2025-08-31:
dup #3045

xumixu, 2025-09-10:
> It reports summation_delivered incorrectly (too small)

Seems there different firmware version, or maybe is the local language decimal point, but in my case the above code shown both voltage too large and power too large and this fixed it. So probably each person will have to check their units and correct them accordingly.