zigpy/zha-device-handlers #3046: New Device Support Request - CK-BL702-AL-01(7009_Z102LG03-1) by eWeLink - GU10 model

milkywade, 2024-03-17:
### Problem description

ZHA has no options to control light colors of this GU10 bulb from AliExpress:

https://fr.aliexpress.com/item/1005006099930203.html?spm=a2g0o.order_list.order_list_main.5.5f795e5bHoQ1qo&gatewayAdapt=glo2fra

### Solution description

When this lamp is connected to ZHA, home assistant should provide the handlers to change the lamp colors.

Also, a "color-to-set when power up" option should be available as well so when you switch on the lights, the color you preset should be used.

### Screenshots/Video

<details><summary>This is what I see on the ZHA page of my HASS setup</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4742, maximum_buffer_size=127, maximum_incoming_transfer_size=242, server_mask=11264, maximum_outgoing_transfer_size=242, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010d",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00",
        "0xfc11",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "CK-BL702-AL-01(7009_Z102LG03-1)",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Core",
    "version": "2024.3.1",
    "dev": false,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.12.2",
[... 475 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I found a related entry on zigbee2mqtt: https://github.com/Koenkk/zigbee2mqtt/issues/20999

milkywade, 2024-09-13:
Still not working using HA Core 2024.8.3

wbmk78, 2024-09-24:
I also would like this functionality...

ahgartner, 2024-11-11:
I also would like this functionality

closed 2025-08-14 by github-actions[bot]