zigpy/zha-device-handlers #3048: [BUG] Tuya TS0012 no entities for external switch

JanKrl, 2024-03-17:
### Bug description

External switch entities (S1 and S2) do not show up in Home Assistant.

### Steps to reproduce

Add device to HA using Zigbee

### Expected behavior

Both switches show up as HA entities

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_llfaquvp",
  "model": "TS0012",
  "class": "zhaquirks.tuya.ts001x.Tuya_Double_No_N_Plus"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[... 445 lines cut ...]
            "unsupported_attributes": {
              "0xd030": {
                "attribute_name": "external_switch_type"
              }
            }
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

javicalle, 2024-03-17:
Your device is loading the quirk and signature shows the OnOff clusters for both endpoints.
Have you tried to 'reconfigure' your device?

JanKrl, 2024-03-18:
I did. I also tried removing the device from HA and pairing it again.

Here is the result of reconfiguration. 1st time `Attribute on_off` was showing yellow icon. 2nd time it turned green.
[image]

How can I debug this issue further?

closed 2024-09-21 by github-actions[bot]