zigpy/zha-device-handlers #3049: [BUG] Failed to call service switch/turn_off. Failed to send request: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>

cdalexndr, 2024-03-18:
### Bug description

Device does not respond to command and UI shows error:
"Failed to call service switch/turn_off. Failed to send request: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>"

### Steps to reproduce

Issue switch on/off command

### Expected behavior

to work

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=82, maximum_incoming_transfer_size=1024, server_mask=11264, maximum_outgoing_transfer_size=1024, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0002",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0006",
        "0x0007",
        "0x0b05",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "SONOFF",
  "model": "ZBMINIL2",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.1",
    "docker": true,
    "arch": "x86_64",
[... 383 lines cut ...]
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 96, in async_turn_off
    await self._on_off_cluster_handler.turn_off()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/general.py", line 453, in turn_off
    result = await self.off()
             ^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 83, in wrapper
    with wrap_zigpy_exceptions():
  File "/usr/local/lib/python3.12/contextlib.py", line 158, in __exit__
    self.gen.throw(value)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 75, in wrap_zigpy_exceptions
    raise HomeAssistantError(message) from exc
homeassistant.exceptions.HomeAssistantError: Failed to send request: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>
```

</details>

[home-assistant_zha_2024-03-18T20-28-21.159Z.log](https://github.com/zigpy/zha-device-handlers/files/14641992/home-assistant_zha_2024-03-18T20-28-21.159Z.log)

MattWestb, 2024-03-18:
From the device signature:
```
*is_end_device=True, 
*is_full_function_device=False, 
*is_mains_powered=False, 
*is_receiver_on_when_idle=False, 
*is_router=False
```
So its one sleeping end device and need being waked up then sending things to it or its not listening and can receiving commands.

cdalexndr, 2024-04-11:
Issue self resolved a few days ago, now working fine.

Sixfifty, 2024-06-06:
I'm having this issue while attempting to use a switch on the Dashboard to control an IKEA TRETAKT switch.

[image]

closed 2024-12-10 by github-actions[bot]