zigpy/zha-device-handlers #3052: [Device Support Request] Add support for Giex GX02 water timer

Aazamandius, 2024-03-19:
### Problem description

There are a new water timer Giex GX02

I tied to make it works with quirks - and now it show watering, litres and single switch

It's just need to replace here 
MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"), ("_TZE200_a7sghmms", "TS0601")],
with
MODELS_INFO: [("_TZE204_7ytb3h8u", "TS0601")]

I'm not sure can it show device settings like start or stop watering timers or other local automations (stop after 60 litres for example)

### Solution description

Please add support this device to handler
And if possible make UI settings for locat device automations settings

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.2.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.1",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Moscow",
    "os_name": "Linux",
    "os_version": "6.6.16-haos",
    "supervisor": "2024.03.0",
    "host_os": "Home Assistant OS 12.0",
    "docker_version": "24.0.7",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.34.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "delete": {
      "version": "1.9",
      "requirements": []
    },
    "gismeteo": {
      "version": "2.5.2",
      "requirements": [
        "colorlog==6.7.0",
        "ruff==0.0.291"
      ]
    }
  },
  "integration_manifest": {
[... 409 lines cut ...]
```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

WJ4IoT, 2024-04-04:
Install the latest version of Z2M ([1.36.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) commit: [ffc2ff1](https://github.com/Koenkk/zigbee2mqtt/commit/ffc2ff1)) and this device is supported, so far looks good, but my device is just arrived and garden does not need extra water yet.

pablopezg, 2024-05-28:
I've seen that only battery and ON/OFF features are available right now. When do you expect to have the rest?

Pokeypoet, 2024-05-29:
Hello, Could you please paste your custom quirk here please? I don't see to be able to make it work and unsure if it's the quirk or something else I am doing wrong.

pablopezg, 2024-05-29:
Hello,

The custom quirk that I use is the one here: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_valve.py

Now I realized that after rebooting the HA two new sensors appears:
[image]

But I was wondering about the rest of sensors that should be available on such device:
- battery (available already)
- state (available already)
- mode
- cycle_irrigation_num_times
- irrigation_start_time
- irrigation_end_time
- last_irrigation_duration
- water_consumed (available already)
- irrigation_target
- cycle_irrigation_interval
- linkquality (available already)

Pokeypoet, 2024-05-29:
Great, thanks for the link to quirk, work for me now 👍

I would also be very interested in implementing the remaining sensors.

closed 2024-10-20 by TheJulianJES