zigpy/zha-device-handlers #3053: Moes scene switch TS004F by _TZ3000_xabckq1v

jrmurdoch, 2024-03-20:
### Problem description

Please add support

### Solution description

Support to be added.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    }
[... 506 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

This is a Moes 4 button wireless scene switch as here: [https://moeshouse.com/products/zigbee-wireless-scene-switch-battery-powered?variant=43700045283643](url)

MattWestb, 2024-03-20:
Wot function is missing on your device ?
The system have loaded "zhaquirks.tuya.ts004f.TuyaSmartRemote004FDMS" so you shall having all function the device is having both light searing an "scene" mode.

jrmurdoch, 2024-03-20:
This is what I see under the device.

[image]

I expected to see somthing similar to the following:

[image]

Note the controls section with although I expected to see 4 separate buttons.

Also to note is that the button presses are detected, in the logs somewhat. this is in order left -> right, top -> bottom

[image]

MattWestb, 2024-03-21:
Switch entries you is getting for things you can turn on and off like outlets and lights.
Your device is one controller (sending commands) and can only sending  commands (light or scene ones its depends of the mode it is in) and you is only getting the battery and  identity.
You is having device automations for the device and can changing mode from light controller to "tuya scene" so can  geting more commands from the device.
Read the initial issue for the TS004F there you is having all information.

jrmurdoch, 2024-03-21:
I'm a bit lost sorry. I looked at automations and yes can get the "turn on event" to work. Although is it possible to edit this "turn on event" as it should be "button 1 pressed event"? Essentially the buttons don't register the correct event.

MattWestb, 2024-03-21:
The button 1-4 is only for "tua scene mode" commands so you need changing the working mode to it it by setting the mode attribute and waking the device up or doing it with the button pressing on the device.
Read the initial issues and you is having all there (use search "ts004f"):

closed 2024-09-24 by github-actions[bot]