zigpy/zha-device-handlers #3058: [Device Support Request]  Tuya TS0601 by_TZE204_r0jdjrvi smart curtain motor

YasserAntonio, 2024-03-21:
### Problem description

using homeassistant/home-assistant:2024.3  on a Raspberry pi 3, having no issue with other zigbee devices.

After syncing TS0601 by_TZE204_r0jdjrvi smart curtain motor with HA I had no control feature and only LQI and RSSI with no info. I dont know how to get the signature of the device but I would be glad to provide it if you tell me how.

device details : https://fr.aliexpress.com/item/1005005060508822.html?spm=a2g0o.order_list.order_list_main.5.19de5e5b6IosoP&gatewayAdapt=glo2fra

### Solution description

whathever that could make the device work

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

javicalle, 2024-03-22:
From the device information view:
[image]

For the signature, select the option 'Administrate Zigbee device' ("Administrar dispositivo zigbee" in my screenshoot) and then the 'signature' tab.

imaginepast, 2024-03-31:
[zha-0375ed0f63267e3bcdaaff17a1b96c7e-_TZE204_r0jdjrvi TS0601-78cac4173774a17ddd60e2ed39e14ab2.json](https://github.com/zigpy/zha-device-handlers/files/14816925/zha-0375ed0f63267e3bcdaaff17a1b96c7e-_TZE204_r0jdjrvi.TS0601-78cac4173774a17ddd60e2ed39e14ab2.json)

imaginepast, 2024-03-31:
I have the same problem, I am attaching the document.

mandishshah, 2024-04-27:
I've trying to configure the same motor type. Have you been able to figure this out?

YasserAntonio, 2024-08-10:
not for the moment

YasserAntonio, 2024-09-24:
solved here : https://github.com/zigpy/zha-device-handlers/issues/1953 
solved by adding  a custom quirk describing and interacting with the motor using a python. Works for me now

pixselious, 2024-09-24:
Here is a step by step guide for complete noobs like myself.

HASS 2024.9.3
SONOFF Zigbee 3.0 USB Dongle Plus V2
TS0601 by _TZE204_r0jdjrvi aka Tuya zigbee smart curtain motor

1. cd to the directory with your configuration.yaml file
2. create a new directory called `custom_zha_quirks`
3. cd into that directory 
4. create a new file called `TZE204_r0jdjrvi.py` (i don't know for sure, but I'm guessing the file name has to match your device code)
5. paste the code from this comment into the new file https://github.com/zigpy/zha-device-handlers/issues/1953#issuecomment-1666933123
[backup](https://github.com/user-attachments/files/17116440/TZE200_r0jdjrvi.py.txt)

7. go to line 261 and adjust the strings to match your device code exactly. in our case it should be like this:
```
MODELS_INFO: [
    ("_TZE204_r0jdjrvi", "TS0601"),
],
```
8. save the new file
9. paste this into your configuration.yaml file:
  ```
  zha:
    custom_quirks_path: /config/custom_zha_quirks/
  ```
10. if you have previously paired the device to ZHA, unpair it.
[image]

11. check the config file and restart home assistant
12. pair the device like you did before (to put the motor into pairing mode press the button 3 times, that hold it for a few seconds)
13. done! now you should have some new buttons and a slider:

[image]
[image]

YasserAntonio, 2024-09-24:
Nice work ! It will help a lot of people, thanks !

vothar, 2025-04-24:
thanks @pixselious  ..it work also for my gen 4 tuya cover motor model : TS0601 by _TZE204_2rvvqjoa  ( [](https://www.aliexpress.com/item/1005007863513974.html) )

open / close / stop button are ok .  but i can't set a start or stop position so the % open didn't work

davidliau, 2025-12-04:
Thanks so much @pixselious !!

**"Closing" state disappears after a few seconds but motor is still spinning**

After adapting the custom quirk you indicated, I have an issue with the "Close" button when pressed: The "closing" state is displayed BUT only for a few seconds and then the state disappears, even though the motor is still spinning.

Conversely, the "Open" button when pressed will continuously display the state "opening" until the motor is stopped manually or stop spinning after sometime. Do you or anyone encounter this issue ??

I'm testing the TS0601_TZE204_r0jdjrvi motor on its own (that is, powered up via AC mains but not connected to any curtain rails yet).

Any help is much appreciated! Thanks!