zigpy/zha-device-handlers #3061: [Device Support Request]

agvg1, 2024-03-22:
### Problem description

Tuya PIR have no functions, pairs fine and according to ZHA have the latest firmware.

### Solution description

Get it to have some more function than using battery and displaying firmware version :)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ghynnvos",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

agvg1, 2024-03-22:
[image]

Seems to have a sensor in the corner,  the seller has very little information on functions and I don't have a Tuya Zigbee controller for testing at the moment

mrnutsen, 2024-07-23:
Any updates or custom quirks available for this? I have the same sensor, only firmware is showing after "successful" pairing in HA.

Haprog, 2024-07-27:
I just bought the same device. I'm not using ZHA (but Zigbee2MQTT) but I want to add some context as this is the only page I could find about this specific device. Z2M doesn't even show firmware version or battery level for this.

This is being sold in Finland (by Power, a Nordic consumer electronics chain) under brand name Conecto with model name: COZIGPMS (I guess Conecto is all just rebranded Tuya devices?).

I'm a developer and interested in helping to add support for this (at least for Z2M, maybe also ZHA if I can help) but I'm pretty new to HA, ZHA and Z2M even as a user so I'm not sure what it requires or how deep reverse engineering would be needed. I'm trying to start from here https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

I'm not sure if having it working in Z2M would help with ZHA side also.

Haprog, 2024-07-27:
I did some digging in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts searching by "TS0601" (there are hundreds of results) and looking for anything related to motion sensing.

I found this [this section](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L7483-L7500) which describes a "Haozee PIR sensor" (Tuya TS0601 by _TZE200_f1pvdgoh). I did a google search for that and found this issue https://github.com/Koenkk/zigbee2mqtt/issues/12883 which links to an [Ali Express listing](https://www.aliexpress.com/item/1005003891641858.html) for a sensor that looks pretty much identical to this one so I guess it's basically the same device (just identified differently).

I was able to get mine to work on Z2M just by making a custom converter by reusing the main parts of the configuration defined for the Haozee PIR sensor.

This is my current JS definition file that seems to work perfectly with Z2M.

```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;

const definition = {
  fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ghynnvos']),
  model: 'TS0601_pir_2',
  vendor: 'Tuya',
  description: 'PIR Motion Sensor',
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  onEvent: tuya.onEvent(),
  configure: tuya.configureMagicPacket,
  exposes: [e.occupancy(), e.illuminance(), e.battery()],
  meta: {
    tuyaDatapoints: [
      [1, 'occupancy', tuya.valueConverter.trueFalse0],
      [4, 'battery', tuya.valueConverter.raw],
      [101, 'illuminance', tuya.valueConverter.raw],
    ],
  },
  whiteLabel: [
    tuya.whitelabel('Conecto', 'COZIGPMS', 'PIR Motion Sensor', ['_TZE200_ghynnvos']),
  ],
};

module.exports = definition;
```

By quick testing at least occupancy and illuminance status works fine. Can't see battery percentage yet, but I guess it might start reporting later after running for some time. I guess the small black dot on the device is the illuminance sensor.

Mastiffen, 2024-08-16:
@Haprog Hej! :) I have added this to my setup, but the occupancy is very flakey. The light detection is working well, though. The strange thing is that sometimes the occupancy is set as occupied for hours, then sudden it won't see occupancy at all. Have you seen anything like that?

Haprog, 2024-08-16:
@Mastiffen Terve! I have not seen such problems. For me the occupancy has been reliable. I use this to automate lights on/off in one room so it's used pretty much every day and has been reliable so far.

Maybe your device is broken? Does it have good signal to the Zigbee network? I also bought two more of the same device, but have not taken those into use yet.

Mastiffen, 2024-08-16:
Aha, Finland, inte Sverige! Min fru er 2/3 finsk.
Thanks! Maybe it's the placement. It's in the ceiling pointing down towards the shower (lights went off when my wife used more than five minutes in there), but it's only 10 cm from the frame of the shower. I'll try to move it.

Haprog, 2024-08-19:
I use mine in a utility/laundry room (kodinhoitohuone) which is mostly used as a corridor to get to the larger bathroom and sauna. I made an automation in Home Assistant to immediately turn off the lights when no motion is detected (without extra timeout) and it seems to take around 20-30 seconds until the lights turn off if I just walk through so there's probably around 20s internal delay in the device until it updates the occupancy state when motion is not detected anymore.

If I stay in the room e.g. to do empty the washing machine, it does sometimes turn off the light while I'm there when I'm not moving much, but it turns back on when I wave or do a bigger movement. I could add a timeout on the automation side so it does not turn off until occupancy has been cleared for X amount of time, but most of the time I want it to turn off quicker so I haven't done that. Your shower use case might work better with some added timeout.

But I think if you have it directly above or too close to the shower the sensor might get "cloudy" from the moisture in the air, when the shower is being used, and thus does not see movement as well which might add to your issue? I also wouldn't expect this device to last long in so humid environment as it's probably not that well insulated and moisture might get inside it with some time and cause electrical issues.

Mastiffen, 2024-08-22:
@Haprog  Of course you have it in a corridor to the sauna... ;) But I gave up this one and put up a Fibaro "Eye" that I had laying around instead. Problem solved, no more showers in the dark. I will take down this one and see if it works better in my house, where I am using a ConBee II, not a cheap Ali Express Zigbee. That may be related.

valzzu, 2024-12-01:
> I did some digging in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts searching by "TS0601" (there are hundreds of results) and looking for anything related to motion sensing.
> 
> I found this [this section](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L7483-L7500) which describes a "Haozee PIR sensor" (Tuya TS0601 by _TZE200_f1pvdgoh). I did a google search for that and found this issue [Koenkk/zigbee2mqtt#12883](https://github.com/Koenkk/zigbee2mqtt/issues/12883) which links to an [Ali Express listing](https://www.aliexpress.com/item/1005003891641858.html) for a sensor that looks pretty much identical to this one so I guess it's basically the same device (just identified differently).
> 
> I was able to get mine to work on Z2M just by making a custom converter by reusing the main parts of the configuration defined for the Haozee PIR sensor.
> 
> This is my current JS definition file that seems to work perfectly with Z2M.
> 
> ```js
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> 
> const definition = {
>   fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ghynnvos']),
>   model: 'TS0601_pir_2',
>   vendor: 'Tuya',
>   description: 'PIR Motion Sensor',
>   fromZigbee: [tuya.fz.datapoints],
>   toZigbee: [tuya.tz.datapoints],
>   onEvent: tuya.onEvent(),
>   configure: tuya.configureMagicPacket,
>   exposes: [e.occupancy(), e.illuminance(), e.battery()],
>   meta: {
>     tuyaDatapoints: [
>       [1, 'occupancy', tuya.valueConverter.trueFalse0],
>       [4, 'battery', tuya.valueConverter.raw],
>       [101, 'illuminance', tuya.valueConverter.raw],
>     ],
>   },
>   whiteLabel: [
>     tuya.whitelabel('Conecto', 'COZIGPMS', 'PIR Motion Sensor', ['_TZE200_ghynnvos']),
>   ],
> };
> 
> module.exports = definition;
> ```
> 
> By quick testing at least occupancy and illuminance status works fine. Can't see battery percentage yet, but I guess it might start reporting later after running for some time. I guess the small black dot on the device is the illuminance sensor.

this seems to have worked, thx!

valzzu, 2024-12-04:
now that it has been running a while its not reporting battery

BeeBerg, 2025-01-12:
Decision
I tried many different options, only this one worked. https://4pda.to/forum/index.php?showtopic=871505&view=findpost&p=133184799

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_gjldowol',
        },
    ],
    model: 'TZE200_gjldowol',
    vendor: 'Tuya',
    description: 'Luminance motion sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.occupancy(),
            e.illuminance().withUnit('lx'),
            e.battery(),
            e
                .enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
                .withDescription('PIR sensor sensitivity (refresh and update only while active)'),
            e
                .enum('keep_time', ea.STATE_SET, ['10', '30', '60', '120'])
                .withDescription('PIR keep time in seconds (refresh and update only while active)'),
            e
                .numeric('illuminance_interval', ea.STATE_SET)
                .withValueMin(1)
                .withValueMax(720)
                .withValueStep(1)
                .withUnit('minutes')
                .withDescription('Brightness acquisition interval (refresh and update only while active)'),
        ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'occupancy', tuya.valueConverter.trueFalse0],
            [4, 'battery', tuya.valueConverter.raw],
            [9, 'sensitivity', tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
            [101, 'keep_time', tuya.valueConverterBasic.lookup({'10': tuya.enum(0), '30': tuya.enum(1), '60': tuya.enum(2), '120': tuya.enum(3)})],
            [12, 'illuminance', tuya.valueConverter.raw],
            [102, 'illuminance_interval', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

prairiesnpr, 2025-01-12:
@BeeBerg are you saying this is the same as TZE200_gjldowol? If so see if https://github.com/zigpy/zha-device-handlers/pull/3705 works for you.

mingalsuo, 2025-07-23:
> I did some digging in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts searching by "TS0601" (there are hundreds of results) and looking for anything related to motion sensing.
> 
> I found this [this section](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L7483-L7500) which describes a "Haozee PIR sensor" (Tuya TS0601 by _TZE200_f1pvdgoh). I did a google search for that and found this issue [Koenkk/zigbee2mqtt#12883](https://github.com/Koenkk/zigbee2mqtt/issues/12883) which links to an [Ali Express listing](https://www.aliexpress.com/item/1005003891641858.html) for a sensor that looks pretty much identical to this one so I guess it's basically the same device (just identified differently).
> 
> I was able to get mine to work on Z2M just by making a custom converter by reusing the main parts of the configuration defined for the Haozee PIR sensor.
> 
> This is my current JS definition file that seems to work perfectly with Z2M.
> 
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> 
> const definition = {
>   fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ghynnvos']),
>   model: 'TS0601_pir_2',
>   vendor: 'Tuya',
>   description: 'PIR Motion Sensor',
>   fromZigbee: [tuya.fz.datapoints],
>   toZigbee: [tuya.tz.datapoints],
>   onEvent: tuya.onEvent(),
>   configure: tuya.configureMagicPacket,
>   exposes: [e.occupancy(), e.illuminance(), e.battery()],
>   meta: {
>     tuyaDatapoints: [
>       [1, 'occupancy', tuya.valueConverter.trueFalse0],
>       [4, 'battery', tuya.valueConverter.raw],
>       [101, 'illuminance', tuya.valueConverter.raw],
>     ],
>   },
>   whiteLabel: [
>     tuya.whitelabel('Conecto', 'COZIGPMS', 'PIR Motion Sensor', ['_TZE200_ghynnvos']),
>   ],
> };
> 
> module.exports = definition;
> 
> By quick testing at least occupancy and illuminance status works fine. Can't see battery percentage yet, but I guess it might start reporting later after running for some time. I guess the small black dot on the device is the illuminance sensor.

Any idea what to do if Z2M considers this invalid and refuses? ...Other than Finnish 🤬-words 🤣 

I'm a total beginner and was quite lucky to find instructions on where to make this script available to Z2M, `external_converters/*.mjs` right, but find myself quite uneducated again.

Kiitos ja anteeksi...