zigpy/zha-device-handlers #3067: Ignore problematic firmware updates of Aqara MAEU01 plugs

FlorianFreudiger, 2024-03-24:
### Bug description

As already established, newer firmware versions (>v32) of the Aqara MAEU01 plugs have the problem of toggling on/off when certain remotes are used. Because v32 is already well supported, ZHA should not offer firmware updates beyond v32.

Relevant information on the toggling issue:

https://github.com/zigpy/zha-device-handlers/blob/4b3d6591c72302fd82e1b5a6e4ee30eb2e08dd9c/zhaquirks/xiaomi/aqara/plug_eu.py#L44-L47

- https://github.com/zigpy/zha-device-handlers/pull/1656#issuecomment-1244750465
- https://github.com/zigpy/zha-device-handlers/pull/1706
- https://github.com/Koenkk/zigbee2mqtt/issues/13903

### Steps to reproduce

1. Have Aqara MAEU01 plug with v32 firmware version connected.
2. Browse to the Settings in Home Assistant
3. See ZHA offer plug update from 0x00000020 (v32) to 0x00000029 (v41)

### Expected behavior

ZHA to not offer an update beyond v32 for these plugs.

TheJulianJES, 2024-03-24:
What do you have in your `configuration.yaml`?

FlorianFreudiger, 2024-03-24:
> What do you have in your `configuration.yaml`?

```
zha:
  zigpy_config:
    ota:
      ikea_provider: true
      ledvance_provider: true
      salus_provider: true
      inovelli_provider: true
      otau_directory: /config/zigpy_ota
```
I previously used this configuration to be able to manually downgrade to the v32 version. The `/config/zigpy_ota` has since been empty.

TheJulianJES, 2024-03-24:
ZHA doesn't offer any updates for Xiaomi/Aqara devices by default. Double check that your OTA directory is empty.

FlorianFreudiger, 2024-03-24:
> ZHA doesn't offer any updates for Xiaomi/Aqara devices by default. Double check that your OTA directory is empty.

```
[core-ssh ~]$ cd /config/zigpy_ota
[core-ssh zigpy_ota]$ ls -la
total 8
drwxr-xr-x    2 root     root          4096 Jan 25  2023 .
drwxr-xr-x   14 root     root          4096 Mar 24 21:45 ..
[core-ssh zigpy_ota]$
```

Back when I downgraded my plugs, I only uploaded a v32 firmware file modded to v42 id, but I have removed the firmware file from the directory since then.

[image]

So I'm not sure where this update originates from, but the update popped up after I updated from Home Assistant 2024.2 to 2024.3 a few minutes earlier.

FlorianFreudiger, 2024-03-24:
I also don't have the `allow_file_providers: "I understand I can *destroy* my devices by...` config option in my `configuation.yaml` which is apparently now required, so it really shouldn't be a local update causing this.

puddly, 2024-04-04:
ZHA doesn't include any Aqara OTA updates and most built-in OTA providers filter by model and manufacturer to prevent something like this from happening.

Can you enable ZHA debug logging, click the "check for updates" button in `/config/updates`, wait 30s for it to show up, and then disable debug logging to download a log?

FlorianFreudiger, 2024-04-10:
Not sure why but the updates disappeared after upgrading to Home Assistant 2024.4, I did not change anything beyond that.

Will re-open if the updates appear again.

closed 2024-04-10 by FlorianFreudiger