zigpy/zha-device-handlers #3073: [Device Support Request] TS0601 by _TZE200_s1xgth2u for Nedis SmartLife Zigbee Climate Sensor ZBSC30WT

zifban, 2024-03-28:
### Problem description

I bought a Nedis SmartLife Zigbee Climate Sensor ZBSC30WT. HomeAssistant/ZHA shows only LQI and RSSI when paired.

### Solution description

Please add support to this model variant.

### Device signature

<details><summary>Device signature without custom zha quirk</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_s1xgth2u",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details><summary>Device signature with custom zha quirk</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_s1xgth2u",
  "model": "TS0601",
  "class": "ts0601_sensor.TuyaTempHumiditySensorVar03"
}
```

</details>
[... 659 lines cut ...]

<details><summary>Custom quirk</summary>

```diff
  class TuyaTempHumiditySensorVar03(CustomDevice):
      """Tuya temp and humidity sensor (variation 03)."""
  
      signature = {
          # "profile_id": 260,
          # "device_type": "0x0051",
          # "in_clusters": ["0x0000","0x0004","0x0005","0xef00"],
          # "out_clusters": ["0x000a","0x0019"]
          MODELS_INFO: [
              ("_TZE200_yjjdcqsq", "TS0601"),
              ("_TZE200_9yapgbuv", "TS0601"),
              ("_TZE200_qyflbnbj", "TS0601"),
+             ("_TZE200_s1xgth2u", "TS0601"),  #  Nedis ZBSC30WT
          ],
```

</details>

### Additional information

Link to product: https://nedis.com/en-us/product/household-and-living/climate/humidity-control/550783946/smartlife-climate-sensor-zigbee-30-battery-powered-android-ios-white

pullasuti, 2024-04-22:
There's already one in the repository, which is pretty close. Some datapoints are different, but seems to work with the Nedis version (and also Connecto with id _TZE200_zppcgbdj).

Noticed that my old extension had stopped working, so tried with the new template. No idea what the extra datapoints do. Can't get the temperature unit to update, no matter what 🤷 (debug data added for interested parties)

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_s1xgth2u',
        },
    ],
    model: 'ZBSC30WT',
    vendor: 'Nedis',
    description: 'Automatically generated definition',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime, 
    configure: tuya.configureMagicPacket,
    exposes: [
        e.temperature(), 
        e.humidity(), 
        tuya.exposes.temperatureUnit(), 
        tuya.exposes.temperatureCalibration(),
        tuya.exposes.humidityCalibration(), 
        tuya.exposes.batteryState(), 
        e.battery_low(), 
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [4, 'battery', tuya.valueConverter.batteryState],  //just "1", though low voltage rechargeable batteries in use. Needs checking with actual batteries :)
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnit], // different datatype from others, this has 4, others 2 (not changing unit, though)
           // [19, 'temperature_unit', tuya.valueConverter.temperatureUnit], // seen only once, ever, original value "4"
            [10, 'temperature_calibration', tuya.valueConverter.divideBy10], // no idea if it's working (original values seem indicative though)
            [11, 'humidity_calibration', tuya.valueConverter.divideBy10], // no idea if it's working (original values seem indicative though)
                ],
    },
};

module.exports = definition;

//debug 2024-04-22 16:53:05Received Zigbee message from 'LCD temp', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,250],"type":"Buffer"},"datatype":2,"dp":10}],"seq":59953}' from endpoint 1 with groupID 0
//debug 2024-04-22 16:53:05Received Zigbee message from 'LCD temp', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,150],"type":"Buffer"},"datatype":2,"dp":11}],"seq":60209}' from endpoint 1 with groupID 0
//debug 2024-04-22 16:53:06Received Zigbee message from 'LCD temp', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,4],"type":"Buffer"},"datatype":2,"dp":19}],"seq":60465}' from endpoint 1 with groupID 0
//debug 2024-04-22 16:53:06Received Zigbee message from 'LCD temp', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":4}],"seq":60721}' from endpoint 1 with groupID 0
//debug 2024-04-22 16:53:06Received Zigbee message from 'LCD temp', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,228],"type":"Buffer"},"datatype":2,"dp":1}],"seq":60977}' from endpoint 1 with groupID 0
//debug 2024-04-22 16:53:06Received Zigbee message from 'LCD temp', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,94],"type":"Buffer"},"datatype":2,"dp":2}],"seq":61233}' from endpoint 1 with groupID 0
//debug 2024-04-22 16:53:07Received Zigbee message from 'LCD temp', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":61489}' from endpoint 1 with groupID 0
```

jouko999, 2024-04-29:
This sensor is widely available, cheap and looking good.
So please add it to supported devices!

Mikkora1, 2024-10-11:
> This sensor is widely available, cheap and looking good. So please add it to supported devices!

jouko999, while this issue is still open, I managed to add Nedis ZBSC30WT with following steps: (I made instructions for non coding person)

Before starting, I strongly recommend to create backup of your current Home Assistant.

1. Add following lines to "configuration.yaml" to the end of file
[image]
 
 2. Create folder "zha_quirks" to same folder where your configuration.yaml locates. 
[image]

3. Download this: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py and modify it:
- Line 102 modify: `RH_MULTIPLIER = 100 #(originally 10)`
- Line 113 add: ` ("_TZE200_s1xgth2u", "TS0601")  #  Nedis ZBSC30WT `
[image]

4. Remove device from ZHA if device is still connected. 

5. Home assistant developer tools -> yaml -> check configurations. If ok, restart HA fully.

6. Search device again via ZHA
 
7. Enjoy
[image]

bangert, 2024-12-24:
@Mikkora1 the suggested quirk is no longer applicable since `ts0601_sensor.py has changed fundamentally.

a more recent device signature

```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_s1xgth2u",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

PR #3634 (2024-12-26): Add Nedis ZBSC30WT sensor `_TZE200_s1xgth2u` variant
## Proposed change
Add support for new device.

## Additional information
I have the device and added the same change as a custom quirk.
Requesting help wrt unit tests.
Fixes #3073 

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2024-12-26 by TheJulianJES