zigpy/zha-device-handlers #3081: [Device Support Request] For Tuya curtain motor

hami89, 2024-04-01:
### Problem description

My Tuya curtain motor is not recognized by HA.

### Solution description

Please add the device (manufacturer) to the supported list.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_yia0p3tr",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.3.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
[... 325 lines cut ...]
    }
  }
}```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

javicalle, 2024-04-02:
According to Z2M it must be added to the `TuyaMoesCover0601` quirk:
* https://github.com/zigpy/zha-device-handlers/blob/9ccb69ad64da9bcbaae28201cc9efa6589565300/zhaquirks/tuya/ts0601_cover.py#L351

Are you able to check it?

hami89, 2024-04-02:
Even before I found that post, I tried to make a custom quirk, but there are some problems.

He is right that the  TuyaMoesCover0601 is the closest one to it, but my device has a secondary endpoint (242).

Simply adding it to the TuyaMoesCover0601 the integration logs an error upon startup, the it is considering my custom version of the quirk, but there is a signature mismatch, as TuyaMoesCover0601 does not define endpoint 242.

So after looking around in the repos, I tried to find the proper definition for 242.

I ended up adding an extra import to the quirk:
```python
from zigpy.zcl.clusters.general import GreenPowerProxy
```

and then I added this modified version:

```python
class TuyaHTCover0601(TuyaWindowCover):
    """Tuya blind controller device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098,
        #                    maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        #                    maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
        # "endpoints": {
        # "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004","0x0005","0xef00"], "out_clusters": ["0x000a","0x0019"] }
        # },
        # "manufacturer": "_TZE200_zah67ekd",
        # "model": "TS0601",
        # "class": "zigpy.device.Device"
        # }
        MODELS_INFO: [
			("_TZE200_yia0p3tr", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
			242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
		}
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
			242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
[... 26 lines cut ...]
2024-04-01 09:32:30.146 DEBUG (MainThread) [zigpy.zcl] [0xBBD8:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x18>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=3, command_id=11, *direction=<Direction.Server_to_Client: 1>)
2024-04-01 09:32:30.146 DEBUG (MainThread) [zigpy.zcl] [0xBBD8:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2024-04-01 09:32:30.147 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140206046492864] Failed to close cover: <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
response = await hass.services.async_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2319, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2356, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 905, in entity_service_call
single_response = await _handle_entity_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in _handle_entity_call
result = await task
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 289, in async_close_cover
raise HomeAssistantError(f"Failed to close cover: {res[1]}")
homeassistant.exceptions.HomeAssistantError: Failed to close cover: <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>
```

So I probably messed up something along the way.

whatisthisUT, 2024-04-10:
@hami89 did you manage to get the _TZE200_yia0p3tr motor running on ZHA?

hami89, 2024-04-16:
> @hami89 did you manage to get the _TZE200_yia0p3tr motor running on ZHA?

Unfortunately no, it still does not work.

hami89, 2024-04-17:
One interesting addition is that while I cannot control the motor with my custom quirk, if I operate it with it's 433Mhz remote, it reports the position correctly through zigbee to HA.

hami89, 2024-07-08:
Can you provide any help to figure out what can be the problem with this motor integration? As it works for zigbee2mqtt it probably should work for mine too.
I can extract any logs needed and do some testing, but I need guidance.

hami89, 2024-08-15:
@whatisthisUT I managed to make it work, but I don't know how to do it the correct way so I can create a PR from it.

Besides the quirk, I figured that the server commands for this particular model should go out as "is_manufacturer_specific = False", otherwise the device rejects them. Unfortunately, for tuya it si set to True on the module init script.

For testing, I modified the /usr/local/lib/python3.12/site-packages/zha/application/platforms/cover/__init__.py init script file from a root shell in HA. After restart, it works as expected.

Can this be achieved from the quirk itself? I got lost in the class hierarchy of zha/zigpy. I would appreciate some guidance.

whatisthisUT, 2024-08-27:
Hey @hami89 awesome that you managed to make it work. I will try to replicated that on my instance this week if I find time.
I'm not involved in development for IOT so I can't be a lot of help to you except for testing.

closed 2025-08-13 by github-actions[bot]