zigpy/zha-device-handlers #3083: MiBoxer PIR1-ZB Motion Detektor TS0202

SkyDroVV, 2024-04-01:
### Problem description

[image]

### Solution description

Generally support.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

SkyDroVV, 2024-04-05:
I sent the motion detector back because of the poor connection and the long response time.

jekader, 2024-06-05:
I have this device and it works just fine for me. It shows extra controls which are not in use but otherwise all good

closed 2024-12-09 by github-actions[bot]