zigpy/zha-device-handlers #3086: [BUG] Warning about "Min mireds is 0, setting to 153"

EPinci, 2024-04-03:
### Bug description

Reporting bug per HA log:

```
2024-04-04 00:12:15.724 WARNING (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x766B:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected
2024-04-04 00:12:15.731 WARNING (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x766B:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected
```

### Steps to reproduce

Appears in HA logs

### Expected behavior

Not sure what to report in here, I created the bug as instructed by the log entry.
How do I trace it back to the offending device?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2024-04-04:
Pleas provide device signature and / or diagnostic.

This is not one HA error, its one not well done device firmware (very likely not Zigbee certified device) that is reporting its color propitiates outside the allowed range so shall being reported to the device manufacture so they can fixing ther device firmware.

EPinci, 2024-04-07:
Those lines on the log are all I have. Can you tell me how to trace them back to a device?

TheJulianJES, 2024-04-07:
Search `0x766B` on the ZHA network visualization map to find the device that's in a broken state.
Then, go to the device page of that and download diagnostics information and upload that info.
(It might be a device such as a ZBDongle-E router.)

EPinci, 2024-04-07:
You are spot on, it's a ZBDongle-E router. Info attached.
[zha-3630fb3d98a51a5a4bfab9d7b1d2741a-SONOFF DONGLE-E_R-a33f3faaf7988ab11c31c5d4082cb77d.json](https://github.com/zigpy/zha-device-handlers/files/14899033/zha-3630fb3d98a51a5a4bfab9d7b1d2741a-SONOFF.DONGLE-E_R-a33f3faaf7988ab11c31c5d4082cb77d.json)

AlexW75, 2024-04-15:
Have the same issue also using Sonoff ZBDongle_E.

gdrapp, 2024-05-26:
Seeing the same with my Sonoff ZBDongle-E running router firmware Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl.

StingRayZA, 2024-07-06:
In case it helps, Similar issue here with the same. here's a `json`
[zha-ed8218d35d87057379ed64de725e603b-SONOFF DONGLE-E_R-859c4730571388a83952e6347d3f7554.json](https://github.com/user-attachments/files/16115772/zha-ed8218d35d87057379ed64de725e603b-SONOFF.DONGLE-E_R-859c4730571388a83952e6347d3f7554.json)

brackishwaters, 2024-08-31:
Same issue here in logbook.  
Logger: zha.zigbee.cluster_handlers
Source: components/zha/__init__.py:151
First occurred: 6:56:57 AM (2 occurrences)
Last logged: 6:56:57 AM

[0xE1CA:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

Im starting to feel like I need a quirk for this due to how ZHA is seeing its configuration:
[image]

semaf, 2024-09-14:
Same issue here..