zigpy/zha-device-handlers #3092: Tuya device TS0601 by _TZE200_2gtsuokt

smittong, 2024-04-08:
### Problem description

When connecting with ZHA to HA, there are no entities available. I have searched through the current quirks and can't find a similar product.  This is an LED constant current driver with outputs for warm and cool whites. The expected controls would be color temperature, diming and on/off control.

### Solution description

Some help please to create a quirk for the device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_2gtsuokt",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.4.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Pacific/Auckland",
    "os_name": "Linux",
    "os_version": "6.1.73-haos-raspi",
    "supervisor": "2024.03.1",
    "host_os": "Home Assistant OS 12.1",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "plant": {
      "version": "2024.2.0",
      "requirements": [
        "async-timeout>=4.0.2"
      ]
    },
    "unifigateway": {
[... 959 lines cut ...]
              access_acl: 5
              attribute_value: 4294967295
            "0x0006":
              attribute_id: "0x0006"
              attribute_name: image_upgrade_status
              value_type:
                - "0x30"
                - enum8
                - Discrete
              access: READ|REPORT
              access_acl: 5
              attribute_value: 0
            "0xfffd":
              attribute_id: "0xfffd"
              attribute_name: cluster_revision
              value_type:
                - "0x21"
                - uint16_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 3
          commands_received: {}
          commands_generated: {}
success: true

cepresso, 2024-10-14:
Hi, I have the following quirk which allows to switch the driver on and off and also choose the intensity. However, the color temperature changes are not working. Maybe someone could take a look and share some advice?
```python
"""Tuya dimmable led controller dual channel."""
from typing import Dict
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    PowerOnState,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaLevelControl
)

class CCTColorCluster(CustomCluster, Color):
    """CCT Lighting custom cluster."""

    # Remove RGB color wheel for CCT Lighting: only expose color temperature
    _CONSTANT_ATTRIBUTES = {0x400A: 16}

class CCTControllerManufCluster(TuyaMCUCluster):

    attributes = TuyaMCUCluster.attributes.copy()

    attributes.update(
        {
            0x8002: ("power_on_state", PowerOnState),
        }
    )
    
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        3: DPToAttributeMapping(
            TuyaLevelControl.ep_attribute,
            "current_level",
            converter=lambda x: (x * 255) // 1000,
            dp_converter=lambda x: (x * 1000) // 255,
        ),
        4: DPToAttributeMapping(
            CCTColorCluster.ep_attribute,
            "color_temperature",
            converter=lambda x: int((x - 250) * 1000 / (454 - 250)),
            dp_converter=lambda x: int(250 + x * (454 - 250) / 1000),
        ),
        14: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "power_on_state",
            converter=lambda x: PowerOnState(x),
        ),
    }
    
    set_time_offset = 1970

    data_point_handlers = {
        1: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        14: "_dp_2_attr_update"
    }

class DimmableLedCCTController(CustomDevice):
    """Tuya dimmable led controller dual channel."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_2gtsuokt", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaMCUCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaLevelControl,
                    CCTControllerManufCluster,
                    CCTColorCluster,
                    #LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

smittong, 2024-12-30:
Hi @cepresso, thank you for that.  It has been a while before I got back to my project using this.  I just re-connected that CCA driver and can confirm it works how you describe.  Not sure how to solve the light temperature.  Both channels appear to operate as one.  Did you make any further progress?

closed 2025-08-13 by github-actions[bot]