zigpy/zha-device-handlers #3093: [BUG] TS110E by _TZ3210_3mpwqzuu | 2 gang dimmer

miit86, 2024-04-08:
### Bug description

Using ZHA.

Problems that occur.
1. Device (automatically) turns on light but not the state. 
    Reproduced when I Read the 0x0006 (cluster) and the 0x0000 (atrribute) 
    when i change the value to "false" the "automatically on" problem is gone.
2. Device (automatically) turns off the state but not the lights.
3. When turning on with brightness from automation the state is good, but the lights stay off. when changing the brightness with the slider the lights go on.
4. gui state is sometimes not correct with real state

### Steps to reproduce

by just using it.

### Expected behavior

1. gui state and real state should be the same
2. when using turn_on with brightness. it should turn on with the brightness

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
[... 725 lines cut ...]
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

elimo2401, 2024-04-16:
Any update here?

iroger, 2024-10-19:
I have the same issue with TS110E
by _TZ3210_3mpwqzuu

GUI state changes to ON while light stays off

Henrik1986, 2024-11-26:
My device switch randomly to "on" in the GUI . The bulb itself dosen't turns on. If the bulb is "on" in the GUI I can turn on the bulb on the wall-switch. It's just one of the channels that has this behavior. The other channels is off/on in the GUI when it should show as off/on.  

The logg shows this. 

```
2024-11-26 21:52:07.621 DEBUG (MainThread) [zha.zigbee.device] [0x7DC5](TS110E): Device seen - marking the device available and resetting counter
2024-11-26 21:52:07.621 DEBUG (MainThread) [zha.zigbee.device] [0x7DC5](TS110E): Update device availability -  device available: True - new availability: True - changed: False
```

Henrik1986, 2024-11-28:
I found a "solution" to my problem on home Assistant community. 

https://community.home-assistant.io/t/light-ts110e-zigbee-device-status-is-changing-while-device-stays-off/702420/3?u=henrik1986

closed 2025-08-13 by github-actions[bot]