zigpy/zha-device-handlers #3094: Support fot NOUS a4z   (_TZ3000_uwkja6z1)

Jassir-ben-Hussein, 2024-04-08:
### Problem description

With the default ZHA configuration the Device delivers the wrong values for Current and Energy deliverd for both ports.

The Factor seems to be 3600

### Solution description

The Values should be corrected by the factor value probably

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_uwkja6z1",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

easter-blok, 2024-06-11:
I have the same device, if you need more info or diagnostic logs let us know! Thanks a lot!

Sup3rMario, 2024-06-16:
I have the same problems - Can someone help?

Gounick, 2024-10-09:
Same problems, same device. Here's my 2 cents:

### Problem description

This connected socket has 2 electrical outlets, and its sensors normally measure values independently of each other:
* current
* voltage
* power
* energy

Currently, in my case, I have kept one switch off and the other permanently on.
As far as I can see, the sensors can't differentiate between socket n°1 and n°2. I have the impression that the values are the sum of the 2 sockets.
The electric current seems to be on the wrong scale.

### Solution description

Obtain values per electrical outlet.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]
</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0003",
[... 3064 lines cut ...]
                "attribute": "ZCLAttributeDef(id=0x0A0B, name='active_power_ph_c', type=<class 'zigpy.types.basic.int16s'>, zcl_type=<DataTypeId.int16: 41>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x050f": {
                "attribute": "ZCLAttributeDef(id=0x050F, name='apparent_power', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x090f": {
                "attribute": "ZCLAttributeDef(id=0x090F, name='apparent_power_ph_b', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0a0f": {
                "attribute": "ZCLAttributeDef(id=0x0A0F, name='apparent_power_ph_c', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0805": {
                "attribute": "ZCLAttributeDef(id=0x0805, name='average_rms_over_voltage', type=<class 'zigpy.types.basic.int16s'>, zcl_type=<DataTypeId.int16: 41>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0512": {
                "attribute": "ZCLAttributeDef(id=0x0512, name='average_rms_over_voltage_counter', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0912": {
                "attribute": "ZCLAttributeDef(id=0x0912, name='average_rms_over_voltage_counter_ph_b', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.u

neoscaler, 2025-03-04:
I created quirk for this device today, see https://github.com/neoscaler/zha-device-handlers/blob/feature/nous-a4z-quirk/zhaquirks/nous/ts011f_2plug.py

Going to add tests and then create a PR.

neoscaler, 2025-03-07:
PR is up, rewrote the quirk with v2 syntax and added all other optimizations from Z2M definition.

Jassir-ben-Hussein, 2025-12-23:
The wrong energy values are still there .