zigpy/zha-device-handlers #3095: [Device Support Request] Nimly doorlock (Touch, Touch Pro etc)

Fredrik81, 2024-04-08:
### Problem description

Add support for Nimly doorlocks.
According to support this should be possible from ZHA/z2m.
- Add remove user
- Open/Lock
- Change sound volume
- Activate autolock
- Se who unlocked and what method (code, finger, nfc etc)

In ZHA today only lock/unlock works. z2m seems to have more and user report they can do all except see who unlocked today.
Would be nice to get more features and at least battery level.

Here is link to z2m support for Nimly: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/onesti.ts

### Solution description

At least be able to see battery status but support for all the listed features would be great!

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.MainsPowered|RxOnWhenIdle|AllocateAddress: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": "0x0104",
      "device_type": "0x000a",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0101",
        "0xfea2"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Onesti Products AS",
  "model": "NimlyPRO",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

wealthystudent, 2024-09-04:
Yes, this is needed!

Is there any way to issue Zigbee commands for the features that are not supported by ZHA?

gustavostmo, 2024-11-20:
This would be fantastic if someone could fix it. Unfortunately I don't have the ability today but would like to learn.

misse, 2025-03-26:
I am also very interested and would be happy to contribute, but I'm not sure how to get started.

misse, 2025-05-11:
[image] Baby steps. Setting this value to 0 prevents the door from auto locking

Clusters
Endpoint id: 11
Id: 0x0101
Type: In

Attribute
auto_relock_time (id 0x0023)
Value: 0 = auto-relock disabled

Now to figure out how to make this available to Home assistant. 

Relevant code over at zigbee2mqtt: https://github.com/Koenkk/zigbee-herdsman-converters/blob/a7f23169d662b478d8cdab9e9ae530051b8b98c2/src/converters/toZigbee.ts#L2804

misse, 2025-05-11:
Turns out it was easier than I thought. Working code running in my custom_zha_quirks folder:

```python
"""Device handler for Nimly Smart Locks."""

from zigpy.quirks.v2 import QuirkBuilder
from zigpy.zcl.clusters.closures import DoorLock
from zigpy.zdo.types import NodeDescriptor

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.nimly import NIMLY

# clears the mains powered mac capability flag
NIMLY_LOCK_NODE_DESCRIPTOR = NodeDescriptor(
    logical_type=2,
    complex_descriptor_available=0,
    user_descriptor_available=0,
    reserved=0,
    aps_flags=0,
    frequency_band=8,
    manufacturer_code=4660,
    maximum_buffer_size=108,
    maximum_incoming_transfer_size=127,
    server_mask=11264,
    maximum_outgoing_transfer_size=127,
    descriptor_capability_field=0,
    mac_capability_flags=NodeDescriptor.MACCapabilityFlags.AllocateAddress
    | NodeDescriptor.MACCapabilityFlags.RxOnWhenIdle,
)

(
    QuirkBuilder(NIMLY, "EasyFingerTouch")
    .applies_to(NIMLY, "EasyCodeTouch")
    .applies_to(NIMLY, "easyCodeTouch_v1")
    .node_descriptor(NIMLY_LOCK_NODE_DESCRIPTOR)
    .add_to_registry()
)

(
    QuirkBuilder(NIMLY, "NimlyPRO")
    .applies_to(NIMLY, "NimlyCode")
    .applies_to(NIMLY, "NimlyTouch")
    .applies_to(NIMLY, "NimlyIn")
    .node_descriptor(NIMLY_LOCK_NODE_DESCRIPTOR)
    .replaces(DoublingPowerConfigurationCluster, endpoint_id=11)
    .switch(
        attribute_name="auto_relock_time",  # Expose the attribute as a switch
        cluster_id=DoorLock.cluster_id,  # Use the cluster ID from DoorLock
        endpoint_id=11,  # Specify the endpoint ID
        translation_key="auto_relock",  # Translation key for Home Assistant
        fallback_name="Automatically lock after unlock",  # Fallback name
    )
    .add_to_registry()
)
````

Mordog89, 2025-05-19:
I would appreciate some development on this as well. I can control some functions in the "manage sig we device" view as someone else also commented, but full support would be great, specifically to also add things like week schedule for adding a temporary code for handyman to use for example...

comatose-tortoise, 2025-05-30:
Also super interested in this. I've been trying to do some quirk development with LLMs but can't get it to work so far. Anyone else trying?

wealthystudent, 2025-05-30:
> Also super interested in this. I've been trying to do some quirk development with LLMs but can't get it to work so far. Anyone else trying?

Sadly I bought an SLZB-06 and switched to Z2M, so I stoped looking into this..

comatose-tortoise, 2025-05-30:
Too bad. Any reflections on Z2M vs ZHA? Which is better?

ABEC7, 2025-05-30:
> Too bad. Any reflections on Z2M vs ZHA? Which is better?

Z2M. It works great.

wealthystudent, 2025-05-30:
> Too bad. Any reflections on Z2M vs ZHA? Which is better?

Z2M is best for compatibility imo

comatose-tortoise, 2025-05-30:
So can I ask you guys that now use this lock with Z2M: What features are available for it there?

misse, 2025-05-30:
I found that the most features seems to be available to zigpy with the existing code, the only thing I did was add exposure of an endpoint to home assistant (see .switch).

So I think It's pretty easy to add more as long as the Doorlock.cluster already have a mapping for it. 

What would you like to be able to control or read?

PR #4138 (2025-06-24): Expose extra entities for Nimly Locks
## Proposed change

- Last action source
- Last action
- Last action user
- Last pin code
- Auto relock
- Sound volume

[image]

## Additional information

Fixes #3095
Fixes #2354

Related to #3465, #3580, #3457

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-06-24 by TheJulianJES