zigpy/zha-device-handlers #3097: [Device Support Request] Tuya TS0201 by _TZ3210_ncw88jfq quirk request for 2 issues (humidity reporting and temperature units display)

jeremyzone, 2024-04-08:
### Problem description

I have purchased this small Zigbee temperature/humidity sensor with screen from Aliexpress https://a.aliexpress.com/_mL7xBbs
[image]
It has a perfect form factor for my gecko tank. Unfortunately there are two issues that I have found with using these devices on ZHA

1. Humidity reporting does not display properly in Home Assistant, but only when the humidity is above 65.5%. When the humidity is below 65.5%, it works perfectly. See the following graph as the humidity rises at night in my gecko tank. Once the humidity rises above 65.5%, the humidity drops to a truncated value of 0 then continues to rise even though the display on the device shows the correct, higher value. 
[image]
Someone else reported the issue here but didn't understand why it was happening https://community.home-assistant.io/t/ts0201-by-tz3210-ncw88jfq-reports-wrong-humidity-still/654814

2. The screen defaults to celsius but supports switching to Fahrenheit if sent a Zigbee command. This is not currently supported in this device in ZHA. I was able to successfully switch them to Fahrenheit when connected to a Zuya Zigbee hub, but they switch back to Celsius when they are reset to connect to ZHA. 

### Solution description

For the problems outline above:
1. I would like humidity to be reported accurately in ZHA. I have done some debugging of this issue and I have figured out how to convert the humidity to the correct values. This sensor has a weird reporting scheme:
For humidity values below 65.5%, the device sends out the humidity multiplied by 100. For example, 11.1% humidity is 1110, and 62.6% humidity is 6260. The last digit is **always** a 0. That's easy enough, HA reports these values fine.
HOWEVER for humidity values above 65.5%, the device CHANGES its reporting scheme to a new one where the last digit is **always** a 6 and the remaining digits start over at 0. So for example, if the actual humidity is 67.6, then the device is sending a reported humidity value of 216. If the actual humidity is 95.3%, the device is sending a reported humidity value of 2986.
2. I would like the ability to switch the device display to Fahrenheit in ZHA. I'm not sure if this is helpful for ZHA, but I followed the directions for Zigbee2MQTT to find the Tuya DP (https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_5-link-your-tuya-account-with-your-new-tuya-project) and the datapoint was 101. 
[image]

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_ncw88jfq",
  "model": "TS0201",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.4.1",
[... 346 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

honza73, 2024-04-15:
Just a proposal of custom quirk. Solves only the humidity issue.
Based on [zha-device-handlers]/zhaquirks/tuya/ts0201.py (MoesTemperatureHumidtySensorWithScreen).

_Note: What about temperatures higher that 65.5 °C/°F?_

"""https://github.com/zigpy/zha-device-handlers/issues/3097 devices support."""

```
import logging

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufCluster,
)
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    Ota,
    Time,
    PowerConfiguration, 
)
from zigpy.zcl.clusters.measurement import (
    RelativeHumidity,
    TemperatureMeasurement,
)

_LOGGER = logging.getLogger(__name__)

class RelativeHumidityFolded(CustomCluster, RelativeHumidity):
    """Handles invalid humidity values."""

    def _update_attribute(self, attrid, value):
        if attrid == 0:
            """See https://github.com/zigpy/zha-device-handlers/issues/3097"""

            _LOGGER.debug(
                "[0x%04x:%s:0x%04x] Received value of %d: %d",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
                value,
            )
            
            if value % 10 == 6:
                value = 10 * (value // 10)
                value += 6550

            _LOGGER.debug(
                "[0x%04x:%s:0x%04x] Saved value of %d: %d",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
                value,
            )
            
        super()._update_attribute(attrid, value)

class Aliexpress(CustomDevice):
    """Aliexpress temperature and humidity sensor with screen."""

    signature = {
        #  <SimpleDescriptor endpoint=1, profile=260, device_type="0x0302"
        #  input_clusters=["0x0000", "0x0001", "0x0402", "0x0405", "0xef00"]
        #  output_clusters=["0x000a", "0x0019"]>
        MODELS_INFO: [
            ("_TZ3210_ncw88jfq", "TS0201"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id, 
                    Ota.cluster_id
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidityFolded,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id, 
                    Ota.cluster_id
                ],
            }
        },
    }

```

jeremyzone, 2024-04-16:
> Just a proposal of custom quirk. Solves only the humidity issue. Based on [zha-device-handlers]/zhaquirks/tuya/ts0201.py (MoesTemperatureHumidtySensorWithScreen).
> 
> _Note: What about temperatures higher than 65.5 °C/°F?_

Thanks so much! The humidity reporting does seem to be working now with this custom quirk! This is awesome. I will continue to test the humidity reporting 

To answer your question, I have seen no errors with temperature values so far

honza73, 2024-04-17:
> OK I have figured out how to switch the display from Celsius to Fahrenheit! But I don't know how to set this up using the GUI "Configuration" menu on the device settings. The only way I have been able to figure it out is by sending a command through the Manage Zigbee device menu
> 
> I am not a programmer! Just like playing with these devices
> 
Well... I am in similar situation as you. I am programmer, but not of the Python code. I own this sensor, so I coded only what I need. Also by trial and error. I don't need to switch the display to Fahrenheit. So I didn't solve this. :-) But as I see, you have shifted the solution by another step. In fact you are on a half way. You only need to map the Tuya cluster to such a control cluster what is understood by the HA core. I made something simillar for my TRV (thermostatic radiator valve). And I was inspired by https://github.com/jacekk015/zha_quirks.

zumwild, 2024-07-17:
I had the problem with the Humidity values in ZHA (The current HA, 12.4) being off honza73's quirk above fixed it. Thanks honza.

mag2352, 2024-08-14:
I converted it to Quirk V2, and added an enumerator for the temperature display units. Also, for my specific unit, it reports the humidity with three digits, unlike the issue originally reported by @honza73. In my case, I simply needed to multiply the received value by 10 (it was reporting 926 for 92.6%, rather than 9260). Maybe there was a firmware update (I bought this unit this month)? However, I left the original fix in place, as it seems other users need it to have values of over 65.5 % show correctly.

`
"""https://github.com/zigpy/zha-device-handlers/issues/3097 devices support."""

import logging
from typing import Dict, Final, Any, Optional, Union
from zigpy.quirks.v2 import QuirkBuilder
import zigpy.types as t
from zigpy.zcl.clusters.measurement import RelativeHumidity
from zigpy.zcl.foundation import ZCLAttributeDef
from zigpy.quirks import CustomCluster
from zigpy.zcl import foundation
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA, 
)

_LOGGER = logging.getLogger(__name__)

class TempDisplayUnits(t.enum8):
    """Temperature Display units."""
    CELSIUS = 0x00
    FAHRENHEIT = 0x01

class TuyaTempSwitcher(TuyaMCUCluster):
    """Tuya Temperature Display units switcher."""

    class AttributeDefs(TuyaMCUCluster.AttributeDefs):
        """Manufacturer specific attribute definitions."""
        temperature_display_units: Final = ZCLAttributeDef(
            id=0x0065,  # DP 101
            type=TempDisplayUnits,
            is_manufacturer_specific=True,
        )

    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
        **kwargs: Any,
    ):
        """Override the default Cluster command."""

        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "temperature_display_units",
            converter=lambda x: TempDisplayUnits(x),
        ),
    }

    data_point_handlers = {
        101: "_dp_2_attr_update",
    }

class RelativeHumidityFolded(CustomCluster, RelativeHumidity):
    """Handles invalid humidity values."""

    def _update_attribute(self, attrid, value):
        if attrid == 0:
            _LOGGER.debug(
                "[0x%04x:%s:0x%04x] Received value of %d: %d",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
                value,
            )
            
            if value > 100 and value < 1000:
                value *= 10

            if value % 10 == 6:
                value = 10 * (value // 10)
                value += 6550

            _LOGGER.debug(
                "[0x%04x:%s:0x%04x] Saved value of %d: %d",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
                value,
            )
            
        super()._update_attribute(attrid, value)

(
    QuirkBuilder("_TZ3210_ncw88jfq", "TS0201")
    .skip_configuration()
    .adds(TuyaTempSwitcher)
    .replaces(RelativeHumidityFolded)
    .enum(TuyaTempSwitcher.AttributeDefs.temperature_display_units.name, TempDisplayUnits, TuyaTempSwitcher.cluster_id)
    .add_to_registry()
)
`

jeremyzone, 2024-11-07:
> I converted it to Quirk V2, and added an enumerator for the temperature display units. Also, for my specific unit, it reports the humidity with three digits, unlike the issue originally reported by [...]

This sounds great! Thanks for working on this. I can't figure out how to test it because I haven't messed with the Quirk V2 before. I tried to put the code into a .py file in my zha_quirks directory but it didn't work. I have spent the night trying to figure out how to use Quirk V2 via the documentation but still haven't figured it out...

That is interesting that yours doesn't need the humidity correction. My devices report firmware 0x00000083

mag2352, 2024-11-07:
> > I converted it to Quirk V2, and added an enumerator for the temperature display units. Also, for my specific unit, it reports the humidity with three digits, unlike the issue originally reported by @honza73. In my case, I simply needed to multiply the received value by 10 (it was reporting 926 for 92.6%, rather than 9260). Maybe there was a firmware update (I bought this unit this month)? However, I left the original fix in place, as it seems other users need it to have values of over 65.5 % show correctly.
> 
> 
> 
> This sounds great! Thanks for working on this @mag2352 . 
> 
> 
> 
> That is interesting that yours doesn't need the humidity correction. My devices report firmware 0x00000083
> 
> 
> 
> Ignore the previous edit: I was able to get your quirk to work once I made a small change to line 107. I think that this is a new thing from 2 weeks ago in the latest code. I had to add a translation_key and fallback_name. I have no idea if I'm doing it right per the proper coding protocol, but this is what I changed it to and it is at least loading:
> 
> `    .enum(TuyaTempSwitcher.AttributeDefs.temperature_display_units.name, TempDisplayUnits, TuyaTempSwitcher.cluster_id, translation_key="TS0201Temp", fallback_name="TS0201temp")`
> 
> 

I'm glad you were able to get it to load, does the switcher work? I know quirks V2 has significant activity, so syntax might change for it to work. As far as I can tell, the one I posted above still works for me, I can see the custom quirk is loaded in HA logs and the device signature. In order to get it to switch, I usually press the pair button on the device quickly to force it to send an update to HA, then switch the unit in the enumerator (sometimes I have to press the pair button one or two more times after switching the unit for it to register). Also, how can I determine the firmware version of the device? I apologize if that's a trivial question.

jeremyzone, 2024-11-10:
> > > I converted it to Quirk V2, and added an enumerator for the temperature display units. Also, for my specific unit, it reports the humidity with three digits, unlike the issue originally reported by @honza73. In my case, I simply needed to multiply the received value by 10 (it was reporting 926 for 92.6%, rather than 9260). Maybe there was a firmware update (I bought this unit this month)? However, I left the original fix in place, as it seems other users need it to have values of over 65.5 % show correctly.
> > 
> > 
> > This sounds great! Thanks for working on this @mag2352 .
> > That is interesting that yours doesn't need the humidity correction. My devices report firmware 0x00000083
> > Ignore the previous edit: I was able to get your quirk to work once I made a small change to line 107. I think that this is a new thing from 2 weeks ago in the latest code. I had to add a translation_key and fallback_name. I have no idea if I'm doing it right per the proper coding protocol, but this is what I changed it to and it is at least loading:
> > `    .enum(TuyaTempSwitcher.AttributeDefs.temperature_display_units.name, TempDisplayUnits, TuyaTempSwitcher.cluster_id, translation_key="TS0201Temp", fallback_name="TS0201temp")`
> 
> I'm glad you were able to get it to load, does the switcher work? I know quirks V2 has significant development activity, so syntax might change for it to work. As far as I can tell, the one I posted above still works for me, I can see the custom quirk is loaded in HA logs and the device signature. In order to get it to switch, I usually press the pair button on the device quickly to force it to send an update to HA, then switch the unit in the enumerator (sometimes I have to press the pair button one or two more times after switching the unit for it to register). Also, how can I determine the firmware version of the device? I apologize if that's a trivial question.

I was having trouble adding a test device to my Home Assistant using the V2 quirk so I removed the line ".skip_configuration()" and now it seems to be working to add a new device.

I am seeing the same behavior as you with regard to changing the temp units where I have to push the pairing button to wake the device at the same time as I change the temp units on HA, but it is working!

I can see the firmware version by clicking on "Firmware" under the Configuration section on the device info page.

Here is the full V2 quirk that is working for me:
```
"""https://github.com/zigpy/zha-device-handlers/issues/3097 devices support."""

import logging
from typing import Dict, Final, Any, Optional, Union
from zigpy.quirks.v2 import QuirkBuilder
import zigpy.types as t
from zigpy.zcl.clusters.measurement import RelativeHumidity
from zigpy.zcl.foundation import ZCLAttributeDef
from zigpy.quirks import CustomCluster
from zigpy.zcl import foundation
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA, 
)

_LOGGER = logging.getLogger(__name__)

class TempDisplayUnits(t.enum8):
    """Temperature Display units."""
    CELSIUS = 0x00
    FAHRENHEIT = 0x01

class TuyaTempSwitcher(TuyaMCUCluster):
    """Tuya Temperature Display units switcher."""

    class AttributeDefs(TuyaMCUCluster.AttributeDefs):
        """Manufacturer specific attribute definitions."""
        temperature_display_units: Final = ZCLAttributeDef(
            id=0x0065,  # DP 101
            type=TempDisplayUnits,
            is_manufacturer_specific=True,
        )

    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
        **kwargs: Any,
    ):
        """Override the default Cluster command."""

        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "temperature_display_units",
            converter=lambda x: TempDisplayUnits(x),
        ),
    }

    data_point_handlers = {
        101: "_dp_2_attr_update",
    }

class RelativeHumidityFolded(CustomCluster, RelativeHumidity):
    """Handles invalid humidity values."""

    def _update_attribute(self, attrid, value):
        if attrid == 0:
            _LOGGER.debug(
                "[0x%04x:%s:0x%04x] Received value of %d: %d",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
                value,
            )
            
            if value > 100 and value < 1000:
                value *= 10

            if value % 10 == 6:
                value = 10 * (value // 10)
                value += 6550

            _LOGGER.debug(
                "[0x%04x:%s:0x%04x] Saved value of %d: %d",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
                value,
            )
            
        super()._update_attribute(attrid, value)

(
    QuirkBuilder("_TZ3210_ncw88jfq", "TS0201")
    .adds(TuyaTempSwitcher)
    .replaces(RelativeHumidityFolded)
    .enum(TuyaTempSwitcher.AttributeDefs.temperature_display_units.name, TempDisplayUnits, TuyaTempSwitcher.cluster_id, translation_key="TS0201Temp", fallback_name="TS0201temp")
    .add_to_registry()
)
```

mag2352, 2024-11-10:
That makes sense. I was following following an another v2 quirk as an example and it had skip_configuration() in it. It happened to work fine with my device, but I'm reading that it can cause problems. I apologize for that, but I am glad you were able to figure out why the quirk wasn't working. The firmware on my model is the same as yours (83). I had a version of the quirk where I printed all of the humidity values and it never had the rollover effect above 65.5%. Either way, being able to switch the display units has made them into a good solution for per room management. I've been running 5 of these for months now, and they've been working great. I've attached an image of the humidity readings from one of them, and it is able to read above 65.5% humidity without issue. [image]

jeremyzone, 2024-11-11:
> That makes sense. I was following following an another v2 quirk as an example and it had skip_configuration() in it. It happened to work fine with my device, but I'm reading that it can cause problems. I apologize for that, but I am glad you were able to figure out why the quirk wasn't working. The firmware on my model is the same as yours (83). I had a version of the quirk where I outputted all of the humidity values to the HA log, and it never had the rollover effect above 65.5%. Either way, being able to switch the display units has made them into a good solution for per room management. I've been running 5 of these for months now, and they've been working great. I've attached an image of the humidity readings from one of them, and it is able to read above 65.5% humidity without issue.

It sounds like we are getting close. But one question: You are using a different humidity reporting code in your custom quirk that's different from the quirk I've poster, right?

mag2352, 2024-11-11:
> > That makes sense. I was following following an another v2 quirk as an example and it had skip_configuration() in it. It happened to work fine with my device, but I'm reading that it can cause problems. I apologize for that, but I am glad you were able to figure out why the quirk wasn't working. The firmware on my model is the same as yours (83). I had a version of the quirk where I outputted all of the humidity values to the HA log, and it never had the rollover effect above 65.5%. Either way, being able to switch the display units has made them into a good solution for per room management. I've been running 5 of these for months now, and they've been working great. I've attached an image of the humidity readings from one of them, and it is able to read above 65.5% humidity without issue.
> 
> 
> 
> It sounds like we are getting close. But one question: You are using a humidity reporting code in your custom quirk that's different from the quirk I'm using, right?

No, I'm actually using the same one you posted. I added the change needed for my version of the sensor, which does not conflict with the calculation done for the one you use. This is what was necessary for my device to work:

```

if value > 100 and value < 1000:
                value *= 10

```

You can see that the line is present in the code you posted (and mine), but isn't present in the v1 quirk originally posted.

honza73, 2024-12-05:
> I was able to implement @honza73 solution, thanks for that man. Just a question out of curiosity: i see the point of putting an offset of 6550 to the reading as i exactly saw the the difference before the quirk (i.e. seeing 80% on the display and ha recording 15%). How comes %10==6 if readings are 4 digits? What raw readings did you originally see above 65.50% and below 65.50%? Thanks a lot again

Well, I understand, that you have the same unit as I have. Unit which is breaking the reading on the 65.5% limit. So bellow the 65.5% the raw reading for example for 51.2% is 5120 and the raw reading for example for 72.3% is 686.

tricoveri, 2024-12-05:
Thanks man! My problem was that i don't know how to access the raw reading. Indeed we have the same unit. Now i see, it defaults above 65.5%. And the last digit over 65,5% is 6 regardless of the reading itself maybe.
If actual humidity should fall below 35% it could be misleanding unless:

32%= 3200
97,5%=3206

honza73, 2024-12-05:
> Thanks man! My problem was that i don't know how to access the raw reading. Indeed we have the same unit. Now i see, it defaults above 65.5%. And the last digit over 65,5% is 6 regardless of the [...]

Exactly!

The raw and translated values are both logged to log (by our custom quirk). You only have to enable to log debug messages:

```
logger:
  default: info
  logs:
#    homeassistant.components.zha: debug
#    zigpy: debug
#    zhaquirks: debug
    ts0201_aliexpress: debug
```

jmuf, 2024-12-30:
just to say that the @jeremyzone v2 quirk (in https://github.com/zigpy/zha-device-handlers/issues/3097#issuecomment-2466591691 ) may no longer work for the "old" units (i.e with rollover bug, firmware ≠ 0x83):

[image]

(the obviously-wrong "staircase" values in the middle all end in "6": 16.6%, 28.6%, 41.6% etc)

This could be because values for the respective workarounds do overlap, i.e reported "166" with the old unit should have become 67.1%, with new firmware would be 16.6% (and as the new check comes first (`value >100`), this is what "wins"..).

Per @mag2352 (in https://github.com/zigpy/zha-device-handlers/issues/3097#issuecomment-2467032288), the new firmware units can indeed report legitimate values ending in "6", so we cannot easily distinguish both cases.
Unless the quirk can itself check on the firmware version and change behaviour , we might need two separate "quirks"?

Dluc41, 2024-12-30:
> Just a proposal of custom quirk. Solves only the humidity issue. Based on [zha-device-handlers]/zhaquirks/tuya/ts0201.py (MoesTemperatureHumidtySensorWithScreen).
> 
> _Note: What about temperatures higher than 65.5 °C/°F?_
> 
> ```
> """https://github.com/zigpy/zha-device-handlers/issues/3097 devices support."""
> 
> import logging
> 
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.tuya import (
>     TuyaManufCluster,
> )
> from zigpy.quirks import CustomDevice, CustomCluster
> from zigpy.profiles import zha
> from zigpy.zcl.clusters.general import (
>     Basic,
>     Ota,
>     Time,
>     PowerConfiguration, 
> )
> from zigpy.zcl.clusters.measurement import (
>     RelativeHumidity,
>     TemperatureMeasurement,
> )
> 
> _LOGGER = logging.getLogger(__name__)
> 
> 
> 
> class RelativeHumidityFolded(CustomCluster, RelativeHumidity):
>     """Handles invalid humidity values."""
> 
>     def _update_attribute(self, attrid, value):
>         if attrid == 0:
>             """See https://github.com/zigpy/zha-device-handlers/issues/3097"""
> 
>             _LOGGER.debug(
>                 "[0x%04x:%s:0x%04x] Received value of %d: %d",
>                 self.endpoint.device.nwk,
>                 self.endpoint.endpoint_id,
>                 self.cluster_id,
>                 attrid,
>                 value,
>             )
>             
>             if value % 10 == 6:
>                 value = 10 * (value // 10)
>                 value += 6550
> 
>             _LOGGER.debug(
>                 "[0x%04x:%s:0x%04x] Saved value of %d: %d",
>                 self.endpoint.device.nwk,
>                 self.endpoint.endpoint_id,
>                 self.cluster_id,
>                 attrid,
>                 value,
>             )
>             
>         super()._update_attribute(attrid, value)
> 
> 
> class Aliexpress(CustomDevice):
[... 140 lines cut ...]
>                 ],
>             }
>         },
>     }
> 
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     PowerConfiguration.cluster_id,
>                     TemperatureMeasurement.cluster_id,
>                     RelativeHumidityFolded,
>                     TuyaManufCluster.cluster_id,
>                 ],
>                 OUTPUT_CLUSTERS: [
>                     Time.cluster_id, 
>                     Ota.cluster_id
>                 ],
>             }
>         },
>     }
> ```

Is anyone able to help with with this code, I've been searching for the last 2 hours (i'm new to HA) on what to do with custom quirk and I still have no idea where this code goes in HA. Any assistance would be much appreciated

juf00th, 2025-01-05:
> I was having trouble adding a test device to my Home Assistant using the V2 quirk so I removed the line ".skip_configuration()" and now it seems to be working to add a new device.
>
> I am seeing the [...]

Greetings. Using your code I have the following result.

[image]

Peter376, 2025-04-11:
I had a perfectly working quirk for months for this Tuya model. I have 3 identical units. As from sat April 5th, the 3 devices are showing the same malfunction behaviour, despite the quirk being installed. Did anything happen on sat 5th?

wired-dev, 2025-07-17:
trying to get this working myself . cant seem to get it to be detected/used. other quirks work for me. ty

jeremyzone, 2025-09-22:
> ```
> # if value > 100 and value < 1000:
> #                value *= 10
> 
> #            if value % 10 == 6:
> #                value = 10 * (value // 10)
> if value < 2000:
>     value += 6550
> ```
> 
> I did it like this. And it works.

Mine doesn't work with this edit, but seems to be working with the original quirk that I posted. So I guess there's a couple different schemes that Tuya is using with the same model? Can you post the device signature for your sensor? (You can access it by going to "Manage Zigbee device")

jeremyzone, 2025-09-22:
> trying to get this working myself . cant seem to get it to be detected/used. other quirks work for me. ty trying to change it to f from c

Did you get it working?

jeremyzone, 2025-10-08:
> [@jeremyzone](https://github.com/jeremyzone) I just applied your quirk, and the device reports back that the cluster write to change to Fahrenheit worked, but the display stays in Celsius. The LCD appears to support an "F," so I'm wondering if it could be yet another slightly-modified widget. It has yet to be naturally humid enough for me to confirm that correction, but if it doesn't happen organically in the next few days, I will stick it in bag with paper towels.

Thanks for pointing out this problem! I hadn't tested the temperature scale switching feature for a long time, and you're 100% right. It allowed me to switch my screens from Fahrenheit to Celsius, but then wouldn't successfully allow me to go back to Fahrenheit again.

I have made some modifications to the code and now mine are working. Please try the updated code below to see if it works.

For the humidity, I have tested mine by just blowing air from my mouth across the sensor to get it to elevate the humidity level.

```
"""https://github.com/zigpy/zha-device-handlers/issues/3097 devices support."""

import logging
from typing import Dict, Final, Any, Optional, Union
from zigpy.quirks.v2 import QuirkBuilder
import zigpy.types as t
from zigpy.zcl.clusters.measurement import RelativeHumidity
from zigpy.zcl.foundation import ZCLAttributeDef
from zigpy.quirks import CustomCluster
from zigpy.zcl import foundation
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA,
)

_LOGGER = logging.getLogger(__name__)

class TempDisplayUnits(t.enum8):
    """Temperature Display units."""
    Celsius = 0x00
    Fahrenheit = 0x01

class TuyaTempSwitcher(TuyaMCUCluster):
    """Tuya Temperature Display units switcher."""

    class AttributeDefs(TuyaMCUCluster.AttributeDefs):
        """Manufacturer specific attribute definitions."""
        temperature_display_units: Final = ZCLAttributeDef(
            id=0x0065,  # DP 101
            type=TempDisplayUnits,
            is_manufacturer_specific=True,
        )

    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
        **kwargs: Any,
    ):
        """Override the default Cluster command."""
        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "temperature_display_units",
            converter=lambda x: TempDisplayUnits(x),
        ),
    }

    data_point_handlers = {
        101: "_dp_2_attr_update",
    }

class RelativeHumidityFolded(CustomCluster, RelativeHumidity):
    """Handles invalid humidity values from Tuya devices using frame control field context."""

    def deserialize(self, data):
        """Intercept raw frame so we can extract frame control byte."""
        fcf = data[0]  # frame control is always the first byte
        self._last_fcf = fcf  # stash for use in _update_attribute
        return super().deserialize(data)

    @staticmethod
    def _fix_humidity(raw: int, fcf: int, nwk: int) -> int:
        """Convert raw humidity using raw value and ZCL frame control field."""

        # Rule 1: 0–999 and fcf == 0x08 → raw * 10
        if 0 <= raw <= 999 and fcf == 0x08:
            return raw * 10

        # Rule 2: 0–3347 and fcf == 0x18 and ones digit == 6 → strip ones, +656, *10
        if 0 <= raw <= 3347 and fcf == 0x18 and raw % 10 == 6:
            return (raw // 10 + 656) * 10

        # Rule 3: 1000+ and ones digit == 0 → report raw as-is
        if raw >= 1000 and raw % 10 == 0:
            return raw

        # Fallback
        _LOGGER.debug(
            "[0x%04x] TS0201 quirk fallback used | FCF=0x%02x | raw=%d | reported=%d (%.1f%%)",
            nwk, fcf, raw, raw, raw / 100,
        )
        return raw

    def _update_attribute(self, attrid, value):
        if attrid == 0:  # humidity measured_value
            nwk = self.endpoint.device.nwk
            fcf = getattr(self, "_last_fcf", None)
            value = self._fix_humidity(value, fcf or 0, nwk)

        super()._update_attribute(attrid, value)

(
    QuirkBuilder("_TZ3210_ncw88jfq", "TS0201")
    .adds(TuyaTempSwitcher)
    .replaces(RelativeHumidityFolded)
    .enum(
        TuyaTempSwitcher.AttributeDefs.temperature_display_units.name,
        TempDisplayUnits,
        TuyaTempSwitcher.cluster_id,
        translation_key="TS0201_Temperature_Scale",
        fallback_name="TS0201 Temperature Scale",
    )
    .add_to_registry()
)
```

smartyhall, 2025-10-08:
Last night I noticed that the sensor had stopped reporting in any meaningful way, but after applying the new quirk and repairing, it seems to be fully-functional again — including the ability to switch the display between Celsius and Fahrenheit. I'll report back once it's had a time to idle for a while.

BPS-SCHEDULER, 2026-11-03:
### Confirming @honza73's decode with 1028 samples, plus a current-API v2 quirk

Another `_TZ3210_ncw88jfq` TS0201 of the **rollover variant**. Signature matches the OP exactly
(ep 1, device type `0x0302`, in clusters `0x0000 0x0001 0x0402 0x0405 0xef00`,
`manufacturer_code` 4417). I can't contribute a firmware version: ZHA reports
`firmware_version: None` and direct reads of Basic `0x4000`/`0x0001`/`0x0002` all time out, since
it only answers while awake. Temperature has never been affected on mine.

#### Evidence: 8 days, 1028 samples, no encoding mixing on this unit

Pulled from the recorder rather than hand-timed, so the pairs are the device's own series:

```
last digit of measured_value:   {0: 261,  6: 767}      <- only 0 and 6, no third digit
values ending in 0:   53.80 - 65.50 %                  <- decoded correctly as-is
values ending in 6:    0.06 - 16.16 %  -> + 65.44 ->  65.50 - 81.60 %
```

The two branches are **exactly contiguous**: the "ends in 0" branch tops out at 65.50 % and the
decoded "ends in 6" branch starts at 65.50 %. No gap, no overlap, across a week of real humidity.

This is relevant to @jmuf's point about the two workarounds conflicting. I'm not disputing
@jeremyzone's frame-control finding — clearly some units do mix encodings. But **this unit never
did, in 1028 consecutive samples**, so for the rollover variant the final digit alone is a
sufficient discriminator, and a digit-only quirk is viable.

#### A small arithmetic note on the two posted constants

@honza73's `value = 10 * (value // 10); value += 6550` is a **net +6544**, because the trailing 6
is zeroed first. @jeremyzone's Rule 2 (`drop the 6 -> add 656 -> x10`) is a **net +6554**. They
differ by one 0.1 pp step. Checked against every one-decimal example in this thread:

| source | raw | stated actual | +6544 | +6554 |
|---|---|---|---|---|
| OP | 216 | 67.6 % | **67.60** | 67.70 |
| OP | 2986 | 95.3 % | **95.30** | 95.40 |
| @honza73 | 686 | 72.3 % | **72.30** | 72.40 |

All three favour +6544. My own boundary data is consistent with *either* (it cannot resolve a
single 0.1 step — folded `0.06` maps to 65.50 under +6544 and 65.60 under +6554), so I'm going on
those three quoted examples rather than on my own device. Worth noting that @jeremyzone's
`3346 -> 99 %` doesn't settle it either way: +6544 gives 98.90, which also reads as "99" on an
integer display. So this may simply be a further variant rather than an error — but if anyone has
a one-decimal reading above 65.5 % to add, that would pin it.

#### Working quirk on current libraries (zha-quirks 2.3.0 / zigpy 2.4.1)

The code posted in this thread uses import paths that are now deprecated shims. This loads
cleanly today:

```python
"""Unfold the humidity reports of the TS0201 by _TZ3210_ncw88jfq.

https://github.com/zigpy/zha-device-handlers/issues/3097

RH <= 65.50 %  measured_value = RH * 100, last digit always 0  -> already correct
RH >  65.50 %  measured_value restarts near zero, last digit
               always 6                                        -> low by 65.44 % RH
"""

import logging

from zhaquirks.builder import QuirkBuilder
from zhaquirks.clusters import CustomCluster
from zigpy.zcl.clusters.measurement import RelativeHumidity

_LOGGER = logging.getLogger(__name__)

FOLD_OFFSET = 6544  # centi-percent, == 65.44 % RH
MEASURED_VALUE = RelativeHumidity.AttributeDefs.measured_value.id

class FoldedRelativeHumidity(CustomCluster, RelativeHumidity):
    """Add the fold offset back on to the high-humidity branch."""

    def _update_attribute(self, attrid, value):
        attr_id = getattr(attrid, "id", attrid)
        if (
            attr_id == MEASURED_VALUE
            and isinstance(value, int)
            and not isinstance(value, bool)
            and value % 10 == 6
        ):
            _LOGGER.debug("unfolding humidity %s -> %s", value, value + FOLD_OFFSET)
            value = value + FOLD_OFFSET
        super()._update_attribute(attrid, value)

(
    QuirkBuilder("_TZ3210_ncw88jfq", "TS0201")
    .replaces(FoldedRelativeHumidity)
    .add_to_registry()
)
```

Three things that bit me, in case they save someone else the time:

1. `zigpy.quirks.CustomCluster` and `zigpy.quirks.v2.QuirkBuilder` are deprecation shims now —
   quirks v1 moved to `zha-device-handlers`. Use `zhaquirks.clusters` and `zhaquirks.builder`.
2. `_update_attribute` can receive `attrid` as an `int`, a `t.uint16_t` **or a `ZCLAttributeDef`**,
   so a bare `==` against the attribute id silently misses the last case.
3. 🔴 `type(value) is int` **never matches**. `RelativeHumidity.AttributeDefs.measured_value` is
   declared `t.uint16_t`, which is a *subclass* of `int`, not `int` itself
   (`type(t.uint16_t(786)) is int` -> `False`). A quirk written that way loads, reports
   `quirk_applied: True`, and silently does nothing. Use `isinstance`.

This deliberately handles **only** the rollover variant and leaves any value not ending in 6
untouched, rather than guessing — on a unit that mixes encodings a wrong value stays visibly
wrong instead of being converted into a plausible one. It is therefore not a general fix for all
`_TZ3210_ncw88jfq` units, and I think that is the open question for landing anything upstream:
whether the variants can be told apart reliably enough (frame control, or a firmware read, which
I can't perform on a sleepy device) for one quirk to serve all of them.

Happy to open a PR if a maintainer would find it useful, though given the variant problem I
suspect the decision on discrimination needs to come first.