zigpy/zha-device-handlers #3103: [Device Support Request] HLK-LD2410S "in Z2M" ZG-204ZM

Farmand77, 2024-04-10:
### Problem description

Is it possible to get this device working in ZHA it is only listes as a motion sensor + illumination while in Z2M it is listed and working as a battery operated hybrid device. Both a motion sensor and a pencenes sensor.

### Solution description

Is it possible to get this device working in ZHA it is only listes as a motion sensor + illumination while in Z2M it is listed and working as a battery opperated hybrid device. Both a motion sensor and a pencenes sensor.
All option listed for said device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
https://www.youtube.com/watch?v=I5LtyY1zKJM

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

crippollo, 2024-05-07:
Please i jsut got this and didn't realise i wouldnt be able to use it all in ZHA

hfagelnour, 2024-05-07:
Perfectly working with z2mqtt

e8link, 2024-05-20:
Would be great if this worked with ZHA.   A battery mmWave sensor!

vinzent, 2024-05-20:
Probably a duplicate of this: https://github.com/zigpy/zha-device-handlers/issues/3125

vinzent, 2024-05-20:
https://github.com/zigpy/zha-device-handlers/issues/3125#issuecomment-2117681628

https://gist.github.com/vinzent/2cd645b848fd3b6a0c3e5762956ec89f

vinzent, 2024-05-20:
see here: 

https://github.com/zigpy/zha-device-handlers/issues/3125#issuecomment-2117681628

and here:

https://gist.github.com/vinzent/2cd645b848fd3b6a0c3e5762956ec89f

closed 2024-11-23 by github-actions[bot]