zigpy/zha-device-handlers #3108: TS004F by _TZ3000_ja5osu5g

jrmurdoch, 2024-04-12:
### Problem description

This is a single button device, its working in a switch like manner, single press "on" double press "off"

`_TZ3000_ja5osu5g TS004F Switch turned on
11:48:45 - 3 minutes ago
_TZ3000_ja5osu5g TS004F Switch turned off
11:48:42 - 4 minutes ago`

### Solution description

I'd like to use this to open a garage door, is it possible to change the output to: single press / double press / long press

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x1000",
        "0xe001"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_ja5osu5g",
  "model": "TS004F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.4.3",
    "dev": false,
[... 396 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

vJan00, 2024-04-13:
Hey there,
I got the same problem over here and I fixed it, here is how:

The quirk for the button already exists but it assumes that the Button is from the `DeviceType` `DIMMER_SWITCH` (0x0104).
Instead you Button is saying it's an `ON_OFF_SWITCH` (0x0000) which is why ZHA won't load the right quirk.

To Fix this issue you have to install the quirk manually and change the DeviceType:

1. Go into your HomeAssistant configuration directory where the configuration.yaml is located. Then make a new directory if it doesn't already exists called `zha_quirks`.
2. Go into the created directory and add the file from the following link [ts004f.py](https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts004f.py).
3. Edit the File and change line 335 (TuyaSmartRemote004FSK Class -> Signature -> Endpoints -> 1 -> DeviceType) from `zha.DeviceType.DIMMER_SWITCH` to `zha.DeviceType.ON_OFF_SWITCH`
4. Go back into the config folder and edit your configuration.yaml. Add the following lines:
```
zha:
  custom_quirks_path: /config/zha_quirks
```
5. Restart HomeAssistant and see if the quirk now is getting used.

Maybe someone can look up how we can fix this longterm? A custom quirk should not be the longterm solution.

nealhead, 2024-04-13:
> 3\. zha.DeviceType.ON_OFF_SWITCH

I can't get this working for the life of me either. My button presses don't even show up in events.

vJan00, 2024-04-13:
> My button presses don't even show up in events.

If the signature shows the following the quirk isn’t getting loaded correctly:
`"class": "zigpy.device.Device"`

If you post your Signature here I could double check if your button is the one these topic goes about, not ever button „TS004F _TZ3000_ja5oso5g“ is the same.

jrmurdoch, 2024-04-13:
> > 3. zha.DeviceType.ON_OFF_SWITCH
> 
> I can't get this working for the life of me either. My button presses don't even show up in events.

If it doesn't show up at all press the button 3 times fast. This switches the mode then the events should show

nealhead, 2024-04-13:
> > My button presses don't even show up in events.
> 
> If the signature shows the following the quirk isn’t getting loaded correctly: `"class": "zigpy.device.Device"`
> 
> If you post your Signature here I could double check if your button is the one these topic goes about, not every button „TS004F _TZ3000_ja5oso5g“ is the same.

Strike that, following your directions (and actually reading them thoroughly this time) seems to have fixed it. I hadn't changed the ON_OFF switch portion to the correct devices.

closed 2024-11-18 by github-actions[bot]

smezei, 2024-12-02:
Not working with the latest HA in docker conatiner.

dirkclemens, 2025-03-15:
Hi [vJan00](https://github.com/vJan00), what do you mean with "Signature". My _TZ3000_ja5osu5g logs a single click as 
```
_TZ3000_ja5osu5g TS004F Toggle event was fired
_TZ3000_ja5osu5g TS004F Press Type event was fired with parameters: {'press_type': 0}
_TZ3000_ja5osu5g TS004F Remote Button Short Press - Button 1 event was fired
```
but in an automation, I only can work with a double click even when single click is selected in the automation. 
I did the correction described by you in your post from [on Apr 13, 2024](https://github.com/zigpy/zha-device-handlers/issues/3108#issuecomment-2053757968)

Maybe you are looking for this one?
[zha-58b15438d145a76bd25043008140a09b-_TZ3000_ja5osu5g TS004F-3b1a679195ee99808eca8831df427f5b.json](https://github.com/user-attachments/files/19263064/zha-58b15438d145a76bd25043008140a09b-_TZ3000_ja5osu5g.TS004F-3b1a679195ee99808eca8831df427f5b.json)

putimir, 2025-06-05:
> My _TZ3000_ja5osu5g logs a single click as
>
> ```
> _TZ3000_ja5osu5g TS004F Toggle event was fired
> _TZ3000_ja5osu5g TS004F Press Type event was fired with parameters: {'press_type': 0}
> _TZ3000_ja5osu5g [...]

Hi, I have the same situation, did you manage to sort this out to get more events? Thank