zigpy/zha-device-handlers #3112: [Device Support Request]  TS0601  _TZE204_7gclukjs  . Presence Sensor

casa3c, 2024-04-15:
### Problem description

I've bought this  Sensors in AliExpress.
TS0601 _TZE204_7gclukjs, it is detected but not sensor are shown in ZHA nor controls.

### Solution description

Add the new manufacturer "_TZE204_7gclukjs" to the mapping of the "TS0601" Mapping in the file zhaquirks/tuya/ts0601_sensor.py.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]
[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

casa3c, 2024-04-15:
[image]

KLM30330, 2024-04-18:
Hello, I'have the samedi device, do you find a solution ?

casa3c, 2024-04-22:
Thanks for this!

I have tried and it works perfectly. The only thing does not work is the light sensor (not important to me).
How did get it? Did you set up the quirk yourself or found it in internet?

Thanks again,

cygnusb, 2024-06-27:
Works also fine for me, excluding mentioned illuminance sensor which reports 0 or 1. Possible wrong lambda?

cygnusb, 2024-07-04:
It seems that datapoint 104 is not illuminance but 103. See https://github.com/Koenkk/zigbee2mqtt/issues/21738

closed 2025-01-07 by github-actions[bot]

DogFox, 2025-02-06:
Not sure if this is the correct behavior

[image]