zigpy/zha-device-handlers #3115: [Device Support Request] Zemismart/Tuya cover blind battery support ts0601

sushant-here, 2024-04-18:
### Problem description

I have multiple Zemismart roller blind motors that are handled via `Quirk: zhaquirks.tuya.ts0601_cover.TuyaMoesCover0601`. I can view the battery level in the Tuya app which suggests the device does somehow expose the data. Would it be possible to get this added to HomeAssistant so that i can easily see which blinds need to be charged?

### Solution description

Expose new sensor for monitoring battery level.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_68nvbio9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_cover.TuyaMoesCover0601"
}```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.4.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "6.1.73-haos-raspi",
    "supervisor": "2024.04.0",
    "host_os": "Home Assistant OS 12.2",
    "docker_version": "25.0.5",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "bosch_alarm": {
      "version": "1.0.4",
      "requirements": [
        "bosch-alarm-mode2==0.3.37"
      ]
    },
[... 401 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {
              "0x0002": {
                "attribute_name": "current_file_version",
                "value": 72
              }
            },
            "unsupported_attributes": {
              "0x0002": {
                "attribute_name": "current_file_version"
              }
            }
          }
        }
      }
    }
  }
}

```

</details>

stast1, 2024-04-21:
Try it https://github.com/zigpy/zha-device-handlers/issues/3118

sushant-here, 2024-04-21:
Thanks @stast1!... your custom quirk works. I have commented on #3118 with more details.

closed 2024-04-21 by sushant-here