zigpy/zha-device-handlers #3116: How to tell if quirk exist and used

Fredrik81, 2024-04-18:
Hello
Is there a good way to tell if quirk exist for device and if so is it used and I just need to re-pair?

MattWestb, 2024-04-18:
Easiest way is looking on the device card and expanding the device information.
If having the last line "Quirk:" the system have loaded one quirk and if no like its not have loading one quirk (good devices do need quirk then being Zigbee certified ones).
If looks like in the screen shut its from HA and if loading local its giving the path to the quirk.
[image]

Fredrik81, 2024-04-19:
> Easiest way is looking on the device card and expanding the device information.
>
> If having the last line "Quirk:" the system have loaded one quirk and if no like its not have loading one quirk (good [...]

Followup 😄 
I heared that if quirk is added then re-pair is needed. Is that true?

And if yes is there a good way to tell if there quirk is added or updated and repair needed?

closed 2024-10-23 by github-actions[bot]