zigpy/zha-device-handlers #3117: [Device Support Request] Tuya Zemismart TS0601 _TZE200_vdiuwbkq Chain Roller Shades Driver

hobsonlin, 2024-04-20:
### Problem description

 TS0601 
_TZE200_vdiuwbkq

Appears to be an "updated" version of the "Zemismart Chain Roller Shades Driver": https://zigbee.blakadder.com/Zemismart_M515EGB.html

Purchased from AliExpress zemismart Direct Store: 
https://www.aliexpress.com/item/4001343586623.html

### Solution description

Super new to ZHA and not entirely confident about trying to repurpose another quirk (if that might work by just replacing the manufacturer number?) or write one from scratch.

Would greatly appreciate some help to get this device working :)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_vdiuwbkq",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.4.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
[... 373 lines cut ...]
    }
  }
}```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

shuntfield, 2024-09-27:
I have gotten one of these devices too, to replace a broken one, and this one I cannot get working. All you get is the device appear in ZHA but nothing to be able to drive it. I also need to swap its direction then instructions are infuratingly unreadable!

Fidelity88, 2024-10-29:
Ever managed to get this fixed @shuntfield and/or @hobsonlin ?

shuntfield, 2024-10-29:
Nope, I don't think it's been added still.

One of my older ones died mechanically so I swapped the board out on it. Thought that was probably easier than getting the quirk for it.

Fidelity88, 2024-10-29:
I've proposed a change to mark `_TZE200_vdiuwbkq` devices as `_TZE200_gubdgai2` and related. Should fix the issue :)

closed 2025-08-13 by github-actions[bot]