zigpy/zha-device-handlers #3118: [Device Support Request] custom quirks for Tuya Zemismart blind cover motor TS0601 _TZE200_68nvbio9 with battery

stast1, 2024-04-21:
### Problem description

No battery charge display for Tuya Zemismart blind cover motor TS0601 _TZE200_68nvbio9

### Solution description

Add battery charge display

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
"""Tuya based cover and blinds."""
from typing import Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks import Bus
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaPowerConfigurationCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
    TUYA_GET_DATA,
    TUYA_SET_DATA_RESPONSE,
    TUYA_DP_TYPE_VALUE,
)

TUYA_DP_ID_BATTERY = 0x0D

class TuyaZemismartManufacturerWindowCover(TuyaManufacturerWindowCover):
    """Manufacturer Specific Cluster for Zemismart cover device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: tuple[TuyaManufCluster.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        # Tuya Zemismart Specific Cluster Commands
        if (
            hdr.command_id in (TUYA_GET_DATA, TUYA_SET_DATA_RESPONSE) 
            and args[0].command_id == TUYA_DP_TYPE_VALUE + TUYA_DP_ID_BATTERY
        ):
            self.endpoint.device.battery_bus.listener_event(
                "battery_change",
                args[0].data[4],
            )
        else:
            super().handle_cluster_request(hdr, args, dst_addressing=dst_addressing)

class TuyaZemismartSmartCover0601_battery(TuyaWindowCover):
    """Tuya Zemismart blind cover motor."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.battery_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098,
        #                    maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        #                    maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
        # "endpoints": {
        # "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004","0x0005","0xef00"], "out_clusters": ["0x000a","0x0019"] }
        # },
        # "manufacturer": "_TZE200_zah67ekd",
        # "model": "TS0601",
        # "class": "zigpy.device.Device"
        # }
        MODELS_INFO: [
            ("_TZE200_68nvbio9", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZemismartManufacturerWindowCover,
                    TuyaPowerConfigurationCluster,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

### Additional information

I have added support for displaying battery charge for _TZE200_68nvbio9. May be suitable for other motors with batteries. You just need to change _TZE200_68nvbio9 to your model. Haven't tested the charge updates yet, shows 100%

sushant-here, 2024-04-21:
I can definitely confirm that this is reporting something that looks like a battery percentage on a TS0601 _TZE200_68nvbio9. That said... one of my blinds has been plugged in to charger for the last 2 days (so it should be 100%). This quirk is reporting 90% battery.

Blind is 2 years old and has been charged 4-5 times. Not sure if its somehow related to battery degradation.

Other blinds which have not been charged for months are showing lower state of charge (as expected).

Something is better then nothing - I think this can be (cleaned up?) and merged for the wider community to use.

stast1, 2024-04-24:
Tested - the charge data changes.

flybrys, 2024-05-31:
Just wanted to say thanks so much for this quirk. Lost the battery moving from Z2M and now it's back!
Hopefully it gets added to the main repo

Maks1983, 2024-07-11:
Thnak you, this works good!

flybrys, 2025-01-07:
Not stale. How do we merge this into the master?

Floriszz, 2025-07-04:
The screen is reversed now. Close = Open and Open = Closed.
Looking at the name of my original quirk I see an addition of 'inv_position' at the end ;
Quirk: zhaquirks.tuya.ts0601_cover.TuyaMoesCover0601_inv_position
@stast1 , How could I realize this with your custom quirk?
I see that the value below is 'None', but what to change to to set it inverted?

[image]

Floriszz, 2025-07-04:
I found the right syntax for 'Value' on the Home Assistant community site:
https://community.home-assistant.io/t/zigbee-blinds-displays-backwards-frontend/404127/16

[image]
You have to change it to t"Bool.true" for motor.direction AND cover_inverted as well.

Colmin24, 2025-07-19:
i do have the same device - TS0601  By _TZE200_68nvbio9  (blind motor)
restarted homeassistant - also removed all blind motors and re-installed them didnt make a difference. 
Configuration.yamal path 
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/

device signature 
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_68nvbio9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_cover.TuyaMoesCover0601"
}

Colmin24, 2025-07-27:
thanks for your assistance, i have decided to go to Z2M as that shows the battery levels by default.

Joejoeinc, 2025-08-21:
I've found with a slight modification of the file, you can also use this for Blindsmart coverings and it will report the battery. 

MODELS_INFO: [
            ("_TZE200_pw7mji0l", "TS0601"),
        ],

is there a way to have it work for both makes/models without having to create 2 files?

whalemare, 2025-09-21:
How to set "top" and "bottom" positions directly from Home Assistant, but not from Tuya App?
It seems that function is available only through the remote or Tuya app, but it requires another hub only for single settings change 

[image]

ajfhodgson, 2025-12-07:
+1 for this excellent improvement. Should help me catch the blinds running out of battery **before** they stop working and the family raises a ticket!

Notes to potentially save others wasting a few hours:
- Create folder `/config/custom_zha_quirks/`
- Despite some LLMs saying you shouldn't need this, add to configuration.yaml:
`zha:`
`  custom_quirks_path: /config/custom_zha_quirks/`
- create `<any filename, e.g. zemismart_with_battery>.py`  in `/config/custom_zha_quirks/`
- paste into it the new quirk code above
- restart Home Assistant
- wait a significant time for blind(s) to report their battery status (it's NOT immediately polled)
- you DON'T need ZHA Toolkit in HACS (despite some LLMs saying you do)

ogoncalves200874, 2026-04-08:
Thank you so much! It worked perfectly here!

kozmaz87, 2026-08-20:
To me this device does not show up as cover at all under ZHA. How do I get started? It interviews fine and it is a generic end device but it does not expose any meaningful bits.

[image]