zigpy/zha-device-handlers #3121: [QUESTION] How to disable Broadcast?

aaronegger, 2024-04-22:
### Bug description

Hi,

is it possible to disable the possibility of zigbee devices to be able to broadcast signals?
I have the issue that the Osram Lightify Mini Switch switches every single zigbee devices.

I am using deconz with ZHA. I already removed every group (not only the default) group.

But as soon as i add the osram switch it turns all devices on or off (depending on which button is pressed).

Is there a way to completely disable this behaviour?
It somehow scares me that a device could do this without any automation for it configured.

### Steps to reproduce

Connect a Osram Lightify Mini Switch over ZHA.

### Expected behavior

To not be able to switch any device on or off without an automation for it.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2024-04-22:
Z2M is binding it to group 1801 and then the coordinators IEEE.
Shall working the same in ZHA by putting the switch in configuration mode and then doing bindings.
But first do one new group and adding all devices from the default light group in it and then deleting all devices in the default light group. Shall working but if resetting one device or adding new it must being redone for the device.

So the switch cant being disabling broadcast and from ZHA we cant do anything about that only trying doing the same configure that Osram is doing in the Z"M issue but best not using this old switches.

MattWestb, 2024-04-23:
All information i was finding is from the other issue you was writing in and have links that describing configuration mode and how they was adding the no default group to it so its not sending commands to it.

Binding of group its one part of the light system and is making the system working also if the coordinator / host system is not working OK:
I have most  light with bounded controller so it working if coming home at night after one power frailer and my PI have killed the SD-card.
If doing automatons in the upper system its not working until you have getting the system up and also some need working internet connection that i dont like for "critical lights but is OK for many other things like putting the fan on and of in the toilet.

aaronegger, 2024-04-25:
In my case i don't use groups, as i have a hybrid approach. My approach was to never use individual bulbs or devices. Therefore if i have a network issue i am still able to toggle the lights in the traditional way and i use an usv to keep everything up in case of a power shortage. But of course i will never be safe from dying sd cards. This reminded my to create an download a full backup, which i did today. But i now understand why groups can be important and i learned more about how zigbee works :D

I also managed to get it working without toggling everything.
But i wanted to know if binding and unbinding works proberly and therefore used the bind/unbind to device in the zigbee configuration. Binding it to a device worked and the device was controlled immediatly after binding it. But when i tried to unbind it, it didn't work. It show a green checkmark when doing so, an i kept pressing buttons during the process to keep the switch active, but it kept being binded to the test device.

Do you know why this could happen?

I have little hope that there is maybe a solution how this device can be used despite it's quality issues.
The only reason is the size and that it is completely wrapped in a rubber material.

MattWestb, 2024-04-25:
Then you is getting green by sending unbind the device have getting the command and acking it so its out of our control (firmware function).
Perhaps its cant deleting the last bind then its being lost and cant communicating in the network. Adding one more / new device binding is its using the new or the old or both ?
Some devices cant do group bindings other only devices and some only one device
/ group (the number is hard coded in the firmware).
The hardware and software in this device is old and buggy so from that point not recommended (one of the worse device on the market is OSRAM pugs that destroying the network).

Its some nice controllers out that sending toggle and can doing device binding but i have not trying them (IKEA short cut button was one of them).

aaronegger, 2024-04-29:
I haven't tested this yet. I will try this what happes if i bind it to multiple devices.
But i read in another thread that if i get a point where the switch doesn't bind to anything at first connect, i have a good chance that it will keep working without switching other and new devices.
I got my new switches from Aliexpress, and they feel not very suitable for the environment they need to be used.

Thank you very much for your assistance and explanations :)

closed 2024-04-29 by aaronegger