zigpy/zha-device-handlers #3122: [Device Support Request] Amina S EV charger

VVVVVVVip, 2024-04-23:
### Problem description

New cool charger for electric cars that is independent of cloud connections and other third party solutions and instead can be controlled thru Zigbee.

> **How Amina S behaves in a Zigbee network**
> _Amina S is an routing (end) device.
> Amina S implements 3 endpoints with multiple clusters, see below.
> At manufacturing time each Amina S will be given a unique install code which is printed labels both on the device and in the user guide. The install code is available in printed format and also machine readable (QR-code).
> The manufacturing code for Amina Distribution AS is 0x143B._

Manufactures zigbee documentation; [https://doc.clickup.com/9004130215/p/h/8cb07x7-18308/c750786359b035a](url)

They claim in documentation above that
_> Tech savvy amina S customers have used our Zigbee documentation to set up their own automations in Home Assistant. Some have also published their work for others to use, such as a Zigbee2MQTT converter._
but I can't find any trace of information that this is correct.

Amina S website [https://aminacharging.com/products/amina-s/](url)

### Solution description

Add Amina special clusters in linked documentation (and other clusters if needed)
**Amina S Control (0xFEE7)**
_This cluster is an Amina specific cluster, and contains attributes not belonging naturally to any of the other clusters. It is used to control different aspects of the charger. The following attributes are available:_
[image]

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

VVVVVVVip, 2024-04-28:
[https://github.com/somlioy/amina_s](url)
Here is a Zigbee2MQTT Converter, could this be applied in ZHA?

VVVVVVVip, 2024-04-29:
Tested some more and updated first post with more information, diagnostic and signature.

VVVVVVVip, 2024-05-06:
In my way trying to find some way forward, I tried to chat with Google Gemini. My first try with AI.
Can someone tell if the result is even useful somehow?
This is the code it generated for cluster: **Level Control (0x0008)** (but it looks like the formatting changes when copied here)

```
from zigpy.endpoints import BaseEndpoint
from zigpy.clusters import LevelControlCluster

class AminaSLevelControlCluster(LevelControlCluster):
  """Custom cluster implementation for Level Control on Amina S device."""

  attribute_map = {
      # Map attribute IDs from the device to ZHA attributes
      # This section might require adjustments based on the diagnostic info
      0x0000: 'current_level',  # Replace with the actual attribute ID for current level
      # Add entries for other attributes based on diagnostic information
  }

  @property
  def current_level(self):
      """Read the current level from the device and convert it to a ZHA format."""
      # Replace this with logic to read the current level from the device
      # Refer to explanations in the previous code block.
      pass

  def set_level(self, level):
      """Send a command to the device to set the target level."""
      # Replace this with logic to send a Zigbee command to set the level
      # Refer to explanations in the previous code block.
      pass

  # Implement methods for the additional attributes based on their functionalities:
  @property
  def remaining_time_to_level(self):
      """Read the remaining time to reach the target level (if supported)."""
      # Replace with logic to read the remaining time data from the device
      # This might not be supported by the Amina S device, check documentation.
      # Consult Zigbee Cluster Library specification for the format of the data.
      pass

  def move_to_level(self, level, transition_time):
      """Send a command to move to a target level with a transition time."""
      # Replace with logic to send a Zigbee command for move_to_level
      # Refer to Zigbee Cluster Library specification for the command format.
      pass

  # Similar methods for other attributes (replace placeholders with actual logic):
  @property
  def on_level(self):
      pass

  @property
  def on_off_transition_time(self):
      pass

  @property
  def ramp_rate(self):
      pass

  def step_with_on_off(self, updown, amount, transition_time):
      pass

  def stop_move(self):
      pass

class AminaSDevice(BaseEndpoint):
  """Basic device class for the Amina S device."""

  # Define other clusters and functionality based on your analysis of the diagnostic information
```
Comment from Gemini.
> Explanation of Added Functionality:
> 
> We've added properties and methods for the seven attributes listed under Command Support:
> remaining_time_to_level: This reads the remaining time to reach the target level (if supported). You'll need to check the diagnostic information and potentially the Amina S documentation to see if this attribute is actually reported by the device.
> move_to_level: This sends a command to move to a target level with a transition time. You'll need to implement this based on the Zigbee Cluster Library specification for the move_to_level command.
> Similar methods are included for the other attributes. However, these are currently placeholders, and you'll need to fill them in with logic to:
> Read the data from the device (if supported) using appropriate Zigbee library methods.
> Convert the data to a format suitable for Home Assistant attributes (refer to ZHA documentation).
> 
> Important Note:
> The provided information (diagnostic information and Amina S documentation pages 2-15) might not be sufficient to fully implement all these functionalities. You might need to consult additional resources or the device manufacturer for specifics.
> Remember to adjust the code further based on the actual attribute IDs and data formats identified through the diagnostic information.
> 
> I hope this expanded version helps you get started with writing the quirk for the Amina S Level Control cluster!_

dennisyakkz, 2024-06-18:
I have added the charger in ZHA.

Inside the "Manage device" function in ZHA all relevant endpoints and commands are exposed. 
I can query charging state, power consumption. Using the commands it's also possible to control the charger (start/stop, specify max charging power)
However these endpoints/commands are not exposed in the generated home assistant device.

In order to make best use of the device the following endpoints and commands need to be exposed:

**Commands:**
- OnOff  (enable / disable charging)
- Move To Level (Set allowed charging power in Amps, (value from 6A up to Level control MaxLevel)

**Endpoints:**
energy usage:
- Total Active Energy  (total power consumption in kWh)
- Last Session Energy (last session power consumption in kWh)

statistics:
- EV Connect Status
- Current Power (total power drawn in kW)
- Voltage per phase
- Current per phase (in A. This is needed to be able to do load management or solar charging)
- Alarm (show if there's an alarm and the corresponding alarm message)

**Desirable:**
- Single Phase (specify to charge at single phase or 3 phase. I wasn't able to test this endpoint)
- OTA firmware update

I can help out in engineering, testing and technical stuff. But I'm zero at building the quirk itself.

If anyone is upto developing the actual quirk so that the device is properly supported in Home Assistant. I'm able to donate a device for that.

arossmann, 2024-11-27:
> I have added the charger in ZHA.
> 
> Inside the "Manage device" function in ZHA all relevant endpoints and commands are exposed. I can query charging state, power consumption. Using the commands it's also possible to control the charger (start/stop, specify max charging power) However these endpoints/commands are not exposed in the generated home assistant device.
> 
> In order to make best use of the device the following endpoints and commands need to be exposed:
> 
> **Commands:**
> 
> * OnOff  (enable / disable charging)
> * Move To Level (Set allowed charging power in Amps, (value from 6A up to Level control MaxLevel)
> 
> **Endpoints:** energy usage:
> 
> * Total Active Energy  (total power consumption in kWh)
> * Last Session Energy (last session power consumption in kWh)
> 
> statistics:
> 
> * EV Connect Status
> * Current Power (total power drawn in kW)
> * Voltage per phase
> * Current per phase (in A. This is needed to be able to do load management or solar charging)
> * Alarm (show if there's an alarm and the corresponding alarm message)
> 
> **Desirable:**
> 
> * Single Phase (specify to charge at single phase or 3 phase. I wasn't able to test this endpoint)
> * OTA firmware update
> 
> I can help out in engineering, testing and technical stuff. But I'm zero at building the quirk itself.
> 
> If anyone, who resides in the Netherlands, is upto developing the actual quirk so that the device is properly supported in Home Assistant. I'm able to donate a device for that.

are you able to control the behaviour of the Amina S with the toggle? based of the [documentation](https://doc.clickup.com/9004130215/p/h/8cb07x7-18308/c750786359b035a) they state for the toggle 
> Will toggle between on and off. SHOULD NOT BE USED as it may result in unwanted results, but is mandatory in ZCL rev 2

closed 2025-08-13 by github-actions[bot]