zigpy/zha-device-handlers #3125: TS0601 _TZE200_kb5noeto Tuya Presence and Motion Sensor with illuminance

jdm09, 2024-04-26:
### Problem description

Hi,
this is another Zigbee Device which is recognized as a default motion sensor in ZHA, but without any configurations. The Device is a battery powered presence and motion detector including illuminance sensor.

I bought this from here:
https://de.aliexpress.com/item/1005006190626320.html

Thought, it should be near by this device quirk:
https://github.com/zigpy/zha-device-handlers/blob/bfbfb4876d1811440b4a05cac9e3b83311d194a8/zhaquirks/tuya/ts0601_motion.py#L353

but its not working.

Signature of this device:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0500"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_kb5noeto",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Solution description

I tried to put the device model and manufacturer to the quirk above.

Would be glad, if someone could help with the quirk for this device.
Thanks.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

cobirnm, 2024-05-16:
@jdm09 I've managed to get this working using the below quirk. I get illuminance and motion. I don't know if I should get ocupancy sensor or settings for the sensitivity. For that I can't help but would be verry interested if someone could help with this. Please note that I'm not a coder and I just did some research and added the model to this quirk.

```
"""ZY-M100 Human Presence Sensor"""
import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    #DPToAttributeMapping,
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaNewManufCluster,
    TuyaDPType,

)

from zhaquirks.tuya.mcu import (
    TuyaAttributesCluster,
    DPToAttributeMapping,
    TuyaMCUCluster,
)

ZONE_TYPE = 0x0001

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
[... 264 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarSensitivity,
                ],
                OUTPUT_CLUSTERS: [],
            },

			#### end of new replacements.
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

vinzent, 2024-05-16:
@cobirnm  how does your `_TZE200_kb5noeto` device look like? Is it a battery powered device? your definition shows much more functionality than the zigbee2mqtt defines: https://github.com/13717033460/zigbee-herdsman-converters/blob/6c9cf1b0de836ec2172d569568d3c7fe75268958/src/devices/tuya.ts#L5730-L5762

also the ZY-M100 in the name look more like: https://smarthomescene.com/reviews/tuya-zigbee-human-presence-sensor-zy-m100-review/ instead of https://de.aliexpress.com/item/1005006190626320.html .

cobirnm, 2024-05-17:
> oes your `_TZE200_kb5noeto` device look like? Is it a battery powered devic

Hello, 

this is my device https://www.aliexpress.com/item/1005006851082995.html

Regarding functionality I'm still testing but something strange is happening. Without the above quirk I can't add the device. If I add the quirk the device gets added but when I look at HA the quirk is not being used. Never the less it is working and I can't explain it.

[image]

constapel, 2024-05-17:
TS0601 by _TZE200_kb5noeto Also known as Tuya ZG-204ZM. (PIR+radar 24Ghz)
https://smarthomescene.com/blog/best-and-worst-presence-sensors-for-home-assistant/#best-battery-presence-sensor
Please look into zha drivers.
Zigbee2mqtt is no option. Parameters can not be addjusted. Others say device is constantly sending out its state to the coordinator

https://nl.aliexpress.com/item/1005006803782882.html

vinzent, 2024-05-17:
my first try to add a quirk for this ZG-204ZM 

```python
"""
 * TS0601 ZG-204ZM
 * _TZE200_kb5noeto
 * https://de.aliexpress.com/item/1005006174074799.html ("Color": Mmwave PIR)
 * https://github.com/13717033460/zigbee-herdsman-converters/blob/6c9cf1b0de836ec2172d569568d3c7fe75268958/src/devices/tuya.ts#L5730-L5762
 * https://www.zigbee2mqtt.io/devices/ZG-204ZM.html
 * https://smarthomescene.com/reviews/zigbee-battery-powered-presence-sensor-zg-204zm-review/
 * https://doc.szalarm.com/zg-205ZL/cntop_zigbee_sensor.js
 * https://github.com/Koenkk/zigbee2mqtt/issues/21919
"""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    Basic,
    PowerConfiguration,
    Identify,
    Ota,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import (
    DPToAttributeMapping,
    TuyaLocalCluster,
    TuyaPowerConfigurationCluster2AAA,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster

class MotionState(t.enum4):
    """MotionState enum."""

    NONE = 0x00
    LARGE = 0x01
    SMALL = 0x02
    STATIC = 0x03

class StaticDetectionSensitivity(t.uint8_t):
    """Static detection sensitivity: 1-10"""

class StaticDetectionDistanceCm(t.uint16_t):
    """Static detection distance in centimeter: 0-10000 cm"""

class FadingTimeSec(t.uint16_t):
    """Fading time: 0-28800 seconds"""

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

    attributes = OccupancySensing.attributes.copy()
    attributes.update({
        0xFF00: ("manufacturer_motion_state", MotionState, True),
        0xFF01: ("manufacturer_fading_time", FadingTimeSec, True),
        0xFF02: ("manufacturer_static_detection_distance", StaticDetectionDistanceCm, True),
        0xFF03: ("manufacturer_static_detection_sensitivity", StaticDetectionSensitivity, True),

    })

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaPir24GhzMmwaveSensorBatteryPoweredManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
        ),
        2: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "manufacturer_static_detection_sensitivity",
        ),
        4: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "manufacturer_static_detection_distance",
        ),
        101: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "manufacturer_motion_state",
        ),
        102: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "manufacturer_fading_time",
        ),
        106: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
        ),
        121: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
        ),
    }

class TuyaPir24GhzMmwaveSensorBatteryPowered(CustomDevice):
    """PIR + Millimeter wave occupancy sensor. ZG-204ZM"""

    signature = {
        #  endpoint=1, profile=260, device_type=81, device_version=1,
        #  input_clusters=[0, 1, 3, 1024, 10280], output_clusters=[]
        MODELS_INFO: [
            ("_TZE200_kb5noeto", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IlluminanceMeasurement.cluster_id, # (1024)
                    IasZone.cluster_id, # (1280)
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPir24GhzMmwaveSensorBatteryPoweredManufCluster,
                    TuyaOccupancySensing,
                    TuyaIlluminanceMeasurement,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

[image]

cobirnm, 2024-05-17:
> my first try to add a quirk for this ZG-204ZM.
> 
> Copied into gist: https://gist.github.com/vinzent/2cd645b848fd3b6a0c3e5762956ec89f#file-zg-204zm-py
> 

I used your quirk and it seems better. only the setting parameters are missing.

In this review: https://smarthomescene.com/reviews/zigbee-battery-powered-presence-sensor-zg-204zm-review/ it says that Z2M has settings. It would be great to have it! Never the less this looks quite good!

[image]

vinzent, 2024-05-17:
@cobirnm the attributes arent shown "nicely". You should be able to read/write the attributes by visiting the "Manage Zigbee-Device" link in the 3-dots menu:

[image]

then:

[image]

Honestly I really don't know if everything is working fine. The first time I read the attributes, it reported "None". then i wrote some value. Afterwards if I read it will show the value i've saved. But I don't know if it has any effect on the device.

vinzent, 2024-05-17:
Something doesnt seem to be quite right:

``` 
2024-05-17 16:08:13.116 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE4BF:1:0x0400]: cluster_handler[illuminance] attribute_updated - cluster[Illuminance Measurement] attr[measured_value] value[2570]
2024-05-17 16:08:13.444 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE4BF:1:0x0400]: cluster_handler[illuminance] attribute_updated - cluster[Illuminance Measurement] attr[measured_value] value[34099]
2024-05-17 16:12:13.169 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE4BF:1:0x0400]: cluster_handler[illuminance] attribute_updated - cluster[Illuminance Measurement] attr[measured_value] value[2560]
2024-05-17 16:12:13.647 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE4BF:1:0x0400]: cluster_handler[illuminance] attribute_updated - cluster[Illuminance Measurement] attr[measured_value] value[34082]
2024-05-17 16:16:13.219 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE4BF:1:0x0400]: cluster_handler[illuminance] attribute_updated - cluster[Illuminance Measurement] attr[measured_value] value[2532]
2024-05-17 16:16:13.778 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE4BF:1:0x0400]: cluster_handler[illuminance] attribute_updated - cluster[Illuminance Measurement] attr[measured_value] value[34035]
```

The illuminance seems to be reported 2x each time.

vinzent, 2024-05-17:
I've updated https://gist.github.com/vinzent/2cd645b848fd3b6a0c3e5762956ec89f

seems the IlluminanceMeasurement is binding to the 0x0000 measured_value too and does some calculations. whereas the tuya datapoint would deliver the already calculated value. Don't have enough insights to understand whats going on here. disabling the illuminance measured_value datapoind did fix this issue.

BambamNZ, 2024-05-26:
Loaded the quirk which changed it from motion to presence however, it goes to presence detected and does not change at all. 

[image]

I have removed it and readded it as a new name but that did not change anything.

vinzent, 2024-05-26:
@BambamNZ it sometimes takes 20-30 minutes to change to unoccupied/no-motion. with or without the quirk.  with the quirk enabled, did you try to change some of the settings (distance, detection sensitivity)?

BambamNZ, 2024-05-26:
No matter which setting you tweak it detects presence after you install the batteries and no amount of tweaking of settings get it to clear the presence. I put it in a drawer for 4 hours and still shows presence detected.

Motion_State changes between "Large" & "Static" but never goes to "None" which is suspect is why occupancy value remains "1" and not changes to "0". Force occupancy to "0" which clears it in HA, but it never fires again until you remove the batteries and the cycle starts all over with it stuck with presence detected.

cobirnm, 2024-05-26:
That is strange. Mine has been verry stable and reliable. Right now it's not at it's final location but rather in a test position and it detects presence with a verry arcuate degree:

[image]

These are the settings that I have:
[image]

[image]
[image]

BambamNZ, 2024-05-27:
Based off your shared config @cobirnm - Thank you. 
I've been able to make some progress, been able to get it clear presence automatically, but the time it takes to clear is still strange. It takes between 5 and 15 minutes to clear from when I leave the room, it works just not very predictably right now.

tipofthesowrd, 2024-05-29:
I have the same issue as above. The occupancy state (not motion) is being cleared and detected but the clearing time takes a long time. The time to clear seems to be unrelated to the fading time. 

Attributes can be written but seem to have no effect. After removing the battery to reboot the device the attributes return to the default state.

vinzent, 2024-05-29:
i've ordered a tuya zigbee gateway to test the device with the tuya app and see if it works there as expected.

vinzent, 2024-06-04:
large rewrite. i've got the tuya  zigbee gatewaay and i was able to access the IOT Dev thing (iot.tuya.com) to discover  the datapoints:

https://gist.github.com/vinzent/2cd645b848fd3b6a0c3e5762956ec89f

still need to figure out how to implement it as a quirk v2 with exposed config entries. (https://github.com/zigpy/zha-device-handlers/pull/3019)

lukasdchass, 2024-06-21:
Thanks a lot for working on this vinzent. Is there any use for the "HumanPresenceSensorManufCluster" on 0xef00? It has loads of settings but I haven't seen it mentioned, but it only appears for me after installing your quirk. 

I run zigbee2mqtt on the side for testing and compatibility and haven't been able to get this device working in ZHA as reliably as MQTT sadly. Did you have any plans to continue with your quirk or do you have it working perfectly? Would love to know what settings you have after your V7 as some of the names changed I think as you introduced new settings? Cheers!

digitaltopo, 2024-08-17:
I can't seem to get the quirk to work.

I've deleted and re-added the device several times and tried reconfiguring several times. HA logs report that quirks are loaded. Anything I can try?

[image]

thefunkygibbon, 2024-09-02:
So did anyone get this device working correctly?  the quirk at the top that was posted doesn't seem to be picked up by zha for the device (have restarted HA and re-added the device).  
trying the other script on the gist gives me a little more but all of the "config"  aspects don't have names and are blank,  I tried to change one to a value and it bizarrely killed my ZHA (had to remove the quirk and reboot before it would successfully initialize again.

thefunkygibbon, 2024-09-02:
> I've updated https://gist.github.com/vinzent/2cd645b848fd3b6a0c3e5762956ec89f
>
> seems the IlluminanceMeasurement is binding to the 0x0000 measured_value too and does some calculations. whereas the [...]

not sure if you've tested your latest one, but my ZHA just doesn't start if i have it in my quirks folder at all.

vinzent, 2024-09-02:
> not sure if you've tested your latest one, but my ZHA just doesn't start if i have it in my quirks folder at all.

latest does just work with >= HA 2024.08 due to api changes. 

if you have that, you might provide error messages.

thefunkygibbon, 2024-09-02:
> > not sure if you've tested your latest one, but my ZHA just doesn't start if i have it in my quirks folder at all.
> 
> latest does just work with >= HA 2024.08 due to api changes.
> 
> if you have that, you might provide error messages.

yes i'm on the latest HA. I've managed to get it working now and its not rebooting, although its not actually working.  I've re-paired the sensor, it doesnt imply its unavailable, but the lux value and occupancy never change 

[image]

BambamNZ, 2024-09-02:
> > > not sure if you've tested your latest one, but my ZHA just doesn't start if i have it in my quirks folder at all.
> > 
> > 
> > latest does just work with >= HA 2024.08 due to api changes.
> > if you have that, you might provide error messages.
> 
> yes i'm on the latest HA. I've managed to get it working now (as in that it is not rebooting the zha constantly), but its not actually working correctly. I've re-paired the sensor, it doesn't imply its unavailable, but the lux value and occupancy never change
> 

Must admit, I gave up after weeks of trying and struggling, the quirk really helped to get things going, however the sensor was just not reliable. Mine would get stuck on "Occupancy Detected"  with no state changes for hours and even days. Had to power cycle (remove batteries) every couple of days.

Found this Zigbee [motion Sensor](https://www.aliexpress.com/item/1005006704018593.html?spm=a2g0o.order_list.order_list_main.10.53311802zeL0lf) that is doing just fine for the application I was intending to use this one.

thefunkygibbon, 2024-09-03:
> Must admit, I gave up after weeks of trying and struggling, the quirk really helped to get things going, however the sensor was just not reliable. Mine would get stuck on "Occupancy Detected" with [...]

ah, yeah thats just PIR,  the whole point in using this one was it was Radar with PIR as an "assist" to help it with battery life.  

tbh i've had nothing but problems with any radar based (bloody tuya) devices , so far this is the 3rd type i've tried.   and yet all 3 seem to apparently work great in z2m.  i really am in two minds whether to just bite the bullet and move everything over to z2m, despite it being a gigantuan task.

lukasdchass, 2024-09-03:
Just wanted to inject a bit of hope here with this sensor! It took me weeks of trying different configurations of detection distance, detection sensitivity, and motion detection sensitivity, but I finally got this working perfectly - no false positives, always on when I'm in the room no matter how still I am, etc. It was just about fiddling with those 3 numbers. Mine are 3, 8, and 5 respectively for a small 2 meter by 2 meters ish bathroom.

One thing I did resort to was cleaning and lightly re-soldering the connections for the PIR and RADAR modules after seeing this mentioned elsewhere. Not sure that helped though as I still had issues until I stumbled upon the right combination of settings. Before I got it right my wife described it as "the bathroom is having disco time" with it going on and off! Battery life has also been amazing which is worth noting considering it's a more complex device than a simple PIR.

Finally, one thing I'd recommend is setting the fading time as low as possible (10 seconds) and then just do your fading time manually via the off automation you create, ie: IF: Presence sensor is none FOR 30 seconds THEN turn of lights. Somehow that fading time was always wrong no matter what you set it to. So I decided to take that out of the equation as far as possible.

It's certainly not the best product experience I've had but it's very satisfying once it works. Keep slogging!

lukasdchass, 2024-09-03:
ps: thanks a lot to @vinzent for your hard work on the quirk !

petarlaf, 2024-09-17:
> Just wanted to inject a bit of hope here with this sensor! It took me weeks of trying different configurations of detection distance, detection sensitivity, and motion detection sensitivity, but I [...]

is this on ZHA ? Or MQTT?

thefunkygibbon, 2024-09-17:
I would assume zha given the GitHub page we are in.   would be nice if he shared his working quirk though.

nerumo, 2024-09-17:
@lukasdchass congrats on your success. You mentioned your magic three numbers: 3,8 and 5. 

Do you mean:
- 0x0123, motion_detection_sen: 8
- 0x0003, near_detection: 5
- 0x0002, sensitivity: 3

What values are you using?

petarlaf, 2024-09-17:
> I would assume zha given the GitHub page we are in. would be nice if he shared his working quirk though.

I'd hope so, but I just can't see how what he is saying would work with ZHA atm - unless he has a separate quirk as you say.

Brunsben, 2024-11-08:
Hello, has anyone managed to create a quirk that can be used to set the devices? 
I've tried a few of the ones listed here, but they all don't seem to work. 
For example, I don't have an entry with “TuyaOccupancySencing”

fuatakgun, 2024-12-14:
I have the same one (_tze200_kb5noeto) and this configuration has been working perfectly after long battle.

```
{
    "fading_time": 10,
    "indicator": "ON",
    "motion_detection_mode": "only_pir",
    "motion_detection_sensitivity": 4,
    "static_detection_distance": 5,
    "static_detection_sensitivity": 4
}
```

Brunsben, 2024-12-18:
> I have the same one (_tze200_kb5noeto) and this configuration has been working perfectly after long battle.
>
> ```
> {
>     "fading_time": 10,
>     "indicator": "ON",
>     "motion_detection_mode": [...]

Where should I use this configuration? 
I can't find a place where I can include it. 
And do you also use a custom quirk?

fuatakgun, 2024-12-18:
> Where should I use this configuration? 
> I can't find a place where I can include it. 
> And do you also use a custom quirk?

These are device settings to have proper presence/motion state, but I assumed that the device has been recognized by zha. I am using z2m, so no need for a quirk there.

mopahle, 2025-01-08:
I used the following quirk to add some configurations:
```
import zigpy.types as t
from zigpy.quirks.v2 import EntityPlatform, EntityType
from zigpy.quirks.v2.homeassistant import UnitOfTime,   UnitOfLength
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaMotionDetectionMode(t.enum8):
    only_pir = 0x00
    pir_and_radar = 0x01
    only_radar = 0x02
(
    TuyaQuirkBuilder("_TZE200_kb5noeto", "TS0601")
    .tuya_number(
        dp_id=102,
        attribute_name="fading_time",
        translation_key="fading_time",
        fallback_name="Fading time",
        entity_type=EntityType.CONFIG,
        type=t.uint16_t,
        device_class=SensorDeviceClass.DURATION,
        unit=UnitOfTime.SECONDS,
        min_value=10,
        max_value=28800,
        step=1
    )
    .tuya_number(
        dp_id=4,
        attribute_name="static_detection_distance",
        translation_key="static_detection_distance",
        fallback_name="Static detection distance",
        entity_type=EntityType.CONFIG,
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0,
        max_value=10,
        step=0.01,
        multiplier=0.01
    )
    .tuya_number(
        dp_id=2,
        attribute_name="static_detection_sensitivity",
        translation_key="static_detection_sensitivity",
        fallback_name="Static detection sensitivity",
        entity_type=EntityType.CONFIG,
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1
    )
    .tuya_enum(
        dp_id=122,
        attribute_name="motion_detection_mode",
        enum_class=TuyaMotionDetectionMode,
        translation_key="motion_detection_mode",
        fallback_name="Motion detection mode",
        entity_type=EntityType.CONFIG,
        entity_platform=EntityPlatform.SELECT
    )
    .skip_configuration()
    .add_to_registry()
)

The datapoints are captured from the [herdsman converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts). The are more Datapoints to add...

rdehuyss, 2025-01-26:
Any chance of getting a Tuya Quirk also for this device in the next awesome release?

Thanks to the whole ZHA team for adding support for all those new Tuya devices!!! 🙇 👏 💌

marvelloard, 2025-02-01:
That would be awesome.

rdehuyss, 2025-02-08:
I'm using [this quirk](https://gist.github.com/vinzent/2cd645b848fd3b6a0c3e5762956ec89f) for now. Would a PR to add it help?

PR #3886 (2025-02-26): Add Tuya Motion ZG-204ZM `_TZE200_kb5noeto`
## Proposed change

Adds Tuya PIR 24Ghz human presence sensor, ZG-204ZM. `_TZE200_2aaelwxk` and `_TZE200_kb5noeto`

Based on https://github.com/Koenkk/zigbee-herdsman-converters/blob/6c3d7e2aa1fba6f6342baea21e5b3bbd3688cb0e/src/devices/tuya.ts#L8858

## Additional information

Closes: #3125

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-02-26 by TheJulianJES