zigpy/zha-device-handlers #3127: [Device Support Request]  Aqara Z1 Pro ZNQBKG45LM

HarvsG, 2024-04-28:
### Problem description

This device doesn't seem to be supported

### Solution description

Support to include, on off, different button presses, dimming, coupled vs uncoupled mode, and ideally bindings

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

kuminda, 2024-08-25:
I look forward to see ZHA integration for this device as I want to use the slider function

KwFung7, 2025-04-09:
Please assist with supporting ZHA integration for the Z1 Pro.
Actually I’d like to contribute to this community, but I’m unsure where to begin. Thank you so much!

KwFung7, 2025-04-09:
Found more references online:
[Example configuration from zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt/issues/21687)
[Model Info](https://opendoc.aqara.cn/docs/SDK%E5%BC%80%E5%8F%91/Android%E5%BC%80%E5%8F%91%E6%8C%87%E5%8D%97/%E8%AE%BE%E5%A4%87%E6%8E%A7%E5%88%B6/%E8%AE%BE%E5%A4%87%E6%8E%A7%E5%88%B6SDK%E6%94%AF%E6%8C%81%E8%AE%BE%E5%A4%87%E6%B8%85%E5%8D%95.html)

brofeelgood, 2025-04-12:
May I know if these switches are already recognised in ZHA and if basic functions like on/off are working?

KwFung7, 2025-04-15:
> May I know if these switches are already recognised in ZHA and if basic functions like on/off are working?

Yes, the on/off function work for me, but the slider entity didn't show up and the power consumption is always zero

[image]

brofeelgood, 2025-04-15:
> > May I know if these switches are already recognised in ZHA and if basic functions like on/off are working?
> 
> Yes, the on/off function work for me, but the slider entity didn't show up and the power consumption is always zero

I just found a [Youtube review](https://www.youtube.com/watch?v=C0XdbgY33es&t=833s) and it more or less corresponds to what you've shared. He did mention, however, that the power readings only come on after a minute or so.

I hope someone picks this up and works out a solution for it. Without the slider, this is just a Z1 non-pro switch.

oxelwhy, 2025-05-18:
I'm stuck in step4, unable to see the Quirk. My model is lumi.switch.acn059. Any idea?

KwFung7, 2025-05-18:
@oxelwhy Did you find any log or error in **/config/home-assistant.log** related to **AqaraZ1ProTripleRockerSwitch**

`2025-05-17 14:46:46.169 DEBUG (SyncWorker_4) [switch_z1_pro_triple] AqaraZ1ProTripleRockerSwitch quirk module is being loaded! ZHA Profile ID: 0x0104, Device Type: 0x0000`

[image]

oxelwhy, 2025-05-18:
Nope, did not see any error. All I found is this, 
2025-05-18 13:55:23.373 DEBUG (SyncWorker_3) [aqara_z1_pro_double] AqaraZ1ProDoubleRockerSwitch quirk module is being loaded! ZHA Profile ID: 0x0104, Device Type: 0x0000
2025-05-18 13:55:23.378 DEBUG (SyncWorker_3) [aqara_z1_pro_quadruple] AqaraZ1ProQuadrupleRockerSwitch quirk module is being loaded! ZHA Profile ID: 0x0104, Device Type: 0x0000
2025-05-18 13:55:23.383 DEBUG (SyncWorker_3) [aqara_z1_pro_single] AqaraZ1ProSingleRockerSwitch quirk module is being loaded! ZHA Profile ID: 0x0104, Device Type: 0x0000
2025-05-18 13:55:23.390 DEBUG (SyncWorker_3) [aqara_z1_pro_triple] AqaraZ1ProTripleRockerSwitch quirk module is being loaded! ZHA Profile ID: 0x0104, Device Type: 0x0000

KwFung7, 2025-05-18:
@oxelwhy How about this debug log ? if it doesn't show up then it prove the custom quirk was not being loaded properly.
`2025-05-18 17:05:08.061 DEBUG (MainThread) [switch_z1_pro_triple] AqaraZ1ProTripleRockerSwitch device initialized with IEEE:`

Besides you could try to compare your z1 provided endpoint in "Manage Zigbee device" with the signatures in added python.
[image]

**My "Manage Zigbee Device" signatures:**
```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.switch.acn058",
  "class": "switch_z1_pro_triple.AqaraZ1ProTripleRockerSwitch"
}
```

KwFung7, 2025-05-18:
@oxelwhy I owned the single, double and triple rocker version but I don't have any quadruple version (acn059), that's why I just simply guessed the signature endpoint with the pattern I observed in other version. Perhaps there is something special in the quadruple version.

Could you send me the available signatures in your quadruple version so that I could help to cross check with my [aqara_z1_pro_quadruple.py](https://gist.github.com/KwFung7/4eeb4190cdabebc340c15a183a1ea928#file-aqara_z1_pro_quadruple-py)

[image]

oxelwhy, 2025-05-18:
I see, Then not sure apart from slider, will the wireless switch works as well. 

Anyhow the signature as below. 
```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.switch.acn059",
  "class": "zigpy.device.Device"
}
```

KwFung7, 2025-05-18:
@oxelwhy Done, I just found the misalignment, kindly copy the [aqara_z1_pro_quadruple.py](https://gist.github.com/KwFung7/4eeb4190cdabebc340c15a183a1ea928#file-aqara_z1_pro_quadruple-py) again and then restart your Home Assistant.

_**Changes:**_
[image]

oxelwhy, 2025-05-18:
Ok, tested your updated copy as well. It works. Thx much.

Btw, just out of curiosity, will the Sensors works as well?

[image]

KwFung7, 2025-05-18:
It does not work for me at the beginning and I'm not sure how to make it work, actually I'm not that familiar with ZHA, simply followed existing quirk to develop the script for z1

oxelwhy, 2025-05-18:
No worry, thanks for your contribution and able to use slider. Will see if the wireless switch now able to work, so far single click only, not sure if double click etc will able to work.

oxelwhy, 2025-05-25:
1. On the slider, how do you configure the slide up and slide down for automation? I want to archieve I.e. to set shade like 50% (drag bottom to middle), 25% (drag quarter from bottom) etc. Seems need a complicated or advanced script.
2. On the wireless switch, I notice Zigbee2mqtt can support single, double and hold. Current only support single, are you able to port it ?

HarvsG, 2025-09-26:
Anyone managed to enable or disable decoupled or detached mode?