zigpy/zha-device-handlers #3129: Tuya_ TS0601_TZE200_adquvgsi

nonamelee1003, 2024-05-01:
### Problem description

請求支持Tuya 人體存在傳感器含光照設備，Tuya_ TS0601_TZE200_adquvgsi

### Solution description

新增後能顯示人體存在及光照感應，並設備距離及敏感度

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2024-05-01:
Pleas update the issue with English.

andersje, 2024-05-01:
problem description:
Request support for Tuya human presence sensor with lighting device, Tuya_ TS0601_TZE200_adquvgsi

solution description:

After adding, it can display the presence of human body and light sensor, as well as the device distance and sensitivity.

closed 2024-11-07 by github-actions[bot]