zigpy/zha-device-handlers #3132: [Device Support Request] _TZE200_p2qzzazi - TS0601

slothking87, 2024-05-02:
### Problem description

Sold as a zigbee blind driver from Kogan (https://www.kogan.com/au/buy/kogan-smarterhometm-zigbee-smart-blinds-driver-kogan-bndl/). 

Have tried to add the model number into the existing quirk, but that hasnt helped. I can view buttons in HA once added, but the bottons dont function - 
Error Failed to call service cover/close_cover. Failed to send request: Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>

### Solution description

To be able to use the basic up, down and stop functions of the blind driver

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_p2qzzazi",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 26210,
    "manufacturer": "_TZE200_p2qzzazi",
    "model": "TS0601",
    "name": "_TZE200_p2qzzazi TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4098,
    "power_source": "Battery or Unknown",
    "lqi": null,
    "rssi": null,
    "last_seen": "2024-05-02T19:51:40",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_p2qzzazi",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "update.tze200_p2qzzazi_ts0601_firmware",
        "name": "_TZE200_p2qzzazi TS0601"
      }

```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

slothking87, 2024-05-05:
I got it working fine with Z2MQTT with a few adjustments to an existing script

praetoros, 2024-05-24:
Hey @slothking87 could you please share some details for the setup. Looking to do the same.

slothking87, 2024-05-25:
> Hey @slothking87 could you please share some details for the setup. Looking to do the same.

I was able to use an existing definition and adjust the model number and it worked. I have loaded my external definition JS file [here](https://gist.github.com/slothking87/472127a1e34a05afb1923891bd80a226)

praetoros, 2024-05-27:
thanks a bunch @slothking87!

closed 2024-11-30 by github-actions[bot]