zigpy/zha-device-handlers #3135: IKEA STYRBAR with FW 2.4.11 needs a new quirk

marianhlavac, 2024-05-05:
### Bug description

IKEA has pushed a new OTA FW **v2.4.11** (`0x02040011`) on 3rd May 2024 [^1] for their STYRBAR remote controls, which removes a lot of functionality, which was previously working in Home Assistant.

**tl;dr:** L+R buttons stopped working, Zigbee groups stopped working, the device signature changed -> quirks are not being applied, but even when the current V2 quirk is applied, the buttons doesn't have the same behaviour as previously.

---

What stopped working (at least what I've noticed):

- Left & Right short presses are not reported as events anymore (which makes four-button remote into two-button effectively).
- Zigbee groups stopped working completely, apparently – after the update, all LightLink groups stopped working (that's how I noticed something's up) and I was unable to make it work again. Only direct 1-1 binding works, allows multiple devices to be linked, but for some reason it feels very buggy and unreliable (not all lights turns on/off and for some, dimming does not work – but that's not related). Reminds me the situation with the firmware updates for different IKEA buttons, as described in [one of the threads](https://community.home-assistant.io/t/ikea-tradfri-on-off-group-binding-firmware-versions/580345).
- Left & Right long presses reported as [_turn on_ event, together with _recall_ event](https://github.com/zigpy/zha-device-handlers/assets/3010503/d5280d3f-40cf-457d-8a24-1da8407fa904) (I don't remember the original behaviour, but I feel like this was not happening before).

The device signature has changed and the current quirk (`zhaquirks.ikea.IkeaTradfriRemoteV2`) is not applied. It seems that there is now `0x0004` cluster included, and `0xFC57` is missing from the signature.

I tried to update the signature in the quirk and it got applied fine, but even with the V2 quirk active, the remote doesn't have the same behaviour as previously.

[^1]: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

### Steps to reproduce

1. Keep OTA enabled and let your STYRBAR remotes to be updated to 0x02040011
2. Watch how basically everything stops working overnight for no reason

### Expected behavior

-

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "Remote Control N2",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

[... 370 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {
              "0x0002": {
                "attribute_name": "current_file_version",
                "value": 33816593
              }
            },
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

MattWestb, 2024-05-05:
Did you deleting the device and waiting one minute (or restart HA) before adding it so the system is adding its as new and not using the old cluster settings in the DB ?
I have looking little then its 6 updates in TF feed but most is already in the Dirigera but not all so must updating the firmware matrix.

marianhlavac, 2024-05-05:
Yes, numerous tries to reconfigure or remove/re-add, it didn't help. I have even tried it on a clean instance of Home Assistant, just to be sure.

> I have looking little then its 6 updates in TF feed but most is already in the Dirigera but not all so must updating the firmware matrix.

I don't have any experience with how the firmware works, is there anyone in the community who contributed the current quirks and would be able to help with the new firmware?

---

On the side note, is it possible to downgrade the current remotes or I'm stuck with this unsupported version?

Thanks!

MattWestb, 2024-05-05:
All IKEA controllers i have trying and also in the Zigbee certificate its saying they is protected for roll back the firmware only light is not and can being done (for getting the magic IKEA scene).
I have trying updating my Styrbar and Vallhorn but its looks like Dirigera feed (that ZHA is using) is not having it also tretakt is missing.
Have moving one Styrbar to TF and updating it and can taking on look on it then its done and back in one of my test system.

MattWestb, 2024-05-05:
Add the `Groups.cluster_id`, in the signature and  in the replacement also need adding it in import as `Groups` with the other normal cluster inport.
Deleting the `WWAH_CLUSTER_ID,` and it shall working and loading the quirk and the device automatons.

chrizzlibaer, 2024-05-05:
I'm on the other side over there @ zigbee2mqtt and the remote also stopped working for left and right. Just to say it's not only related to Zha... 🙄

MattWestb, 2024-05-05:
We is knowing if they have doing some more changes tomorrow then i have getting one updated and can fixing the quirk for it.

marianhlavac, 2024-05-05:
> Add the Groups.cluster_id, in the signature and in the replacement also need adding it in import as Groups with the other normal cluster inport.
Deleting the WWAH_CLUSTER_ID, and it shall working and loading the quirk and the device automatons.

Yes, I tried this, unfortunately, as I mentioned in the issue, even having the quirk working doesn't help with LightLink not working and L+R buttons not reporting events.

MattWestb, 2024-05-05:
I can being needed sniffing how TF and DG is setting up the device if they have changing the function of the scene cluster that i easy can do then have getting it updated and then see what is needed for getting working OK.

MattWestb, 2024-05-05:
Local quirk that loading OK and have the same functionality as V2 (long release up and down is the same). Tale the .TXT away and punting in the local quirk folder
[Styr3.py.txt](https://github.com/zigpy/zha-device-handlers/files/15214228/Styr3.py.txt)

JohnBull-1975, 2024-05-05:
I had the 2.4.11 update a few days ago. Only got it for one of two STYRBAR...so the one still on 1.0.024 still works.
I thought the updated worked after update, but I only tried on/off (up/down).
This evening my son reported left-right didn't work.
I started troubleshooting and excluded it from Z2MQTT  (in HA) and after that no button on that device works!
I tried exclude and include several times. Last seen status updates on every include. But no response in MQTT-console when hitting the buttons.

I tried to delete it with the Force Exclude option, Full rebooted of HA and then Permit all join and hitted the link button x4 and after that it shows as Offline.
Can still get it to leave the network by hitting the link button once or by resetting it by link button x4 and then get it to re-join. Still Offline and no response from buttons.

Tried the trick to direct link it to a bulb with no success. It fails to link.
Also discussed and pasted some logs here:
https://github.com/Koenkk/zigbee2mqtt/issues/18515

Think I'll try to take it back to IKEA tomorrow and see if they give me another and see what firmware that is on and if it works...

JohnBull-1975, 2024-05-05:
Tried the re-configre button and got this in the log:
Info 2024-05-05 22:14:14Configuring '0x287681fffeecf0d8'
Error 2024-05-05 22:14:22Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)
Error 2024-05-05 22:14:22Unparsed frame 0xc4. Skipped
Error 2024-05-05 22:14:24Failed to configure '0x287681fffeecf0d8', attempt 4 (Error: {"address":48854,"clusterId":32801,"sequence":224} after 10000ms at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:727:43) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:567:13 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:464:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:147:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:160:21) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Unfortunately it says me nothing...this is far beyond my knowledge...

MattWestb, 2024-05-05:
If re config is not working take out the battery and putting it back and start the re configure is normally getting it reviving the commands and start working (if some early commands is timing out then its one must for getting it reporting OK).

JohnBull-1975, 2024-05-05:
> If re config is not working take out the battery and putting it back and start the re configure is normally getting it reviving the commands and start working (if some early commands is timing out then its one must for getting it reporting OK).

I have tried unplugging batteries and re-inserted them several times.
Tried agan:
Info 2024-05-05 22:24:40Configuring '0x287681fffeecf0d8'
Info 2024-05-05 22:24:47Successfully configured '0x287681fffeecf0d8'
Still Offline and no response from buttons

MattWestb, 2024-05-05:
>If re config is not working take out the battery and putting it back **and** start the re configure is normally getting it reviving the commands and start working (if some early commands is timing out then its one must for getting it reporting OK).

You was **not** reading and is still using Z2M so not one ZHA problem.

JohnBull-1975, 2024-05-05:
but now it updated "last seen" when I clicked up...tried again and it updated...waited 20 sec and pressed left and it updated "last seen" for left click as well...

But still Offline and couldn't see it as device in MQTT integration i HA
Tried to exclude (link button x4) and inlcude (permit join all + link button x4) and then it came back as device.
Renamed it and came back as device in HA and now Online...but still no response from buttons...

Was this thread ZHA specific?

MattWestb, 2024-05-07:
Have updating one more test network (with RCP coordinator) and with quirk V3 group binding is working well also if bind the scene cluster its looks like IKEAs magic scenes is working default so no need adding the magic group and adding scene is its OK with the default ones.
Have not tested device binding but i think it shall working the Dirigera i using only unicast and routing them to groups in the hub = very bad done !!).

Can some more testing the quirk and i can posting on PR for getting it in next HA release ?

marianhlavac, 2024-05-09:
> Can some more testing the quirk and i can posting on PR for getting it in next HA release ?

@MattWestb If you could open the PR, I will test the V3 quirk right away. Thanks!

MattWestb, 2024-05-10:
PR posted

dlbogdan, 2024-05-17:
Just so I can have some rest, is this firmware (2.4.11) better or worse in terms of zigbee groups binding compared to 2.4.5.
I have one styrbar on 2.4.5 with no group binding capability and was hoping 2.4.11 is going to fix this. I tried my best to upgrade it but can't. I even upload the firmware file myself in home assistant and setup my config file as below: 
`zha:
  zigpy_config:
    ota:
      allow_file_providers: "I understand I can *destroy* my devices by enabling OTA updates from files. Some OTA updates can be mistakenly applied to the wrong device, breaking it. I am consciously using this at my own risk."
      otau_directory: /config/zig-ota
      ikea_provider: true # Auto update Trådfri devices
      Ikea_provider: http://fw.ota.homesmart.ikea.net/feed/version_info.json`

the firmware file /config/zig-ota/styrbar_dimmer_zingo-0x11cb-2.4.11-prod.ota.ota.signed was downloaded from the official ikea ota web service. 

I tried triggering the styrbar with calling its ota cluster with image_notify command. 
In the logs it just says that no firmware is found compatible with the device or something like that.

Help!

fsievers, 2024-05-31:
The above mentioned PR has been superseded by #3175

Nezz, 2024-07-31:
FYI: Ikea released a new firmware that should fix many if the issues introduced in 2.4.11:
https://www.reddit.com/r/tradfri/comments/1e4lq3f/styrbar_firmware_version_2416/

Sadly the update was not published in the usual place.

MattWestb, 2024-07-31:
If configures ZHA OK you shall getting izt then its one of the IKEA feed  ZHA is using.
All my Styrbars is updated for some week ago without problems and the "old" V2 quirk is working OK with them.

uvok, 2024-10-12:
I have an Ikea Styrbar with firmware 0x02040016.

No quirks are applied. Left/right buttons are not reported in the logbook.

Home Assistant OS is @ 2024.10.01.

[image]

I also had to try pairing the device several times (got stuck in Interview Done. Pairing), also reconfiguring several times, while pressing buttons.

uvok, 2024-10-15:
> @uvok how did you get this firmware, did ha offered it? I have a younger version but also with that one the left and right buttons do not work. Did you get it working already?

I had to manually send an "image_notify" command to the ota cluster, as described in e.g. https://www.thing-king.de/zigbee-geraete-ueber-home-assistant-bzw-zha-updaten-ota-update/ section "OTA-Update manuell triggern".

Basically, open the device page in HA, click the three dots, "manage...", select cluster "ota", go to "tab" commands, select command "image_notify", "QueryJitter", set slider to 100, and click the "send command" button. After that, HA should offer the update.

Mind you, the update took 3 hours for me. Be sure to push a button, e.g. left/right, every few minutes, otherwise the remote might go to sleep and the firmware update might fail.

johnkjellberg, 2024-11-09:
I had the same problem after I upgraded to 0x02040016. I fixed it by deleting my button and then re-pair it. Now all buttons works as expected.

closed 2025-08-13 by github-actions[bot]