zigpy/zha-device-handlers #3141: [Device Support Request] Avatto 1/2-ch dimmer. TS0601 _TZE204_o9gyszw2/_TZE204_5cuocqty

gruzzy, 2024-05-08:
### Problem description

I bought a couple of devices from AliEexpress (https://www.aliexpress.com/item/1005006524716136.html)

Devices were added to Home Assistance server without issues, but didn't expose any controls. After reading documentation, I have madew a custom quirk and devices are working fine now.

### Solution description

ZHA logs suggest to contribute changes into the main branch.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]
[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk1</summary>
class TuyaSingleSwitchDimmerGP(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_3p5ydos3", "TS0601"),
            ("_TZE200_ip2akl4w", "TS0601"),
            ("_TZE200_vucankjx", "TS0601"),  # Loratap
            ("_TZE200_y8yjulon", "TS0601"),
            ("_TZE204_n9ctkb6j", "TS0601"),  # For BSEED switch
            ("_TZE204_5cuocqty", "TS0601"), #Avattto ZDMS16-1
        ],
</details>

<details><summary>Custom quirk2</summary>
class TuyaDoubleSwitchDimmerGP(TuyaDimmerSwitch):
    """Tuya double channel dimmer device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_fjjbhx9d", "TS0601"),
            ("_TZE200_gwkapsoq", "TS0601"),  # Loratap
            ("_TZE204_zenj4lxv", "TS0601"),
            ("_TZE204_o9gyszw2", "TS0601"), #Avattto ZDMS16-2
        ],
</details>

NH-Networks, 2024-06-05:
@gruzzy  are custom quirks complete?

seem to miss information? can you send me files or give full code ?

gruzzy, 2024-06-05:
Sorry, I am not good in creating proper tickets yet :)

I found that extending the MODELS_INFO lists in tthe class signatures made devices working in HA.

One switch works as a TuyaSingleSwitchDimmerGP devices, another one is running as TuyaDoubleSwitchDimmerGP.

So, there are the only changes in the file **zhaquirks/tuya/ts0601_dimmer.py**:

> class TuyaSingleSwitchDimmerGP(TuyaDimmerSwitch): """Tuya touch switch device."""
> signature = {
>     MODELS_INFO: [
>         ("_TZE200_3p5ydos3", "TS0601"),
>         ("_TZE200_ip2akl4w", [...]

> class TuyaDoubleSwitchDimmerGP(TuyaDimmerSwitch): """Tuya double channel dimmer device."""
> signature = {
>     MODELS_INFO: [
>         ("_TZE200_fjjbhx9d", "TS0601"),
>         ("_TZE200_gwkapsoq", [...]

NH-Networks, 2024-06-06:
Yeah i figured it out, my bad.. will created a pull request for change, as no one is assigned to this yet.

NH-Networks, 2024-06-06:
Pull request created. lets see if it will be picked up.

vrtsman, 2024-06-16:
does not work for me. How can I see the quirk is in place? I can see it in the studio code server environment and edit the py, but does it get loaded correctly?

coen1111, 2024-11-09:
I see support for the on/off functionality is implemented in [3190](https://github.com/zigpy/zha-device-handlers/pull/3190). However, the Dataopoints in the Tuya manufacturing cluster are not added. Which does not enable changing the External Switch-Type as required by my application. An example implementation is present in Zigbee2MQTT [tuya.ts](https://github.com/RafaelDuncan/zigbee-herdsman-converters/blob/3d43ed3cee2606fd4ab517e215ecf9059b4cc7d8/src/devices/tuya.ts#L1614). I want to extend this support to ZHA. I tried making a custom quirks but did not succeed. Using the zha_toolkit I now try to create the command manually, however I did not succeed yet.

This is my action:

```
action: zha_toolkit.zcl_cmd
data:
  ieee: a4:c1:38:dc:2b:c5:a5:61  # Replace with your device's IEEE address
  endpoint_id: 1                 # The endpoint ID of the device, usually 1 for most devices
  cluster: 61184                 # 0xEF00 in decimal
  cmd: 4                         # The command ID for Send DP
  args:
    - 4                          # DP ID for External Switch-Type (check device specifics)
    - 0                          # Data type (0 for raw/unknown, may vary)
    - 2                          # Data value (set to 2 for Momentary)
```

What am I overlooking?

AOskam, 2025-01-19:
Is there anybody who can help with this?

jthrilly, 2025-03-02:
@coen1111 - I'm encountering the same issue, and also wanting to change the switch type. Did you make any progress with this?

jthrilly, 2025-03-11:
Just incase anyone comes across this, I ended up temporarily pairing the device with zigbee2mqtt, which allowed me to change the switch type. this change persisted when i repaired with ZHA.

PooruTorie, 2026-11-12:
@jthrilly I have the same problem i will create a new TuyaQuirkBuilder quirk coule you supply all know datapoints of the device or send a link to the z2m implementation