zigpy/zha-device-handlers #3142: [BUG] Aqara T1 Temp/Humidity not reporting values

eriknn, 2024-05-08:
### Bug description

The device is recognized as model lumi.sensor_ht.agl02.
The device seems to pair correctly, but no new messages seems to be received. Debugging does say "Device seen".
All values remain at 0.
Reconfiguration gives green at "XiaomiPowerConfiguration", yellow/failed on the rest of the clusters.

### Steps to reproduce

Add device like usual...

### Expected behavior

Device should pair successfully and display values.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0403",
        "0x0405",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.sensor_ht.agl02",
  "class": "zhaquirks.xiaomi.aqara.sensor_ht_agl02.LumiSensorHtAgl02"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.5.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
[... 394 lines cut ...]
              "0x0002": {
                "attribute_name": "current_file_version",
                "value": 29
              }
            },
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

eriknn, 2024-05-13:
It seems that a firmware upgrade of my ConBee III solved the issue!

closed 2024-05-13 by eriknn

MrChunks, 2024-05-14:
In case anyone finds this, my old Conbee I has no current firmware updates and hasn't for a long time. I was able to pair this sensor with my radio, but it never got updates. I have now changed to a new Zigbee Coordinator and the sensor appears to be working fine. At least so far.