zigpy/zha-device-handlers #3148: [Device Support Request] AliExpress Curtain Bot TS0601 by _TZE200_cpbo62rn

RharoLP, 2024-05-12:
### Problem description

Hi, 
I bought this one from AliExpress, I can pair it to Home Assistant but I'm not getting any sensors or diagnostics to use to automate the opening and closure of the curtains.

https://www.aliexpress.com/item/1005006224025259.html?spm=a2g0o.order_list.order_list_main.5.38b81802yYV5zq

### Solution description

Fully functional in homeassistant with ZHA able to create automations

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cpbo62rn",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

amkrafie, 2024-06-20:
Did you figure this out ? or found an alternative ?

RharoLP, 2024-06-23:
Nothing so far, I was hoping to get good traction in here but nothing :(

closed 2024-12-27 by github-actions[bot]