zigpy/zha-device-handlers #3152: [Device Support Request]  TS0601_TZE284_cjbofhxw  PJ-1203 clamp power meter

zibijw23, 2024-05-14:
### Problem description

The energy meter doesn't display any measurements.

### Solution description

It's time for some new quirks

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_cjbofhxw",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

https://pl.aliexpress.com/item/1005005994777032.html?spm=a2g0o.order_list.order_list_main.5.51dc1c24SJTQMe&gatewayAdapt=glo2pol

mike-nani, 2024-05-22:
Hi! I'm very new to HA with level 0.000001 coding skills.
Recently ordered and received this device, although previous reviews show the TZE204 model. I was trying to build a custom quirk for it but can't figure out some details.
The "0xed00" and "0xef00" clusters seem to have no attributes, although one in each cluster appears on the full scan. Is it supposed to be like this? Are the correct attributes only visible after que quirk is applied? What am I missing?

No attributes shown in the "Manage Zigbee Device" section (same for "0xef00"):
[image]

Full ZHA Toolkit device scan:
```
zha_toolkit_version: v1.1.10
zigpy_version: 0.64.0
zigpy_rf_version: 0.38.4
ieee_org: update.tze284_cjbofhxw_ts0601_firmware
ieee: xx:xx:xx:xx:xx:xx:xx:xx
command: scan_device
command_data: null
start_time: "2024-05-22T08:54:19.201159+00:00"
errors: []
params:
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  read_before_write: true
  read_after_write: true
scan:
  ieee: xx:xx:xx:xx:xx:xx:xx:xx
  nwk: "0x55e1"
  model: TS0601
  manufacturer: _TZE284_cjbofhxw
  manufacturer_id: "0x4098"
  endpoints:
    - id: 1
      device_type: "0x0051"
      profile: "0x0104"
      in_clusters:
        "0x0000":
          cluster_id: "0x0000"
          title: Basic
          name: basic
          attributes:
            "0x0000":
              attribute_id: "0x0000"
              attribute_name: zcl_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 3
            "0x0001":
              attribute_id: "0x0001"
              attribute_name: app_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 74
            "0x0002":
              attribute_id: "0x0002"
              attribute_name: stack_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ|REPORT
              access_acl: 5
[... 295 lines cut ...]
            "0x0006":
              attribute_id: "0x0006"
              attribute_name: image_upgrade_status
              value_type:
                - "0x30"
                - enum8
                - Discrete
              access: READ|REPORT
              access_acl: 5
              attribute_value: 0
            "0xfffd":
              attribute_id: "0xfffd"
              attribute_name: cluster_revision
              value_type:
                - "0x21"
                - uint16_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 3
          commands_received: {}
          commands_generated: {}
success: true
```
Thanks!

MattWestb, 2024-05-22:
From the first post i see:
```
  },
  "manufacturer": "_TZE284_cjbofhxw",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```
So you dont have any quirk loaded and then its one tuya MCU device is no use t scanning it with ZHA-ToolKit then you is only getting the information of the standard Zigbee module and the interesting is for putting in the quirk signature but all magic information is from the MCU that you cant getting before getting the signature OK in the quirk.

lucianojss, 2024-06-04:
Hi, got this device also was anyone able to run this device in ZHA? 🙇

JHurk, 2024-06-11:
Tried to add this device through a custom_quirk but after a few hours of trial and error I gave up.
Would be really nice if someone could come up with support for this device through ZHA.
I have not been able to get it working with a custom quirk, somehow I can not get the custom_quirk to recognize the device.
I added the device code tze284_cjbofhxw as model info, in multiple quirks (ts0601_energy_meter.py and ts0601_din_power.py) all with no success.

So following this thread hoping support will be added to finally get this device to work with ZHA.

mike-nani, 2024-06-12:
I also spent several hours trying to create a custom quirk, using ts0601_energy_meter.py as a template. Even adjusting for the differences in clusters, I can't get it to recognize the device. Don't know what else to do...

JHurk, 2024-06-13:
Not really sure of this is helpful but looking at the PR I found this one:
https://github.com/zigpy/zha-device-handlers/pull/2961
It seems like a PR for a similar version of 'our' TZE284_cjbofhxw.
Anyone with a little bit more knowledge on how this works, who can tell me if it is worth connection our device to the PR?

nachogarcia, 2024-06-18:
I used the quirk I found in https://github.com/zigpy/zha-device-handlers/issues/1768 but needed to edit the signature of the 1CH, because it has an extra in put cluster (`0xed00`, I don't know which kind it is).
Disclaimer: I needed to restart it after adding it, and it always had power through the clamp.

```
class TuyaEnergyMeter_1CH(CustomDevice):
    """Tuya PJ-MGW1203 1 channel energy meter."""

    signature = {
        MODELS_INFO: [
            ("_TZE204_cjbofhxw", "TS0601"),
            ("_TZE284_cjbofhxw", "TS0601"),
        ], 
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51
            # device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaMCUCluster.cluster_id,
                    0xed00
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }
```

<details>

<summary>Full file to just paste it for now ;)</summary>

```python
"""Tuya Energy Meter."""

from enum import Enum
from typing import Any, Callable, Dict, Optional, Tuple, Type, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.foundation import ZCLAttributeDef

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaZBElectricalMeasurement,
    TuyaZBMeteringClusterWithUnit,
)
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

# from zigpy.zcl.clusters.homeautomation import MeasurementType

# Manufacturer cluster identifiers for device signatures
EARU_MANUFACTURER_CLUSTER_ID = 0xFF66

[... 1660 lines cut ...]
                    TuyaElectricalMeasurement,
                    TuyaMetering,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaElectricalMeasurement,
                    TuyaMetering,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

mike-nani, 2024-06-18:
Thanks @nachogarcia! Finally some progress! It's now applying the quirk and I can manually read the rms_current and rms_voltage attributes, which seem to present realistic values.
Still no entities, though...

nachogarcia, 2024-06-18:
> Thanks @nachogarcia! Finally some progress! It's now applying the quirk and I can manually read the rms_current and rms_voltage attributes, which seem to present realistic values. Still no entities, though...

Try reseting (I just cut the power to the whole house I think). This is mine.
[image]

apapalillo, 2024-06-19:
> I used the quirk I found in #1768 (edit: yeah, that PR @JHurk ) but needed to edit the signature of the 1CH, because it has an extra in put cluster (`0xed00`, I don't know which kind it is). Disclaimer: I needed to restart it after adding it, and it always had power through the clamp.
> 
> ```
> class TuyaEnergyMeter_1CH(CustomDevice):
>     """Tuya PJ-MGW1203 1 channel energy meter."""
> 
>     signature = {
>         MODELS_INFO: [
>             ("_TZE204_cjbofhxw", "TS0601"),
>             ("_TZE284_cjbofhxw", "TS0601"),
>         ], 
>         ENDPOINTS: {
>             # <SimpleDescriptor endpoint=1 profile=260 device_type=51
>             # device_version=1
>             # input_clusters=[0, 4, 5, 61184]
>             # output_clusters=[10, 25]>
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaMCUCluster.cluster_id,
>                     0xed00
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             }
>         },
>     }
> ```
> 
> Full file to just paste it for now ;)

Great! It's something worth! I can see the entities like yours now
Still no values read :(

It doesn't even seem to reset if I do the long press thing. It just doesn't blink as it should

LIrahara, 2024-07-07:
Hi,
Im still having a few issues getting anything working here. Still new to the back end of HA.
So I have done the custom quirks folder in the config, adjusted the configuration.yml to look at it and set it to enable quirks.
I copied the quirk file from #1768 and added the code from nachogarcia to it. Saved it in the quirks folder (didnt name it anything specific).
Removed the device from ZHA, restarted HA and re added the device.
Still nothing.
Can anyone advise where ive gone wrong here please?

infinitytec, 2024-07-12:
I added this as well and am seeing similar issues to what others have described. I am getting voltage and frequency, but that's it.

However, I am using one of the two channel ones of these, and I am seeing three copies of just about every entity:
[image]

TomasTal, 2024-12-23:
Hello, i bought an 2ch Electricity Energy Monitor from [Aliexpress](https://www.aliexpress.com/item/1005005881727281.html). 
Model PJ-1203A (TS060,  _TZE204_81yrt3lo)
input Voltage: 100-240V AC
Measuring Range 0.2-80A
Clamp: 2pcs
Protocol: ZigBee

**And your custom quirk worked form me too. THANK YOU.**

> ### Problem description
> The energy meter doesn't display any measurements.
> 
> ### Solution description
> It's time for some new quirks
> 
> ### Screenshots/Video
> Screenshots/Video
> ### Device signature
> Device signature
> ### Logs
> Logs
> ### Custom quirk
> Custom quirk
> ### Additional information
> It works for me
> 
> https://pl.aliexpress.com/item/1005005994777032.html?spm=a2g0o.order_list.order_list_main.5.51dc1c24SJTQMe&gatewayAdapt=glo2pol

ImpossivelPT, 2025-01-07:
I also had problems with a power meter (just 1 clamp) from aliexpress, you quirk works perfect, thanks a lot!

TS0601
by _TZE284_cjbofhxw

danielavdalian, 2025-01-19:
Can anybody add a Update frequency parameter? That will be a useful parameter. I have it in another meter and it works like a charm, but in this one, updates some times like every 10-8-15 or even 20 seconds....

IgorMilavec, 2025-02-13:
I can confirm that it works fine for _TZE204_81yrt3lo. @zibijw23 could you open a PR to get this merged into main?

closed 2025-08-21 by github-actions[bot]