zigpy/zha-device-handlers #3158: [Device Support Request] TS011F by _TZ3000_cicwjqth | Greensun Outdoor Plug | Tuya

wurmloch1337, 2024-05-17:
### Problem description

Received a new 16A Tuya plug today from the company "GreenSun" ordered on Amazon.
The default integration worked, except it showed the Current value by x1000.

I made it work with a custom quirk.
It looks nearly like a normal tuya plug, but the order is different and there is no 242 endpoint.

Unfortunately I am not able to get more readings than Power, Current, Voltage.
The product page states, that it should be capable of metering like power delivered.

### Solution description

Please have a look at my custom quirk and maybe find the error why it it not showing the other values you would expect.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x010a",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0702",
            "0x0b04",
            "0xe000",
            "0xe001"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZ3000_cicwjqth",
      "model": "TS011F"
    }
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
[... 649 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,                   
                    TuyaZBE000Cluster,
					TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id,Time.cluster_id],
            },
            
        },
    }
```

</details>

### Additional information

ordered here: https://www.amazon.de/dp/B0CYZL8TQH

ftreil, 2024-06-09:
Same plug not working for me. The plug is detected , added to network then disconnect and become unavailable.

Tell me if I can provide some info to help.

Regards.

jipem01, 2024-07-12:
Having same device.
I can add it to the network and I can get few entities : switch, voltage, current power and intensity.

But : 
    - Intensity should have a unit problem, as it display 845A rather than 0.845A.
    - there is no reporting of power summation.
    - it seems since last update (from 2024.6.1 to 2024.7.2) that the switch is no more working (i could not exclude problems from my radio but nothing else changed on the network).
 
I'm not familar with zigbee integration but if you need details/logs let me know i'll try to find them.

lennert-hd, 2024-09-21:
I'm having the same issue and the plug is also incapable of recieving OTA updates.

shrekars, 2025-04-25:
Not available in energy panel.

closed 2025-10-30 by github-actions[bot]