zigpy/zha-device-handlers #3161: [Device Support Request] Add IKEA Fyrtur roller blind remaining battery status

Eriond, 2024-05-20:
### Problem description

The IKEA Fyrtur roller blind uses a rechargable battery for which you may ask remaining charge percentage in the `DoublingPowerConfigClusterIKEA (Endpoint id: 1, Id: 0x0001, Type: in)`, using the attribute `battery_percentage_remaining (id: 0x0021)`
The current implementation seems to omit this information, because the reported battery level is always 0% in HA frontend/UI.

### Solution description

The current value that is read back, seem to be the percentage expressed in 0.5 percent steps, ie. doubled. The attached screenshot shows a recently charged battery.
It *may* also be so that the percentage is expressed in a 0-255 / byte-wide manner, but this has not been verified.

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0102",
        "0x1000",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "FYRTUR block-out roller blind",
  "class": "zhaquirks.ikea.blinds.IkeaTradfriRollerBlinds2"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.5.4",
    "dev": false,
[... 437 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Eriond, 2024-05-20:
So, here's an interesting fact: 
After finding the relevant attribute, playing around a bit in the "Manage zigbee device" settings, and finally writing this issue, somehow I got a new sensor in the "Diagnostic" list: **Battery 194%**
Previously, there were only the "Identify" and "Window covering type" items, and now I have miraculously gotten a third, much sought after, battery sensor!
For those that may think I have bad vision, I have attached  a screenshot of the history diagram.
[image]
Perhaps the issue can be renamed as "Correct the reported battery value" ?
For now, I have made a helper using a template sensor to circumvent the issue:
`{{ (states('sensor.fyrtur_roller_blind_sovrum_battery') | int)/2 }}`

MattWestb, 2024-05-21:
What is the firmware version the blinds running (diag is showing 2.2.009) ?
The factory is 2.3.X but the last production is 24.4.13 and after doing update you need deleting the device and waiting one minute and adding it  agen so its not using the old cashed settings (then its not using any quirk) and shall working OK then IKEA have fixing the old battery problem issue.
Was doing all yesterday with 3 new ones and its working great.

MattWestb, 2024-05-21:
[image]

Eriond, 2024-05-21:
Thank you for your comment. I don't know how to see the firmware update history/logs, but these are the current values:
[image]
In my configuration, I have
```
zha:
  zigpy_config:
    ota:
      ikea_provider: true
```
so it is possible that while I was fiddling with the various Zigbee clusters and attributes yesterday, it triggered an automated firmware in the background, and voila... Still odd that I have the doubled battery percentage though? Or will that be resolved if I remove and then re-add the device after a couple of minutes?

MattWestb, 2024-05-21:
The device have updating OK but its using the old cashed endpoints / cluster setting so loading the quirk for the old firmware (last line in the device info is showing the quirk in use).

You need deleting the device wait one minute (or restarting HA) and then pairing it and it shall being done as new (its loosing the name it was having) and shall working OK.

closed 2024-11-24 by github-actions[bot]