zigpy/zha-device-handlers #3165: [Device Support Request] VALLHORN Wireless Motion Sensor

fzakfeld, 2024-05-22:
### Problem description

Sensor is either stuck at "Configuring" (in Home Assistant) or - when configured - won't report any state (just shows the state from when it was added). Binding to a light surprisingly works, just doesn't report any actual state to HA.

### Solution description

Fix support for VALLHORN motion sensors.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0850",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0b05",
        "0x1000",
        "0xfc7c",
        "0xfc81"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0019",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0406"
      ],
      "output_clusters": [
        "0x0003"
      ]
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0106",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0400"
      ],
      "output_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "VALLHORN Wireless Motion Sensor",
  "class": "zigpy.device.Device"
}
```

</details>

[... 477 lines cut ...]
          "0x0400": {
            "endpoint_attribute": "illuminance",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 16812
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

danvolchek, 2024-08-16:
I'm also experiencing this

Fraighas, 2024-08-21:
Hi, Did you guys found something to solve this ?  I'm also experiencing this

danvolchek, 2024-08-21:
After reading a bunch of threads, it looked to be a problem with the firmware of my Zigbee dongle - that it was too old.

The one I had was using Texas Instruments based firmware, and after more reading, there didn't seem to be an easy way to update TI based dongle firmware easily.

Instead, I purchased an SI Labs based dongle. The firmware it shipped worked out of the box - I didn't even need to upgrade the firmware (I was going to use https://darkxst.github.io/silabs-firmware-builder/). It just came 30 minutes ago, you commented at the perfect time :)

So this doesn't look like a zigpy or device handler problem..

Also note: replacing the dongle required me to re-pair all my zigbee devices. There is a backup/import from backup flow when migrating to a new radio, but it didn't work for me.

LukStarkiller, 2024-12-01:
Another one, I'm using an SLZB-06 and it gets stuck at configuring.

martinhoyer, 2025-05-06:
Same
```
File "/usr/local/lib/python3.13/site-packages/zigpy_znp/zigbee/application.py", line 996, in send_packet
    raise DeliveryError(f"Failed to send request: {status!r}", status=status)
zigpy.exceptions.DeliveryError: Failed to send request: <Status.NWK_NO_ROUTE: 205>
```

MattWestb, 2025-05-06:
<Status.NWK_NO_ROUTE: 205> = Your mesh network is not working OK so it cant find one route to the device.
Update the (TI) coordinator firmware to latest recommended and use only good Zigbee 3 routers in the network for fixing the problem.
And TI coordinator like having end devices as direct child and have problems with latest Zigbee 3 network searing functionality (aka Zigbee PRO 2023).

martinhoyer, 2025-05-06:
Thanks @MattWestb! I've tried latest firmware as well as one from https://github.com/Koenkk/Z-Stack-firmware/discussions/545.  
I need to read into how to set device as router as I thought it's set automagically :)

MattWestb, 2025-05-06:
You cant setting one normal sleeping end device as router.
The problem is that some of your network routers is not working OK so the coordinator cant finding one route to the device.
Some tuya devices and old HUE is not reporting there router tables OK and not doing there routing OK.
Try adding it very near the coordinator or open one good router for joining so it using it for the pairing and not other routers that  is making problems.

closed 2025-11-23 by github-actions[bot]