zigpy/zha-device-handlers #3177: [Device Support Request] TO-Q-SY2-JZT

daurnimator, 2024-05-28:
### Problem description

My [`TO-Q-SY2-JZT`](https://elcb.net/product/zigbee-smart-circuit-breaker-over-current-under-voltage-protection-to-q-sy2-jzt/) has the wrong units for summation power delivered.

I note that my `TO-Q-SY1-JZT` works correctly, and has the quirk `zhaquirks.tuya.ts011f_plug.Plug_CB_Metering`.

### Solution description

Maybe the `TO-Q-SY2-JZT` just needs the same quirk(s) as `TO-Q-SY1-JZT`? https://github.com/zigpy/zha-device-handlers/blob/151970c4b45f9588e811a87a4928cb8aaf5ae601/zhaquirks/tuya/ts011f_plug.py#L1245 
But I *do* seem to get the correct temperature for this model unlike the existing comment in the quirk.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_cayepv1a",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

neildsb, 2024-05-28:
is this linked? https://github.com/zigpy/zha-device-handlers/issues/2652#issuecomment-2132445490

daurnimator, 2024-05-28:
> is this linked? [#2652 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2652#issuecomment-2132445490)

ah yes. i think this is a duplicate

closed 2024-11-27 by daurnimator