zigpy/zha-device-handlers #3178: [Device Support Request] TS0601 by _TZE200_nojsjtj2

Urinprobe, 2024-05-28:
### Problem description

This button is registered as an IAS Device. It will never turn off once activated, so it needs to be removed an re-added every time it is used.

### Solution description

Custom quirk that either switches the IAS Zone alarm off when the button is pressed again or simply transforms the device into a switch.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_nojsjtj2",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.6.28-haos-raspi",
    "supervisor": "2024.05.1",
    "host_os": "Home Assistant OS 12.3",
    "docker_version": "25.0.5",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
[... 290 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

HackyPaddy, 2024-08-08:
I need also a ddf. I use the button as a normal switch.
The manufacturer intended it as an SOS button.

MartinRisbyVR, 2024-08-29:
yep, bought 2 of these with the intention of dotting them round the house eventually encase my wife felt a seizure coming on. unfortunately they are as op describes. they are on permanently once clicked/activated.

dummyjmitbe, 2025-04-21:
hi, any update on this device ?

Niek, 2025-04-22:
Following, bought 2 of these and they only show as permanently `on`.

DEVAXTATOR, 2025-06-22:
I gave up used a tuya smart hub and added it and it wkrks

NdK73, 2025-06-23:
> I gave up used a tuya smart hub and added it and it wkrks

Not a solution if there's already a working network... Moreover it would encourage Tuya putting out more quirky devices to sell more hubs.

It's probably waiting for a command to reset its state (like when in the hospital you push the button to call a nurse: you can't turn it off, it's the nurse that cancels the call).