zigpy/zha-device-handlers #3181: [Device Support Request] TS0601 Cover. Tuya _TZE204_xu4a5rhj curtain motor

QuadGhoST, 2024-05-30:
### Problem description

Request to add a quirk for the "DS82-Tuya" curtain motor (cover). This is a zigbee device that looks exactly as other tuya curtain motors.  

### Solution description

A custom quirk that works and allows to control the curtain motor.

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_xu4a5rhj",
  "model": "TS0601",
  "class": "tuya.ts0601.TuyaCover0601_GP"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Asia/Irkutsk",
    "os_name": "Linux",
    "os_version": "6.6.29-haos",
    "supervisor": "2024.05.1",
    "host_os": "Home Assistant OS 12.3",
    "docker_version": "25.0.5",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "documentation": "https://hacs.xyz/docs/configuration/start",
      "version": "1.34.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
[... 622 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

### Additional information

custom quirk gives an error 
 File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 290, in async_close_cover
    raise HomeAssistantError(f"Failed to close cover: {res[1]}")
homeassistant.exceptions.HomeAssistantError: Failed to close cover: <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>

rickd1994, 2024-06-02:
I have the exact same problem

HA-smart, 2024-06-03:
I confirm, same problem

QuadGhoST, 2024-06-07:
This custom quirk works for me
```
"""Tuya MCU based cover and blinds."""
from typing import Dict, Optional, Union
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
    TuyaWindowCover,
    TuyaManufacturerWindowCover,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaMCUCluster,
)

# Maps OPEN/CLOSE/STOP cover commands from Tuya to Zigbee
# https://github.com/zigpy/zigpy/blob/master/zigpy/zcl/clusters/closures.py#L558
# https://developer.tuya.com/en/docs/iot-device-dev/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv#title-7-DP1%20and%20DP4%20Curtain%20switch%201%20and%202
class TuyaCC(t.enum8):
    """Tuya cover commands."""

    OPEN = 0x00
    STOP = 0x01
    CLOSE = 0x02

class ZclCC(t.enum8):
    """ZCL cover commands."""
    OPEN = 0x00
    CLOSE = 0x01
    STOP = 0x02

TUYA2ZB_COMMANDS = {
    ZclCC.OPEN: TuyaCC.OPEN,
    ZclCC.CLOSE: TuyaCC.CLOSE,
    ZclCC.STOP: TuyaCC.STOP,
}

class TuyaWindowCovering(NoManufacturerCluster, WindowCovering, TuyaLocalCluster):
    """Tuya MCU WindowCovering cluster."""

    """Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
        {
            0xF000: ("curtain_switch", t.enum8, True),  # 0: open, 1: stop, 2: close
            0xF001: ("accurate_calibration", t.enum8, True),  # 0: calibration started, 1: calibration finished
            0xF002: ("motor_steering", t.enum8, True),  # 0: default, 1: reverse
[... 132 lines cut ...]
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWindowCoverManufCluster,
                    TuyaWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

closed 2024-12-11 by github-actions[bot]

Prozpect, 2025-01-07:
> This custom quirk works for me
>
> ```
> """Tuya MCU based cover and blinds."""
> from typing import Dict, Optional, Union
> from zigpy.profiles import zha
> import zigpy.types as t
> from zigpy.zcl import [...]

Awesome!! this got my _TZE204_1fuxihti tuya curtain device working!

maxustm, 2025-03-31:
How to add this quirk to release?