zigpy/zha-device-handlers #3184: [Device Support Request] Zemismart Tuya Zigbee 3 Phase Energy Meter SPM02-D2TZ (TS0601 by _TZE200_v9hkz2yn)

sltvtr, 2024-06-03:
### Problem description

Hi,
I have device SPM02-D2TZ:
[image]

Device connects to zigbee network and shows up in HA, but by default it has no entities and no quirk assigned.
After adding custom quirk ts0601_zemismart_power_meter.py (file pasted below) with forced model :
```
        MODELS_INFO: [
            ("_TZE200_bcusnqt8", "TS0601"),
            ("_TZE200_v9hkz2yn", "TS0601"),  # SPM02
        ],
```
the quirk is assigned and some entities show but only for one phase - there should be 3 phases:

[image]

I'm using current version:
Home Assistant
    Core 2024.5.5
    Interfejs użytkownika 20240501.1

Any help/ideas will be appreciated :)
Best regards,
Janusz

### Solution description

The device should be recognized by ZHA without custom quirks and should present data from all 3 phases (not just 1, like now).

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0702",
        "0x0b04",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_v9hkz2yn",
  "model": "TS0601",
  "class": "ts0601_zemismart_power_meter.TuyaZemismartPowerMeter"
}
```

[... 900 lines cut ...]
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    ZemismartManufCluster,
                    ZemismartElectricalMeasurement,
                    ZemismartPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

</details>

sltvtr, 2024-06-03:
One additional information - this device works without any issue by default under zigbee2mqtt, I've just checked it:

[image]

[image]

rrataj, 2024-06-19:
Hey There! I created a quirk for `_TZE200_v9hkz2yn` version which supports 3 phase Voltage, Current and Power stats:
https://github.com/rrataj/zha-device-handlers/blob/0a6b19ab591378c53f6a8887e6c65b76ea31084b/zhaquirks/tuya/ts0601_din_power_3.py

Hope that helps someone :)

sltvtr, 2024-06-19:
Hi! 
Thank You for the contribution, it looks really better now, and after some quick testing I can say it reports independent values from 3 phases.
What bothers me - there are four measurments for power, current and voltage, and I think something is off.
I think the first measurment of the same type should be sum (or maybe average for voltage), like:
PowerABC
PowerA
PowerB
PowerC
but I see that always PowerABC = PowerA, the same for Current and Voltage:

[image]

Am I right?

Thanks again for Your work, it looks promising :-)
Best regards.

rrataj, 2024-06-19:
Yes, you are right, there are 4 groups (ABC, A, B, C), where currently 1st group (ABC) is the same as 1st phase (A). 
Maybe I will find time and adapt it to show correct sum, but for now I do this directly in HA via simple sum:

`{{ (states('sensor.miernik_energii_moc_2')|float(2) + states('sensor.miernik_energii_moc_3')|float(2) + states('sensor.miernik_energii_moc_4')|float(2)  )| round(0) }}`

[image]
[image]

juapem, 2024-07-05:
It worked

juapem, 2024-09-20:
Hi @rrataj have you solved the Power ABC issue? Thanks

rrataj, 2024-09-24:
Unfortunately not, Workaround works fine and I didn't have time to implement a proper solution.

tramix32, 2024-11-18:
I got some unknown values, can we somehow fix that?

[image]

Data can be accessed in zigbee2mqtt
[image]

Device signature
<details>
  <summary>Device signature</summary>

  ```
  {
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0702",
        "0x0b04",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "10": {
      "profile_id": "0x0104",
      "device_type": "0x0053",
      "input_clusters": [
        "0x0b04"
      ],
      "output_clusters": []
    },
    "20": {
      "profile_id": "0x0104",
      "device_type": "0x0053",
      "input_clusters": [
        "0x0b04"
      ],
      "output_clusters": []
    },
    "30": {
      "profile_id": "0x0104",
      "device_type": "0x0053",
      "input_clusters": [
        "0x0b04"
      ],
      "output_clusters": []
    }
  },
[... 2839 lines cut ...]
                "attribute": "ZCLAttributeDef(id=0x0303, name='neutral_current', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0405": {
                "attribute": "ZCLAttributeDef(id=0x0405, name='phase_harmonic_current_multiplier', type=<class 'zigpy.types.basic.int8s'>, zcl_type=<DataTypeId.int8: 40>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0403": {
                "attribute": "ZCLAttributeDef(id=0x0403, name='power_divisor', type=<class 'zigpy.types.basic.uint32_t'>, zcl_type=<DataTypeId.uint32: 35>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0510": {
                "attribute": "ZCLAttributeDef(id=0x0510, name='power_factor', type=<class 'zigpy.types.basic.int8s'>, zcl_type=<DataTypeId.int8: 40>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0910": {
                "attribute": "ZCLAttributeDef(id=0x0910, name='power_factor_ph_b', type=<class 'zigpy.types.basic.int8s'>, zcl_type=<DataTypeId.int8: 40>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0a10": {
                "attribute": "ZCLAttributeDef(id=0x0A10, name='power_factor_ph_c', type=<class 'zigpy.types.basic.int8s'>, zcl_type=<DataTypeId.int8: 40>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0402": {
                "attribute": "ZCLAttributeDef(id=0x0402, name='power_multiplier', type=<class 'zigpy.types.basic.u

thomo66, 2025-01-30:
This work works fantastically:
https://app.codecov.io/gh/zigpy/zha-device-handlers/commit/90dcb4e969c96045ad6c12b09dfdf892de75188b

evlas, 2025-02-15:
I slightly modified the file to also get the sum of the three-phase power and the powerfactor, but how do I change the names of the created sensors?

`"""Tuya Din Power Meter."""

from zigpy.quirks.v2.homeassistant import (
    UnitOfElectricCurrent,
    UnitOfElectricPotential,
    UnitOfEnergy,
    UnitOfFrequency,
    UnitOfPower,
)
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t

from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE200_v9hkz2yn", "TS0601")
    .tuya_sensor(
        dp_id=1,
        attribute_name="energy_consumed",
        fallback_name="Total energy consumed",
        type=t.uint16_t,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        divisor=100,
        translation_key="energy_consumed",
    )
    .tuya_sensor(
        dp_id=2,
        attribute_name="energy_produced",
        fallback_name="Total energy produced",
        type=t.uint16_t,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        divisor=100,
        translation_key="energy_produced",
    )
    .tuya_sensor(
        dp_id=111,
        attribute_name="total_active_power",
        fallback_name="total_active_power",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
    )
#    .tuya_sensor(
#        dp_id=32,
#        attribute_name="ac_frequency",
#        fallback_name="ac_frequency",
#        type=t.uint16_t,
#        state_class=SensorStateClass.MEASUREMENT,
#        device_class=SensorDeviceClass.FREQUENCY,
#        unit=UnitOfFrequency.HERTZ,
#        divisor=100,
#    )
    .tuya_sensor(
        dp_id=15,
        attribute_name="power_factor",
        fallback_name="power_factor",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER_FACTOR,
#        unit=PERCENTAGE,
    )
    .tuya_sensor(
        dp_id=102,
        attribute_name="voltage_phase_r",
        fallback_name="voltage_phase_r",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.VOLTAGE,
        divisor=10,
        unit=UnitOfElectricPotential.VOLT,
    )
    .tuya_sensor(
        dp_id=103,
        attribute_name="current_phase_r",
        translation_key="current_phase_r",
        fallback_name="Current (phase A)",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.CURRENT,
        divisor=1000,
        unit=UnitOfElectricCurrent.AMPERE,
    )
    .tuya_sensor(
        dp_id=104,
        attribute_name="power_phase_r",
        fallback_name="power_phase_r",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
    )
    .tuya_sensor(
        dp_id=105,
        attribute_name="voltage_phase_s",
        fallback_name="voltage_phase_s",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.VOLTAGE,
        divisor=10,
        unit=UnitOfElectricPotential.VOLT,
    )
    .tuya_sensor(
        dp_id=106,
        attribute_name="current_phase_s",
        fallback_name="current_phase_s",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.CURRENT,
        divisor=1000,
        unit=UnitOfElectricCurrent.AMPERE,
    )
    .tuya_sensor(
        dp_id=107,
        attribute_name="power_phase_s",
        fallback_name="power_phase_s",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
    )
    .tuya_sensor(
        dp_id=108,
        attribute_name="voltage_phase_t",
        fallback_name="voltage_phase_t",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.VOLTAGE,
        divisor=10,
        unit=UnitOfElectricPotential.VOLT,
    )
    .tuya_sensor(
        dp_id=109,
        attribute_name="current_phase_t",
        fallback_name="current_phase_t",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.CURRENT,
        divisor=1000,
        unit=UnitOfElectricCurrent.AMPERE,
    )
    .tuya_sensor(
        dp_id=110,
        attribute_name="power_phase_t",
        fallback_name="power_phase_t",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
    )
    .add_to_registry()
)
`