zigpy/zha-device-handlers #3185: Tuya T&H devices give in ZHA only half of the battery percentage (so it is 100% end ZHA reports 50%)

WilbertWilbers, 2024-06-03:
### Problem description

The Temperature and Humidity devices give in ZHA only half of the battery percentage that was presented in the Tuya Integration before and also  in the Zigbee device configuration

[image]

[image]

### Solution description

Solve the issue?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

[image]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

HelgeGitHub, 2024-06-09:
Same issue for _TZE204_yjjdcqsq TS0601: From the beginning of usage always battery 50%

WilbertWilbers, 2024-10-16:
It was corrected! Thank you!

closed 2024-10-18 by WilbertWilbers