zigpy/zha-device-handlers #3187: [Device Support Request] Aqara Wall Outlet H2 EU (WP-P01D)

poughkeepsee, 2024-06-03:
### Problem description

Hi,

Requesting, if possible, for support for the new H2 Wall Outlet ([link](https://www.aqara.com/eu/product/wall-outlet-h2-eu/)).

[image]

I can add it to ZHA and it's recognized as _lumi.plug.aeu001_, but seems there are issues with the current config/support:
- Shows 2 switches when this is a single unit (switch_1 does nothing when turning on/off, switch_2 turns device on/off)
- Power reading has ~40sec delay, not sure if intended/expected
- Temperature does not register (always at 0)
- Only exposed entities are: switches (1, 2), Power, Summation delivered, Temperature. Looking at the Zigbee2MQTT [page](https://www.zigbee2mqtt.io/devices/WP-P01D.html) it seems there are many other available options: _switch (state), device_temperature, power_outage_count, power_on_behavior, power, energy, voltage, current, overload_protection, led_indicator, button_lock, charging_protection, charging_limit, linkquality_

Home Assistant version: 
Core - 2024.5.5
Supervisor - 2024.05.1
OS - 12.3

### Solution description

Full support of device/fix existing bugs. Possibly mirror Zigbee2MQTT level of support and options.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.plug.aeu001",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

ChristophCaina, 2024-06-07:
Hi, I would also love to see this device fully supported :)
[image]

I agree with the topics mentioned above:
- Switch 1 is not working
- Switch 2 is working as expected.
If you turn on the device with the built in switch, both switch entities will be turned on in HA.

Power is working (with a small delay)
Energy is not working
Temperature is not working

Identify is working
LQI is working
RSSI is working

poughkeepsee, 2024-06-10:
Just adding in case of interested/usefulness, I tested adding this device to Zigbee2MQTT and below are the exposed entities. Really hoping for future support in this outlet, thanks!

[image]
[image]
[image]

ChristophCaina, 2024-06-25:
ok, not sure where this is comming from, but somehow, it seems, that I do have some kind of kWh reporting (suddenly)

[image]

[image]

ronaldtveen, 2024-07-09:
The Wall Outlet H2 EU was just reviewed by Tweakers.net (largest Dutch website dedicated to technology and electronics). I expect a lot more users buying this device now. Me too, but since I use ZHA I'm on the brink if I want it because of the sloppy support in ZHA.

[Tweakers.net: Dumb sockets become smart (Aqara H2 Quicktest)](https://tweakers.net/reviews/12138/aqara-h2-quicktest-domme-stopcontacten-worden-slim.html)

poughkeepsee, 2024-09-07:
Commenting to keep this alive, still hoping for some development to happen.

ronaldtveen, 2024-09-10:
Also, might this Device Description File (DDF) from Deconz help help? 

[DDF for Aqara wall outlet H2 EU (WP-P01D)](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7846)

homerojleal, 2024-09-10:
Yes also looking fw for this fix/integration/quirk please!

timiman, 2024-10-05:
Any development updates regarding this device's support with ZHA ? Hope someone will tackle this.

n0x31x, 2024-10-14:
Also here some interested to have it fully working

tamasd8, 2024-10-28:
Also some interest from me for a good integration, I haven't bought it yet, but this seems like a good product to invest in.

ChristophCaina, 2024-10-30:
The PR which I've started is still open - unfortunately, I don't have the time to continue with this one - so if anyone with some knowledge on this area might to continue here? ^^

akimtsev, 2024-11-08:
I support it, full support of the device in ZHA is very necessary

dolzhanskiyoa, 2025-02-13:
Please, add full support!

kokrp, 2025-03-12:
Please, add full support, also for the Light Switch H2 EU.

samaniel, 2025-05-17:
Please, add full support!

ChristophCaina, 2025-09-16:
can someone pls. review the linked pr's ?
This issue is now open for over a year (opened in June 24) - and the PR is pending for 9 months now... :-(

edvardasast, 2025-10-02:
Also have a lot of these plugs and can't use it normaly with ZHA

ChristophCaina, 2025-10-02:
currently, I am testing the PR as a custom quirk.
It does provide much more information, but it seems that my plug has some issues (doesn't report the voltage and kwh at all at the moment)...

it is running Firmware '1e' (30)... and I am waiting for an aqara hub to test if a firmware update might help here.

groovee, 2025-11-12:
Thank you @andreibsk 
I confirm it works well.

HA: 2025.11.1
HAOS: 15.2
Device: Wall Outlet H2 EU
Connected via [Skyconnect]
Firmware: 0x0000001e

Before:
Two switches, summation delivered not working

After: Firmware & Temperature unknown.
I did not test the charging protection.
The other things seems to work fine!
Overload protection works well.

[image]
[image]
[image]
[image]
[image]