zigpy/zha-device-handlers #3188: [Device Support Request] SONOFF SNZB-03P

olikami, 2024-06-04:
### Problem description

- Motion is always stuck in detected
- Dim and bright is not reported

### Solution description

Make motion work properly, add the dim and bright values

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0406",
        "0x0500",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "SNZB-03P",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Zurich",
    "os_name": "Linux",
    "os_version": "6.6.28-haos-raspi",
    "supervisor": "2024.05.1",
[... 406 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

g3gg0, 2024-06-09:
With this device and three different USB zigbee dongles (CC-based, SONOFF Zigbee 3.0 USB Dongle Plus V2 and Home Assistant SkyConnect) with older as well as with latest firmware flashable, i experience also problems.
tried re-registering a few times, remigrating between dongles and firmware versions.

This device is practically unusable with ZHA.

*If* i can add it to ZHA, it will take a  few minutes to hours until nothing is received anymore.
it doesnt react to any identify or attribute access, nor does it send any updates.
after some time - not sure, think it was a few hours - the device gets "unavailable"

remoteassist, 2024-07-18:
I was able to change the motion status by polling one of the IasZone attribute, but it follow the same status as the Occupancy. Maybe HA is not polling the motion status, it will be stuck in the last status if I don't poll it. Also needed to use the latest firmware for my Sonoff P Coordinator for it to work.  
[image]

chaot2000, 2024-09-15:
Is there any option to set the motion delay?

Nbird67, 2024-09-19:
+1 for SNZB-03P full support

blair287, 2024-09-20:
+1 too

ThomasChelle, 2024-09-21:
+1 for SNZB-03P full support

wwhite139, 2024-12-05:
+1 for SNZB-03P full support

w1an, 2024-12-06:
Brand new SNZB-03P added to HA 2024.11 using SkyConnect with firmware 7.4.4.0 build 0 and ZHA. Pairing took a while, but eventually completed at first try. But since pairing no motion detected. Battery 100%. Please fix :)

Nbird67, 2024-12-27:
It’s not possible to use a modified snzp-06p Quirk ? It has the light sensor too.

BillyBlaze, 2025-01-12:
I have created a Quirk, please note this is my first time working with ZHA and Quirks. So please only apply this if you know what you're doing (in case I made an oversight). @TheJulianJES if you have time for a pre-review than that would be great. If all good and you want a Pull Request than I can do that ofcourse!

With this Quirk you get the same functionality as the [Zigbee2MQTTT implemenation](https://www.zigbee2mqtt.io/devices/SNZB-03P.html).

<details>
<summary>ewelink_snzb03p.py</summary>

```python
"""Sonoff SNZB-03P - Zigbee presence sensor."""

from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import QuirkBuilder, NumberDeviceClass
import zigpy.types as t
from zigpy.zcl.foundation import BaseAttributeDefs, ZCLAttributeDef
from zigpy.zcl.clusters.measurement import OccupancySensing
from zigpy.zcl.clusters.security import IasZone
from zigpy.quirks.v2.homeassistant import EntityType, UnitOfTime
from zha.application import Platform

class LastIlluminationState(t.enum8):
    """Last measureed state of illumination enum."""

    Dark = 0x00
    Light = 0x01

class SonoffIlluminationCluster(CustomCluster):
    """Custom Sonoff illumination cluster."""

    cluster_id = 0xfc11

    class AttributeDefs(BaseAttributeDefs):
        """Attribute definitions."""

        last_illumination_state = ZCLAttributeDef(
            id=0x2001,
            type=LastIlluminationState,
            access="rp",
            is_manufacturer_specific=True,
        )

    @property
    def _is_manuf_specific(self):
        return False

(
    QuirkBuilder("eWeLink", "SNZB-03P")
    .replaces(SonoffIlluminationCluster)
    .removes(IasZone.cluster_id) #  IasZone cluster 0x0500: remove motion detection
    #
    # I don't know why the detection timeout magicly shows up now
    # it did not do it before and once I manually added this I magicly had 2 
    # configuration sliders
    #
    # .number(
    #     OccupancySensing.AttributeDefs.ultrasonic_o_to_u_delay.name,
    #     OccupancySensing.cluster_id,
    #     min_value=5,
    #     max_value=60,
    #     step=1,
    #     unit=UnitOfTime.SECONDS,
    #     device_class=NumberDeviceClass.DURATION,
    #     translation_key="motion_timeout",
    #     fallback_name="Presence detection timeout",
    # )
    #
    .enum(
        SonoffIlluminationCluster.AttributeDefs.last_illumination_state.name,
        LastIlluminationState,
        SonoffIlluminationCluster.cluster_id,

        entity_platform=Platform.SENSOR,
        entity_type=EntityType.STANDARD,

        translation_key="last_illumination_state",
        fallback_name="Last illumination state",
    )
    .add_to_registry()
)
```
</details>

Device:
[image]

- Initial pairing seems also to be working now without needing to reload ZHA or Home Assistant
- Last illumination state is Unknown before first detection

cubical23, 2025-01-14:
good job, your quirk works like a charm

johngalt991, 2025-01-14:
Works like a charm(2)!

Huge thanks, @BillyBlaze. Appreciate it!

ezcGman, 2025-01-18:
Awesome, great work!! Thanks for this!

Weirdly, I always get two "Presence detection timeout" sensors added:
[image]
* number.ewelink_snzb_03p_presence_detection_timeout
* number.ewelink_snzb_03p_presence_detection_timeout_2

Both of them sync each other, so if I change one or the other, the other one changes, too. So I disabled the one with the "_2" prefix 🤷‍♂ 

Another thing:
The "Last illumination state" only changes if motion is detected. Is that know / not possible anyways? Or maybe a bug?

Thanks again!!

Nbird67, 2025-01-18:
I removed this part on my side that is unnecessary

number(
        OccupancySensing.AttributeDefs.ultrasonic_o_to_u_delay.name,
        OccupancySensing.cluster_id,
        min_value=5,
        max_value=60,
        step=1,
        unit=UnitOfTime.SECONDS,
        device_class=NumberDeviceClass.DURATION,
        translation_key="motion_timeout",
        fallback_name="Presence detection timeout",

BillyBlaze, 2025-01-19:
Yeah, I have contacted TheJulianJES on discord about this problem. The reason why this shows up is because slider of the SNZB-06P shows up in SNZB-03P (not related to this quirk). However, the minimal detection value of the 06P is 15 seconds while the 03P can go all the way down to 5.

If you can live with 15 than remove the number. You might need to add `.adds(OccupancySensing.cluster_id)`. Because without this both sliders would disappear after a while on my side.

<details>
<summary>ewelink_snzb03p.py without number</summary>

```python
"""Sonoff SNZB-03P - Zigbee presence sensor."""

from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import QuirkBuilder, NumberDeviceClass
import zigpy.types as t
from zigpy.zcl.foundation import BaseAttributeDefs, ZCLAttributeDef
from zigpy.zcl.clusters.measurement import OccupancySensing
from zigpy.zcl.clusters.security import IasZone
from zigpy.quirks.v2.homeassistant import EntityType, UnitOfTime
from zha.application import Platform

class LastIlluminationState(t.enum8):
    """Last measureed state of illumination enum."""

    Dark = 0x00
    Light = 0x01

class SonoffIlluminationCluster(CustomCluster):
    """Custom Sonoff illumination cluster."""

    cluster_id = 0xfc11

    class AttributeDefs(BaseAttributeDefs):
        """Attribute definitions."""

        last_illumination_state = ZCLAttributeDef(
            id=0x2001,
            type=LastIlluminationState,
            access="r",
            is_manufacturer_specific=True,
        )

    @property
    def _is_manuf_specific(self):
        return False

(
    QuirkBuilder("eWeLink", "SNZB-03P")
    .replaces(SonoffIlluminationCluster)
    # IasZone.cluster_id (0x0500):
    #  Remove motion detection since this is not
    #  implemented by the manufacturer.
    .removes(IasZone.cluster_id)
    .adds(OccupancySensing.cluster_id)
    # .number(
    #     OccupancySensing.AttributeDefs.ultrasonic_o_to_u_delay.name,
    #     OccupancySensing.cluster_id,
    #     min_value=5,
    #     max_value=60,
    #     step=1,
    #     unit=UnitOfTime.SECONDS,
    #     device_class=NumberDeviceClass.DURATION,
    #     translation_key="presence_detection_timeout",
    #     fallback_name="Presence detection timeout",
    # )
    .enum(
        SonoffIlluminationCluster.AttributeDefs.last_illumination_state.name,
        LastIlluminationState,
        SonoffIlluminationCluster.cluster_id,

        entity_platform=Platform.SENSOR,
        entity_type=EntityType.STANDARD,

        translation_key="last_illumination_state",
        fallback_name="Last illumination state",
    )
    .add_to_registry()
)
```
</details>

BillyBlaze, 2025-01-20:
It's buried deep inside the core of ZHA, at least this is what I found over here: https://github.com/zigpy/zha/blob/dev/zha/application/platforms/number/__init__.py#L888-L901

HansBinBrabant, 2025-01-20:
Ha guys, I have bought some Sonoff stuff, together with two x Sonoff SNZB-03P's https://www.tinytronics.nl/nl/domotica/sensoren/beweging-en-aanwezigheid/sonoff-snzb-03p-bewegingssensor-zigbee.
With zigbee ZHA dongle : https://www.tinytronics.nl/nl/platformen-en-systemen/sonoff/zigbee/sonoff-zigbee-3.0-usb-dongle-plus-zbdongle-e
In a raspberry pi 4.

Nice stuff. For the SonOff SNZB-03P, I do have the issue discussed here. So I am also interested in a solution.
Thanks in advance !
Hans

mrogojanu, 2025-02-07:
@BillyBlaze thank you for provided quirk code. In my case, the sensor seems to work normally for the first few minutes after a battery reset (with max two normal detections and clearings), but then it freezes on occupancy detected. Otherwise this is progress compared to the original firmware :) where it stays always stuck in the detected mode.  I would appreciate any ideas on how to further adjust the quirk code.

Hebichel, 2025-03-11:
> I have created a custom quirk, please note this is my first time working with ZHA and Quirks. So please only apply this if you know what you're doing (in case I made an oversight). [@TheJulianJES](https://github.com/TheJulianJES) if you have time for a pre-review than that would be great. If all good and you want a Pull Request than I can do that ofcourse!
> 
> With this Quirk you get the same functionality as the [Zigbee2MQTTT implemenation](https://www.zigbee2mqtt.io/devices/SNZB-03P.html).
> ewelink_snzb03p.py
> 
> Device: [image]
> 
>     * Initial pairing seems also to be working now without needing to reload ZHA or Home Assistant
> 
>     * Last illumination state is Unknown before first detection

I am new to HA. 

Pls tell me what exactly to do with your  ewelink_snzb03p.py
I have same problems with the snzb-03p and it is driving me crazy *SS*

Brgds
Helge

MorphBonehunter, 2025-03-25:
As this works like a charm, is there something which prevents a creation an PR to intergrade it in the repo?
@BillyBlaze do you have any concerns?

panther7, 2025-03-25:
@MorphBonehunter done #3979

3jam3, 2025-04-30:
Any progress on the horizon?

BillyBlaze, 2025-07-04:
great work @panther7, thanks for picking it up further!

carterbox, 2025-09-28:
I had added support for configuration of the timeout and sensitivity through the UI in October of 2024 in https://github.com/zigpy/zha/pull/215. However, I'm not sure if these changes are still effective for devices added today.

BillyBlaze, 2025-10-21:
@carterbox that change you made did gave me a bit of confusion, see here:  https://github.com/zigpy/zha-device-handlers/issues/3188#issuecomment-2600762664

tjl10, 2025-11-18:
Does anyone know the next steps to resolve this?
Are we waiting for the PR to be merged, or are the reviewers waiting for further changes?

panther7, 2026-01-07:
@3jam3 Good job, could you create new PR? I will close my PR #3979.