zigpy/zha-device-handlers #3189: [Device Support Request] - TS011F by _TZ3000_cehuw1lw

nyxk2028, 2024-06-06:
### Problem description

I bought some Aliexpress zigbee smart plugs awhile ago and they worked mint right out of the box. I decided to get some more and when the new plugs where paired the energy metering current and (assuming therefore) KWh values were off by 1000x. I have seen this is a somewhat common problem but I can not for the life of me follow their methods for solutions.

I tried searching for resources to try and learn but I was finding a whole lot of missing information making it very hard to follow.

The only thing I understand is a quirk is a translator.
Why are the first batch of smart plugs working flawlessly and the second batch not? (Bought from same seller 3 months apart)
How did some of my devices just 'find' a quirk to use?
How do you create a custom quirk?
How do you assign a specific device to use a specific quirk?

It looked like one could change cluster attributes eg, 'ac_current_divisor' from 1 to 1000 but when writing the value and reading it again it shows a value of 1. So that didn't work either

I think my problem is identical to this one but again, can not follow what to do
https://github.com/zigpy/zha-device-handlers/issues/1764

Any help would be greatly appreciated.

### Solution description

Produce actuate value

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4107, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=11264, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0x1888",
        "0xe000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_cehuw1lw",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

BambamNZ, 2024-06-16:
I find myself in the same situation, I tried a number of mentioned solutions but can't seems to solve the problem with the power meter reporting the current draw 1000 x higher than actual.

The thing that is odd, after enabling ZHA Quirks, the Plug is not showing Quirk as applied

`    "manufacturer": "_TZ3000_cehuw1lw",
    "model": "TS011F",
    "name": "_TZ3000_cehuw1lw TS011F",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,`

While I can see quirk is applied for my other plugs

`    "manufacturer": "_TZ3000_0zfrhq4i",
    "model": "TS011F",
    "name": "_TZ3000_0zfrhq4i TS011F",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.tuya.ts011f_plug.Plug_v3",
    "quirk_id": "tuya.plug_on_off_attributes",`

Can also see this is the problem, comparing the diagnostic output from a working "_TZ3000_0zfrhq4i TS011F"

`          "0x0b04": {
            "endpoint_attribute": "electrical_measurement",
            "attributes": {
              "0x0603": {
                "attribute_name": "ac_current_divisor",
                "value": 1`

The value should be 1000 

I tried to create a local quirk for just this model "_TZ3000_0zfrhq4i TS011F" but honestly I have no clue, jut stitched something together from other examples.

nyxk2028, 2024-06-16:
> I find myself in the same situation, I tried a number of mentioned solutions but can't seems to solve the problem with the power meter reporting the current draw 1000 x higher than actual.
>
> The [...]

Yes my 0zfrhq4i plugs have that same quirk applied to them. I just don't understand how to get a quirk applied to a specific device/manufacturer.

In the meantime before this gets resolved, I have created a sensor that just divides the original energy value by 100 and use that entity for the energy monitoring dashboard.

BambamNZ, 2024-06-16:
I've created the follow quirk file in an attempt to solve this, but struggle to get it to match the "TZ3000_cehuw1lw" then I uncomment the "MODELS_INFO" section it matches the existing "TZ3000_0zfrhq4i" devices. As I can see they are loading this quirk. 

`"""tuya TS011F Smart Plug."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)

from zhaquirks.tuya.mcu import EnchantedDevice
from zigpy.zcl.clusters.lightlink import LightLink

class Tuya_TS011F_Custom(EnchantedDevice):
    """Tuya metering support."""

    signature = {
        MODELS_INFO: [
           ("_TZ3000_cehuw1lw", "TS011F"),
        ],
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # input_clusters=["0x0000","0x0003","0x0004","0x0005","0x0006","0x000a","0x0702","0x0b04","0x1000","0x1888","0xe000"]
            # output_clusters=["0x0019"]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,     
                    TuyaZBMeteringCluster.cluster_id,
                    TuyaZBElectricalMeasurement.cluster_id,
                    LightLink.cluster_id, 
                    0x1888,
                    TuyaZBE000Cluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=["0x0021"]>
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Time.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    LightLink.cluster_id, 
                    TuyaZBE000Cluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }`

LAK132, 2024-07-02:
Also having problems with _TZ3000_cehuw1lw models specifically (summation delivered appears to be off by 100x), while _TZ3000_typdpbpg models seem to work just fine.

danielp370, 2024-07-04:
I'll try put a patch up for this

danielp370, 2024-07-04:
Can someone please try https://github.com/zigpy/zha-device-handlers/pull/3242

BambamNZ, 2024-07-04:
> Can someone please try #3242

If you can help with some steps to test, happy to assist.

danielp370, 2024-07-04:
Thanks - if you haven't used zha quirks manually before follow the instructions here: https://community.home-assistant.io/.../how-to.../341226/6 <https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226/6?fbclid=IwZXh0bgNhZW0CMTAAAR1L97KKnnmm0FEC0eiaPv0BgLpLAOFDHhVDrK2v7xEy2_dlwZpV3QxszD8_aem_0KBmh1iUpC1WD28kuF1wKQ> then download the quirk file by downloading the whole raw file here https://github.com/.../24ec.../zhaquirks/tuya/ts011f_plug.py <https://github.com/zigpy/zha-device-handlers/blob/24ec41f833981c74020aa2b5760b0ebb56d9f05c/zhaquirks/tuya/ts011f_plug.py?fbclid=IwZXh0bgNhZW0CMTAAAR1cq_HArCXYJzS6FBj54PbDYo7eLkvi7BUIbkNth-rKDmTJiZIFOvpVE1I_aem_CvJ-OYuw5A1GC7TSYNA_Gg> and put ts011f_plug.py <https://l.facebook.com/l.php?u=http%3A%2F%2Fts011f_plug.py%2F%3Ffbclid%3DIwZXh0bgNhZW0CMTAAAR2uUc2WTxF2dPBOMcDR28UyKinGwGXDsPlLZeqwtMw9TVEtxGtbJ1Xfia4_aem_N-UX2kgoaW-VldWfX0xLvA&h=AT1q4j17EDWFWmU9oySbEnx8d_BLpULlqmOsCCT1_MW_dtqrCOBsKWoRg7vThlNlm3BNFJ2wBh0M210H3rtx_wp46dxhCzVqVsVZPromxhTzhRuLU5vegdqL7Rq-VO8FYY0blIAzAmnqIF-BPUY2DVlIvQ&__tn__=R]-R&c[0]=AT1UDSCJ5tZYymABvtADFeJ6ZPSsFsz4szXjt0sZBzQ6OloQv3qP_DPCGY4R_cvU-ywpS9v5r-bWCv1m7m2ey7_it2kqQWm00L9D_aKkJm_YEYktgia6j8NBWjJ-LndmBVgRDGT6_y9nHVD03_6LatyfYuAd3SZFPlcdHMa4TKDbXw7UqmwYEWUWig> in /config/zha-quirks/

BambamNZ, 2024-07-04:
> if you haven't used zha quirks manually before follow the instructions here: https://community.home-assistant.io/.../how-to.../341226/6 then download the quirk file by downloading the whole raw file here https://github.com/.../24ec.../zhaquirks/tuya/ts011f_plug.py and put ts011f_plug.py in /config/zha-quirks/

Thanks - Just did know where to get the whole ts011.py file but figured that out luckily

danielp370, 2024-07-04:
> Thanks - Just did know where to get the whole ts011.py file but figured that out luckily

Great thanks - I fixed up my links above too.

BambamNZ, 2024-07-04:
Can also now see the configuration entities as well which as been absent 

[image]

danielp370, 2024-07-04:
Great, thanks for testing. I'll update the PR comments.

nyxk2028, 2024-07-04:
Thank you very much! seems to all be working as it should now.
[image]