zigpy/zha-device-handlers #3192: [Device Support Request] TS0601 _TZE284_g2e6cpnw Smart Soil Sensor

menaibluey, 2024-06-08:
### Problem description

Did not find any existing issues for this device [https://www.aliexpress.com/item/1005007061020176.html?spm=a2g0o.order_list.order_list_main.4.260f1802vAnJ2p](url)

Device only shows LQI and RSSI entities.

### Solution description

Would be great to have a quirk for this. I'm new to this and not confident that I would be able to configure this. Any help would be greatly appreciated.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_g2e6cpnw",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.6.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "6.6.29-haos",
    "supervisor": "2024.06.0",
    "host_os": "Home Assistant OS 12.3",
[... 360 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

menaibluey, 2024-06-15:
I've switched to Zigbee2MQTT where this sensor is supported in the zigbee-herdsman-converter, see  

[Tuya Soil sensor - TS0601 TZE284_g2e6cpnw](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7609)

At this stage I have no need for a ZHA quirk for this device as it's now supported in Zigbee2MQTT.

closed 2024-06-15 by menaibluey

llnz, 2024-07-06:
I have purchased this device and would like to use it with ZHA. I will have a go at developing a custom quirk.

tube0013, 2024-07-06:
Looks like it could be added using the existing class here: https://github.com/zigpy/zha-device-handlers/blob/8317dc69c3b36e7ae5cabe40d7937c02ca259f77/zhaquirks/tuya/ts0601_sensor.py#L335

The dps from the tuya side look the be the same based on the z2m converted linked. https://github.com/Koenkk/zigbee-herdsman-converters/blob/ad177a3c45c63713f74703131c1124cc377570bb/src/devices/tuya.ts#L1950

The supported soil sensors don't seem to match the signature, this one looks to have an extra cluster "0xed00"  and the battery calculation may be different as this device has 3x AAA vs 2xAAA in the currently supported devices.